ikwzm/udmabuf #2: CONFIG_ARM

igalkr, 2017-06-24:
Hi,

Great job on the driver.
Could you elaborate on the **CONFIG_ARM**, where is it set and when?

This is actually brings me to the next questions:
1) Why the **dma_mmap_coherent** is utilized when **CONFIG_ARM** is not defined? 
2) There is no mapping applied in the **udmabuf_driver_vma_fault** - as I saw in other samples which have utilized **remap_pfn_range** - https://github.com/claudioscordino/mmap_alloc/blob/master/mmap_alloc.c
3) **https://www.embeddedrelated.com/showthread/comp.arch.embedded/217578-1.php** states that the **vma->vm_pgoff** has to be set to zero prior to invoking **dma_mmap_coherent**, why you do not reset this value?

Thanks,
Igal

ikwzm, 2017-06-26:
Thank you for the issue.

A.1

Originally I made udmabuf assuming ARM architecture only.
At that time, cache coherency control between buffer and CPU was necessary.
For that reason I did not use dma_mmap_coherent (), because I did not know how to specify cache coherency control with dma_mmap_coherent ().

At one time I try to build on the x86 architecture as well as the ARM architecture.
At that time, the following two problems occurred.
 1. pgprot_dmacoherent () was undefined on x86, pgprot_dmacoherent () seems to be for ARM only.
 2. udmabuf used on-demand paging mechanism for page mapping, but it did not work well on x86.

So I changed it so that I do not use on-demand paging outside the ARM architecture.
However, as a side effect, caching coherency control can not be controlled outside the ARM architecture.
That's why we switched the code with the definition of SYNC_ENABLE.

A.2

udmabuf_driver_vma_fault()  referred to [Linux Device Drivers, Third Edition] (https://lwn.net/Kernel/LDD 3/)  , "Chapter 15: Memory Mapping and DMA",  "Mapping Memory with nopage".

A.3

I did not know this information. 
Thank you very much.
 I am going to use it as an example.

igalkr, 2017-06-26:
Thank you very much for your time and explanations.

Igal

Igal Kroyter

closed 2019-02-24 by ikwzm