ikwzm/udmabuf #4: udmabuf initiate transfer?

agamez, 2017-09-05:
Hi,

This question maybe arises from a misconception, so please do tell me where I'm wrong.

Once I've loaded this module, I see that I can read/write from /dev/udmabuf0 and read/write from the segment of memory associated to the DMA device. However, it seems that udmabuf won't launch the DMA transfer itself.

If this is the case, how am I supposed to order from userspace RX or TX transfer to really occur?

Thanks!

ikwzm, 2017-09-05:
Hi,

As you pointed out, udmabuf will not activate DMA transfers. The ```udmabuf``` only provides a DMA buffer. Another means is required to start DMA transfer.

For example, there is a sample project for DMA transfer using ```udmabuf``` to the following URL.

 - https://github.com/ikwzm/ZYBO_UIO_IRQ_SAMPLE

In this project, the DMA controller implemented on the FPGA is controlled from a program operating in user space. The purpose of using ```udmabuf``` is to control the DMA controller without writing device drivers that operate in kernel space.

anish12321, 2018-10-28:
Hi ikwzm,

Do you have a sample project for DMA transfer for ZynqMP? I see that the register configurations are different between ZYBO and ZynqMP and it would be really helpful if you have a sample project to refer to.

Thanks!

ikwzm, 2018-10-28:
Hi,

Here is the example of ZynqMP which does not have DMA, but reads and writes the PL side memory with Vivado HLS.

https://github.com/ikwzm/ZynqMP-FPGA-Linux-Example-2-UltraZed

anish12321, 2018-10-28:
Thank you so much. I’ll check this one out.

Thanks,
Anish

honorpeter, 2019-06-12:
@ikwzm 

when I boot udmabuf with  your code  on ZCU102  (petalinux 2018.2). it does not  work .  I don't kown why.

[    5.830825] [<ffffff8000a4b078>] udmabuf_module_init+0x78/0x1000 [udmabuf]
[    5.837673] [<ffffff8008083980>] do_one_initcall+0x38/0x128
[    5.843229] [<ffffff800810bcd0>] do_init_module+0x58/0x1b0
[    5.848696] [<ffffff800810ac44>] load_module+0x19fc/0x2150
[    5.854165] [<ffffff800810b5b8>] SyS_finit_module+0xa8/0xc0
[    5.859719] Exception stack(0xffffff800ca9bec0 to 0xffffff800ca9c000)
[    5.866144] bec0: 0000000000000008 0000007fb9d8c8f0 0000000000000000 0000000000000008
[    5.873958] bee0: 0000000000000000 60ceffffffffffff ffffffffffffffff ffffffffffffffff
[    5.881771] bf00: 0000000000000111 0000000000000038 0101010101010101 0000000000000001
[    5.889584] bf20: 0000000000000000 ffffffffffff0000 0000000000000033 0000000000000004
[    5.897397] bf40: 0000007fb9d9d2a8 0000007fb9cf5a80 0000000000000000 0000000037af4e30
[    5.905209] bf60: 0000007fb9d8c8f0 0000000000000000 0000000000020000 0000000037af4e30
[    5.913022] bf80: 0000000000000000 0000000000000000 0000000000000001 0000000000431100
[    5.920835] bfa0: 0000000000000000 0000007fda0ba160 0000007fb9d86260 0000007fda0ba160
[    5.928648] bfc0: 0000007fb9cf5aa4 0000000040000000 0000000000000008 0000000000000111
[    5.936460] bfe0: 0000000000000000 0000000000000000 0000000000000000 0000000000000000
[    5.944273] [<ffffff80080837b0>] el0_svc_naked+0x24/0x28
[    5.949565] ---[ end trace d6a40af423b1d8ea ]---
[    5.954194] udmabuf: couldn't create sys class. return=-17

ikwzm, 2019-06-12:
Thank you for the issue.
However, posting to this thread is inappropriate. Please open a new issue.

ikwzm, 2019-06-20:
This issue is confusing, so I will close it once.

closed 2019-06-20 by ikwzm