ikwzm/udmabuf #9: Leaking memory

nitsch, 2018-06-12:
When configuring a udmabuf device in the device tree and using reserved memory, repeatedly loading and unloading the kernel module seems to leak the memory. Here is the relevant portion of the device tree:
```
reserved-memory {
	#address-cells = <1>;
	#size-cells = <1>;
	ranges;

	fpga_reserved1: buffer@20000000 {
		compatible = "shared-dma-pool";
		no-map;
		reg = <0x20000000 0x08000000>;
	};
};

udmabuf@0 {
	compatible = "ikwzm,udmabuf-0.10.a";
	device-name = "udmabuf0";
	minor-number = <0>;
	size = <0x00C80000>;
	memory-region = <&fpga_reserved1>;
};
```
Here is dmesg output after running insmod and rmmod repeatedly:
```
[   20.093466] udmabuf udmabuf@0: driver probe start.
[   20.098066] udmabuf udmabuf@0: assigned reserved memory node buffer@20000000
[   20.109583] udmabuf udmabuf0: major number   = 245
[   20.109597] udmabuf udmabuf0: minor number   = 0
[   20.109606] udmabuf udmabuf0: phys address   = 0x20000000
[   20.109614] udmabuf udmabuf0: buffer size    = 13107200
[   20.109621] udmabuf udmabuf0: dma coherent   = 0
[   20.109629] udmabuf udmabuf@0: driver installed.
[   75.176237] udmabuf udmabuf@0: driver remove start.
[   75.176505] udmabuf udmabuf@0: driver removed.
[   78.847090] udmabuf udmabuf@0: driver probe start.
[   78.847453] udmabuf udmabuf@0: assigned reserved memory node buffer@20000000
[   78.859316] udmabuf udmabuf0: major number   = 244
[   78.859329] udmabuf udmabuf0: minor number   = 0
[   78.859338] udmabuf udmabuf0: phys address   = 0x21000000
[   78.859346] udmabuf udmabuf0: buffer size    = 13107200
[   78.859353] udmabuf udmabuf0: dma coherent   = 0
[   78.859361] udmabuf udmabuf@0: driver installed.
[   82.806212] udmabuf udmabuf@0: driver remove start.
[   82.806493] udmabuf udmabuf@0: driver removed.
[   84.507019] udmabuf udmabuf@0: driver probe start.
[   84.507396] udmabuf udmabuf@0: assigned reserved memory node buffer@20000000
[   84.522188] udmabuf udmabuf0: major number   = 243
[   84.522200] udmabuf udmabuf0: minor number   = 0
[   84.522210] udmabuf udmabuf0: phys address   = 0x22000000
[   84.522218] udmabuf udmabuf0: buffer size    = 13107200
[   84.522225] udmabuf udmabuf0: dma coherent   = 0
[   84.522233] udmabuf udmabuf@0: driver installed.
[   86.336180] udmabuf udmabuf@0: driver remove start.
[   86.336450] udmabuf udmabuf@0: driver removed.
```
Notice how the physical memory address goes up with every load of the kernel module. This would continue until the assigned DMA pool is depleted.

I have tested this on a [4.9.0 kernel](https://github.com/Xilinx/linux-xlnx/tree/xilinx-v2017.4).

ikwzm, 2018-06-12:
Thank you for the issue.

I also checked memory leak on Linux 4.14.34.

I corrected the source code as follows and confirmed that dma_free_coherent () was called in udmabuf, and it was confirmed that it was called.

```C:udmabuf.c
static int udmabuf_driver_destroy(struct udmabuf_driver_data* this)
{
    if (!this)
        return -ENODEV;

    if (this->virt_addr != NULL) {
        dev_info(this->dma_dev, "dma_free_coherent() start");
        dma_free_coherent(this->dma_dev, this->alloc_size, this->virt_addr, this->phys_addr);
        dev_info(this->dma_dev, "dma_free_coherent() done");
        this->virt_addr = NULL;
    }
    cdev_del(&this->cdev);
    device_destroy(udmabuf_sys_class, this->device_number);
    ida_simple_remove(&udmabuf_device_ida, MINOR(this->device_number));
    kfree(this);
    return 0;
}

```

I prepared the following device-tree, secured 16MiB in reserved-memory and secured 15MiB in udmabuf.

```dts:test1.dts
        reserved-memory {
                #address-cells = <0x1>;
                #size-cells = <0x1>;
                ranges;

                zynq-dma-unusable-area {
                        reg = <0x0 0x80000>;
                };

                image_buf@0 {
                        compatible = "shared-dma-pool";
                        reusable;
                        reg = <0x20000000 0x01000000>;
                        phandle = <0x27>;
                };
        };

        udmabuf@0 {
                compatible = "ikwzm,udmabuf-0.10.a";
                device-name = "udmabuf0";
                minor-number = <0x0>;
                size = <0x00F00000>;
                memory-region = <0x27>;
	};
```

I followed insmod, rmmod and insmod as follows.
As a result, I found that memory can not be secured by the second insmod.

```
root@debian-fpga: /home/fpga/udmabuf# insmod udmabuf.ko
[   29.377529] udmabuf udmabuf@0: assigned reserved memory node image_buf@0
[   29.427365] udmabuf udmabuf0: major number   = 245
[   29.432174] udmabuf udmabuf0: minor number   = 0
[   29.436710] udmabuf udmabuf0: phys address   = 0x20000000
[   29.442150] udmabuf udmabuf0: buffer size    = 15728640
[   29.447299] udmabuf udmabuf0: dma coherent   = 0
[   29.451939] udmabuf udmabuf@0: driver installed.
```

```
root@debian-fpga: /home/fpga/udmabuf# rmmod udmabuf.ko
[   38.236690] udmabuf udmabuf@0: dma_free_coherent() start
[   38.242348] udmabuf udmabuf@0: dma_free_coherent() done
[   38.247856] udmabuf udmabuf@0: driver removed.
```

```
root@debian-fpga: /home/fpga/udmabuf# insmod udmabuf.ko
[   72.338724] udmabuf udmabuf@0: assigned reserved memory node image_buf@0
[   72.348059] cma: cma_alloc: alloc failed, req-size: 3840 pages, ret: -12
[   72.354753] dma_alloc_coherent() failed. return(0)
[   72.359469] udmabuf udmabuf@0: driver setup failed. return=-12
[   72.365793] udmabuf: probe of udmabuf@0 failed with error -12
```

Unfortunately, memory seems not to be released by dma_free_coherent().

Since dma_free_coherent() is the Linux Kernel API, it seems that something is going wrong in the Linux Kernel.

It may be a problem to reserve reserved-memory as CMA.

Unfortunately I did not find out any more.

Be careful when using the CMA area secured by reserved-memory from udmabuf until you know the cause.

ikwzm, 2018-06-13:
As a result of survey, we found that the position to call of_reserved_mem_release () is bad.
I released the modified version (v1.2.3).
I appreciate the provision of your issue.

closed 2018-06-13 by ikwzm