ikwzm/udmabuf #11: Is there any plans to upstream this driver?

vanjoe, 2018-06-27:
Have you tried to upstream this driver? Would the linux maintainers be amenable to having it in mainline?

ikwzm, 2018-06-28:
Thank you for the issue

At the moment there is no such plan.

udmabuf only works on arm / arm64 architecture. I do not know if it works on other architectures.

I hope that similar features will be implemented on the Linux mainline.

stuaxo, 2022-08-08:
@ikwzm any idea what needs to be done to get this working on x86 / amd64 ?

With Arm macs now a thing the ARM platform is a little less obscure than it was in 2018.

ikwzm, 2022-08-08:
Thank you for the issue.

I made u-dma-buf because it was needed by Zynq/ZynqMP/CycloneV SoC. 
I have these systems and am able to debug them.

However, I never used u-dma-buf on x86/amd64. 
Also, I'm not familiar with x86/amd64 cache/MMU.

This is why u-dma-buf is not guaranteed to work on architectures other than arm/arm64.

stuaxo, 2022-08-08:
Thanks for the info, that's good context that might be worth adding to the README.

I can see how getting a test for this on CI would be a pain too.

noelpedro, 2023-06-16:
@ikwzm, So the main line driver udmabuf is not compatible with zynqMP (arm-64bit)?

ikwzm, 2023-06-16:
The "u-dma-buf" provided by this repository and the "udmabuf" provided by linux main line are completely different.
Probably, "udmabuf" provided by linux main line works on ZynqMP(arm64).