ikwzm/udmabuf #12: mmap() offset parameter does nothing

tpwrules, 2018-09-19:
As the title states, providing a nonzero offset parameter to mmap() does not change the offset of the mapped region. This appears to be because of the following line:
https://github.com/ikwzm/udmabuf/blob/7861654030d8e1b424f514dab126ccff970ccf9a/udmabuf.c#L573

When I remove this line, the offset parameter works as expected. I don't understand why it was there, maybe to fix another bug?

ikwzm, 2018-09-20:
Thank you for the issue.

I do not understand much about vma-> vm_pgoff.
I am clearing vma-> vm_pgoff exactly as pointed out on issue # 2.

As you pointed out, it may be better not to clear vma-> vm_pgoff.

Please give me some more time.

tpwrules, 2018-09-20:
Ok, thank you for the input. I am working on a 32 bit ARM target (Altera Cyclone V). Maybe the behavior is different for AArch64, but I don't have one to test.

ikwzm, 2018-09-29:
Thank you for the issue.

Probably the problem was fixed with this commit.
https://github.com/ikwzm/udmabuf/commit/853fdd334d304e29b14ac7fb491e247ac6407635

closed 2018-09-29 by ikwzm