ikwzm/udmabuf #13: Usage on kernel 3.10 ?

sktpin, 2018-09-20:
I tried to compile this for kernel 3.10., and got some compile errors.
I looked at the list of supported kernels again, and discovered a gap in versions: 3.6 - 3.8, 3.18, ...
Is there a specific reason for omitting 3.10, i.e. will it definitely be a waste of time to try it with that kernel, or do you just not have a report of anyone succeeding with that kernel version and hence not listed it?
(*if* I get it to work, I will report back for another number to add to your list ;) May take a while though, I'm new to anything kernel related...)

ikwzm, 2018-09-20:
As you pointed out, the current udmabuf could not be compiled for Linux Kernel 3.18 or earlier.
It is because is_device_dma_coherent () is defined only in Linux Kernel 3.19 or later.

udmabuf v1.3.1-rc1 has been temporarily released. Could you please compile with this?

ikwzm, 2018-09-20:
Supplement

I currently do not use Linux Kernel 3.x.x. Also, there is no compilable environment. udmabuf v 0.8.0 or later, Linux Kernel 3.x.x neglected the operation confirmation. This is my mistake. Thank you for pointing out.

sktpin, 2018-09-20:
Thanks!
It appears there is more not present in that kernel, tough: ‘ida_simple_get’, ‘ida_simple_remove’, ‘ida_destroy’, ‘ida_init’.

It looks like I can't avoid getting into driver development myself ;)
(It seems I'm bound to that kernel version, even the last 2018 relese of NVIDIA's L4T Linux for TK1 has kernel 3.10, for whatever reason...)

Btw, maybe interesting, I found your project because it is linked on a Xilinx page about using DMA, and the other options they listed look a lot more involved.

sktpin, 2018-09-20:
I am not sure I understood what you said abut udmabuf v 0.8.0, but I tried to compile that - and it does compile! But with a warning: udmabuf.o(.init.text+0xbc): Section mismatch in reference from the function init_module().
Is that something that was fixed in a later release, which might also have a chance of compiling with kernel 3.10?

ikwzm, 2018-09-20:
Probably the problem was fixed with this commit.
https://github.com/ikwzm/udmabuf/commit/eace167540344a53312890bb6e888052014498d6

ikwzm, 2018-09-20:
I temporarily released udmabuf v1.3.1-rc2 which resolved `idr` related compile error. Could you please compile with this?

sktpin, 2018-09-21:
I had applied the changes you showed to v0.8.0 and it compiled without the warning, then had to go and forgot to post.
This v1.3.1-rc2 also compiles.
Thankyou!

ikwzm, 2018-09-21:
Congratulations. 
And thanks for this issue.
I will close this issue.

closed 2018-09-21 by ikwzm