ikwzm/udmabuf #15: dd command gets stuck

maikonadams, 2019-02-10:
Hi , when a use insmod command,
I get 
udmabuf udmabuf0: driver version = 1.3.2
udmabuf udmabuf0: major number   = 244
udmabuf udmabuf0: minor number   = 0
udmabuf udmabuf0: phys address   = 0x3e500000
udmabuf udmabuf0: buffer size    = 1048576
udmabuf udmabuf0: dma coherent   = 0
udmabuf udmabuf.0: driver installed.

and see the drivers at /dev ...
but then I try the dd command and I get stuck .
I am using 
Linux peta64v2 4.14.0-xilinx-v2018.2 #4 SMP PREEMPT Sat Feb 9 22:53:26 PST 2019 armv7l GNU/Linux

do I have to edit the device tree ?

ikwzm, 2019-02-10:
Thank you for the issue.

How do you use dd? And what will be the result?
Please give me some more detailed information.

maikonadams, 2019-02-12:
root@peta64v2:~# dd if=/dev/urandom of=/dev/udmabuf0 bs=4096 count=1024

and just get stuck .. 
dmesg :
udmabuf udmabuf0: driver version = 1.3.2
udmabuf udmabuf0: major number   = 244
udmabuf udmabuf0: minor number   = 0
udmabuf udmabuf0: phys address   = 0x3c500000
udmabuf udmabuf0: buffer size    = 1048576
udmabuf udmabuf0: dma coherent   = 0
udmabuf udmabuf.0: driver installed.

ikwzm, 2019-02-12:
bs=4096 and count=1024  => transfer size=4194304 byte

transfer size (=4194304 byte) > buffer size(=1048576 byte)

Since the buffer size is smaller than the transfer size, will it stack?

maikonadams, 2019-02-12:
oh thank you.
I guess you are saying it is too large ,
I changed to dd if=/dev/urandom of=/dev/udmabuf0 bs=1024 count=1024
1024+0 records in
1024+0 records out

I have to mention that I did not include any node in my device tree.

closed 2019-02-24 by ikwzm