ikwzm/udmabuf #16: Zynq Ultrascale Buffers are uncached, and slow to access

docjay2015, 2019-04-18:
On the Zynq Ultrascale, memory is mapped uncached for sync modes of 1, 2, 3.  So the access is slow.  For sync mode 0 (or 4) the data is cached, but flushes and invalidates do not work.  So modes 0 (or 4) are not useable.

docjay2015, 2019-04-18:
Benchmark  to copy a dma buffer in sync modes of 1, 2, or 3 is about 250 mb/s.  For mode 0, its 800 mb/s - all using optimzed assembly code to do the copies using load quad word pair instructions and matching stores.  Performance would be much worse if normal pointer addressing was used.

Is there any way to have the memory cached but still support an invalidate and flush set of commands that actually work.

ikwzm, 2019-04-19:
Thank you for the issue.

Udmabuf can explicitly instruct cache flushing and invalidation.

You can specify cache flush explicitly by ```sync_for_device``` as follows before PL(FPGA) starts PS->PL dma.

```C:
    unsigned char  attr[1024];
    unsigned long  sync_for_device = 1;
    if ((fd  = open("/sys/class/udmabuf/udmabuf0/sync_for_device", O_WRONLY)) != -1) {
        sprintf(attr, "%d", sync_for_device);
        write(fd, attr, strlen(attr));
        close(fd);
    }
```

After PL (FPGA) executes PL-> PS dma, you can specify cache invalidation explicitly by doing sync_for_cpu as follows.

```C:
    unsigned char  attr[1024];
    unsigned long  sync_for_cpu = 1;
    if ((fd  = open("/sys/class/udmabuf/udmabuf0/sync_for_cpu", O_WRONLY)) != -1) {
        sprintf(attr, "%d", sync_for_cpu);
        write(fd, attr, strlen(attr));
        close(fd);
    }
```

See the ```sync_offset```, ```sync_size```, ```sync_direction```, ```sync_for_cpu```, ```sync_for_device```, ```sync_owner```, ```dma_coherent``` section of Readme.md.

docjay2015, 2019-04-22:
Thanks so much for the reply.  In review my code was erroneous, I though I
was using those features be was not.  Correcting my internal code errors
seemed to fix the problem.  I did modify the source code to implement
coherence cache commands in a single write rather than writing to
sync_size, sync_offset and sync_direction simultaneously, via a new
attribute called cache_cmd.  I would guess that should be more efficient.
The modified code is attached for your review to see if it is worth
including in the reference.  Regards

/*********************************************************************************
 *
 *       Copyright (C) 2015-2018 Ichiro Kawazome
 *       All rights reserved.
 * 
 *       Redistribution and use in source and binary forms, with or without
 *       modification, are permitted provided that the following conditions
 *       are met:
 * 
 *         1. Redistributions of source code must retain the above copyright
 *            notice, this list of conditions and the following disclaimer.
 * 
 *         2. Redistributions in binary form must reproduce the above copyright
 *            notice, this list of conditions and the following disclaimer in
 *            the documentation and/or other materials provided with the
 *            distribution.
 * 
 *       THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS
 *       "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT
 *       LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR
 *       A PARTICULAR PURPOSE ARE DISCLAIMED.  IN NO EVENT SHALL THE COPYRIGHT
 *       OWNER OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL,
 *       SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT
 *       LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; LOSS OF USE,
 *       DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND ON ANY
 *       THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT 
 *       (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE
 *       OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
 * 
 ********************************************************************************/
#include <linux/cdev.h>
#include <linux/clk.h>
#include <linux/dma-mapping.h>
#include <linux/fs.h>
#include <linux/idr.h>
#include <linux/init.h>
#include <linux/interrupt.h>
#include <linux/io.h>
#include <linux/ioport.h>
#include <linux/kernel.h>
#include <linux/module.h>
#include <linux/mutex.h>
#include <linux/of.h>
#include <linux/of_device.h>
#include <linux/sched.h>
#include <linux/device.h>
#include <linux/platform_device.h>
#include <linux/slab.h>
#include <linux/string.h>
#include <linux/sysctl.h>
#include <linux/types.h>
#include <linux/uaccess.h>
#include <linux/scatterlist.h>
#include <linux/pagemap.h>
#include <linux/list.h>
#include <linux/spinlock.h>
#include <linux/version.h>
#include <asm/page.h>
#include <asm/byteorder.h>

/**
[... 1410 lines cut ...]
    if (retval) {
        printk(KERN_ERR "%s: couldn't register platform driver. return=%d\n", DRIVER_NAME, retval);
        udmabuf_platform_driver_registerd = 0;
        goto failed;
    } else {
        udmabuf_platform_driver_registerd = 1;
    }

    return 0;

 failed:
    udmabuf_module_cleanup();
    return retval;
}

/**
 * udmabuf_module_exit()
 */
static void __exit udmabuf_module_exit(void)
{
    udmabuf_module_cleanup();
}

module_init(udmabuf_module_init);
module_exit(udmabuf_module_exit);

ikwzm, 2019-04-23:
Thank you for the interesting suggestions.

I think that your idea of ​​setting ```sync_offset```/```sync_size```/```sync_direction``` and 'sync' single writing is good.

Inspired by your idea, I am thinking of extending the ```sync_for_device```/```sync_for_cpu``` variable to change the ```sync_offset```/```sync_size```/```sync_direction``` simultaneously.

```C:udmabuf

#define  SYNC_COMMAND_DIR_MASK        (0x000000000000000C)
#define  SYNC_COMMAND_DIR_SHIFT       (2)
#define  SYNC_COMMAND_SIZE_MASK       (0x00000000FFFFFFF0)
#define  SYNC_COMMAND_SIZE_SHIFT      (0)
#define  SYNC_COMMAND_OFFSET_MASK     (0xFFFFFFFF00000000)
#define  SYNC_COMMAND_OFFSET_SHIFT    (32)
#define  SYNC_COMMAND_ARGMENT_MASK    (0xFFFFFFFFFFFFFFFE)
/**
 * udmabuf_sync_command_argment() - get argment for dma_sync_single_for_cpu() or dma_sync_single_for_device()
 *                                  
 * @this:       Pointer to the udmabuf device data structure.
 * @command     sync command (this->sync_for_cpu or this->sync_for_device)
 * @phys_addr   Pointer to the phys_addr for dma_sync_single_for_...()
 * @size        Pointer to the size for dma_sync_single_for_...()
 * @direction   Pointer to the direction for dma_sync_single_for_...()
 * Return:      Success(=0) or error status(<0).
 */
static int udmabuf_sync_command_argment(
    struct udmabuf_device_data*  this     ,
    u64                          command  ,
    dma_addr_t                *  phys_addr,
    size_t                    *  size     ,
    enum dma_data_direction   *  direction
) {
    u64    sync_offset   ;
    size_t sync_size     ;
    int    sync_direction;
    if ((command & SYNC_COMMAND_ARGMENT_MASK) != 0) {
        sync_offset    = (u64   )((command & SYNC_COMMAND_OFFSET_MASK) >> SYNC_COMMAND_OFFSET_SHIFT);
        sync_size      = (size_t)((command & SYNC_COMMAND_SIZE_MASK  ) >> SYNC_COMMAND_SIZE_SHIFT  );
        sync_direction = (int   )((command & SYNC_COMMAND_DIR_MASK   ) >> SYNC_COMMAND_DIR_SHIFT   );
    } else {
        sync_offset    = this->sync_offset;
        sync_size      = this->sync_size;
        sync_direction = this->sync_direction;
    }
    if (sync_offset + sync_size >= this->size)
        return -EINVAL;
    switch(sync_direction) {
        case 1 : *direction = DMA_TO_DEVICE    ; break;
        case 2 : *direction = DMA_FROM_DEVICE  ; break;
        default: *direction = DMA_BIDIRECTIONAL; break;
    }
    *phys_addr = this->phys_addr + sync_offset;
    *size      = sync_size;
    return 0;
} 

/**
 * udmabuf_sync_for_cpu() - call dma_sync_single_for_cpu() when (sync_for_cpu != 0)
 * @this:       Pointer to the udmabuf device data structure.
 * Return:	Success(0)
 */
static int udmabuf_sync_for_cpu(struct udmabuf_device_data* this)
{
    int status = 0;

    if (this->sync_for_cpu) {
        dma_addr_t              phys_addr;
        size_t                  size;
        enum dma_data_direction direction;
        status = udmabuf_sync_command_argment(this, this->sync_for_cpu, &phys_addr, &size, &direction);
        if (status == 0) {
            dma_sync_single_for_cpu(this->dma_dev, phys_addr, size, direction);
            this->sync_for_cpu = 0;
            this->sync_owner   = 0;
        }
    }
    return status;
}

/**
 * udmabuf_sync_for_device() - call dma_sync_single_for_device() when (sync_for_device != 0)
 * @this:       Pointer to the udmabuf device data structure.
 * Return:	Success(0)
 */
static int udmabuf_sync_for_device(struct udmabuf_device_data* this)
{
    int status = 0;

    if (this->sync_for_device) {
        dma_addr_t              phys_addr;
        size_t                  size;
        enum dma_data_direction direction;
        status = udmabuf_sync_command_argment(this, this->sync_for_device, &phys_addr, &size, &direction);
        if (status == 0) {
            dma_sync_single_for_device(this->dma_dev, phys_addr, size, direction);
            this->sync_for_device = 0;
            this->sync_owner      = 1;
        }
    }
    return status;
}

```

Is this change helpful for you?

docjay2015, 2019-04-23:
Yes, I could use that instead, and would rather use a common code base
rather than my customized version.  I assume you would also change the set
method to get a u64 rather than a u32 from the string source (similar to
what I sent you).  Providing all 3 values in on system call likely reduces
overhead BUT ALSO MAKES IT POSSIBLE to use the same buffer (chopped up into
different areas perhaps) in a thread safe way because the call provides all
parameters in one call and mutex locked in the set logic.  Your changes
also are backward compatible which I think is also a plus, and does not
introduce an extra attribute.

So if you wish to make those changes I would like forward to picking up a
revised version.

I don't understand to kernel concepts of direction.  It seems to be that
the firmware should be fetching from DRAM, not from the cache??? - is that
true?  If so then what I am concerned about is invalidating the cache
before reading anything the firmware DMAs in, and flushing (assuming the
cache is not write through) after the PS changes dma buffers before
allowing the firmware to use those buffers.  What does it mean to specify
the direction is CPU to DEVICE when the owner of a buffer is set to the
CPU?  Similarly what does it mean to specify the direction is DEVICE to CPU
when the owner of the buffer is set to the DEVICE?  I also dont understand
what bidirectional does.  My ignorance I suppose.  The bare metal libraries
provide flush and invalidate which I think I understand.  Feel free to
comment on this if you wish.

I did test the invalidate logic with firmware DMA into a cached buffer.
Without the invalidate there were some artifacts.  With the invalidate the
test pattern error checking passed (using your driver).  Also as a side
note, one needs to be careful when doing this type of error checking when
using the debugger.  When stepping through code the cache is invalidated
often and the benefit of calls to invalidate may not be detected.

Regards

ikwzm, 2019-04-23:
Information on DMA Direction is detailed in DMA-API-HOWTO.txt in the Linux Kernel documentation. DMA-API-HOWTO.txt is in the Document/ directory of the Linux Kernel source code.

The part is extracted below.

```
DMA Direction
=============

The interfaces described in subsequent portions of this document
take a DMA direction argument, which is an integer and takes on
one of the following values::

 DMA_BIDIRECTIONAL
 DMA_TO_DEVICE
 DMA_FROM_DEVICE
 DMA_NONE

You should provide the exact DMA direction if you know it.

DMA_TO_DEVICE means "from main memory to the device"
DMA_FROM_DEVICE means "from the device to main memory"
It is the direction in which the data moves during the DMA
transfer.

You are _strongly_ encouraged to specify this as precisely
as you possibly can.

If you absolutely cannot know the direction of the DMA transfer,
specify DMA_BIDIRECTIONAL.  It means that the DMA can go in
either direction.  The platform guarantees that you may legally
specify this, and that it will work, but this may be at the
cost of performance for example.

The value DMA_NONE is to be used for debugging.  One can
hold this in a data structure before you come to know the
precise direction, and this will help catch cases where your
direction tracking logic has failed to set things up properly.

Another advantage of specifying this value precisely (outside of
potential platform-specific optimizations of such) is for debugging.
Some platforms actually have a write permission boolean which DMA
mappings can be marked with, much like page protections in the user
program address space.  Such platforms can and do report errors in the
kernel logs when the DMA controller hardware detects violation of the
permission setting.

Only streaming mappings specify a direction, consistent mappings
implicitly have a direction attribute setting of
DMA_BIDIRECTIONAL.

The SCSI subsystem tells you the direction to use in the
'sc_data_direction' member of the SCSI command your driver is
working on.

For Networking drivers, it's a rather simple affair.  For transmit
packets, map/unmap them with the DMA_TO_DEVICE direction
specifier.  For receive packets, just the opposite, map/unmap them
with the DMA_FROM_DEVICE direction specifier.

Using Streaming DMA mappings
============================

The streaming DMA mapping routines can be called from interrupt
context.  There are two versions of each map/unmap, one which will
map/unmap a single memory region, and one which will map/unmap a
scatterlist.

To map a single region, you do::

	struct device *dev = &my_dev->dev;
[... 196 lines cut ...]
				 * so dma_sync_single_for_device() is
				 * not needed here. It would be required
				 * for DMA_BIDIRECTIONAL mapping if
				 * the memory was modified.
				 */
				give_rx_buf_to_card(cp);
			}
		}
	}

Drivers converted fully to this interface should not use virt_to_bus() any
longer, nor should they use bus_to_virt(). Some drivers have to be changed a
little bit, because there is no longer an equivalent to bus_to_virt() in the
dynamic DMA mapping scheme - you have to always store the DMA addresses
returned by the dma_alloc_coherent(), dma_pool_alloc(), and dma_map_single()
calls (dma_map_sg() stores them in the scatterlist itself if the platform
supports dynamic DMA mapping in hardware) in your driver structures and/or
in the card registers.

All drivers should be using these interfaces with no exceptions.  It
is planned to completely remove virt_to_bus() and bus_to_virt() as
they are entirely deprecated.  Some ports already do not provide these
as it is impossible to correctly support them.

```

docjay2015, 2019-04-23:
Invalidate and flush do work with the driver.

closed 2019-04-23 by docjay2015