ikwzm/udmabuf #18: Zynq Ultrascale+ (ARM Cortex A-53) Kernel 4.19 problem

oguzhancik, 2019-05-31:
There is a compilation problem while it is being used with Petalinux 2019.1. This version includes kernel version 4.19. I want to use it with Zynq Ultrascale+ (ARM A-53) and Linux 4.19 kernel.

Here is the compilation error.

```
| /home/oguzhancik/petalinux-projects/zcu102-works/build/tmp/work/plnx_zynqmp-xilinx-linux/udmabuf/1.0-r0/udmabuf.c: In function 'udmabuf_platform_driver_probe':
| /home/oguzhancik/petalinux-projects/zcu102-works/build/tmp/work/plnx_zynqmp-xilinx-linux/udmabuf/1.0-r0/udmabuf.c:1246:18: error: too few arguments to function 'of_dma_configure'

|          retval = of_dma_configure(&pdev->dev, pdev->dev.of_node);
|                   ^~~~~~~~~~~~~~~~
| In file included from /home/oguzhancik/petalinux-projects/zcu102-works/build/tmp/work/plnx_zynqmp-xilinx-linux/udmabuf/1.0-r0/udmabuf.c:44:
| /home/oguzhancik/petalinux-projects/zcu102-works/build/tmp/work-shared/plnx-zynqmp/kernel-source/include/linux/of_device.h:58:5: note: declared here
|  int of_dma_configure(struct device *dev,
|      ^~~~~~~~~~~~~~~~
| make[3]: *** [/home/oguzhancik/petalinux-projects/zcu102-works/build/tmp/work-shared/plnx-zynqmp/kernel-source/scripts/Makefile.build:312: /home/oguzhancik/petalinux-projects/zcu102-works/build/tmp/work/plnx_zynqmp-xilinx-linux/udmabuf/1.0-r0/udmabuf.o] Error 1
| ERROR: oe_runmake failed

```

I need to used it for my DMA design from userspace and buffer management is based on udmabuf kernel module.

oguzhancik, 2019-05-31:
The problem is caused by here

Kernel function prototype looks like this

`static inline int of_dma_configure(struct device *dev, struct device_node *np, bool force_dma)`

I added false statements to compile it properly but i'm not sure the effect.

```
#if (USE_OF_DMA_CONFIG == 1)
    /*
     * of_dma_configure()
     * - set pdev->dev->dma_mask
     * - set pdev->dev->coherent_dma_mask
     * - call of_dma_is_coherent()
     * - call arch_setup_dma_ops()
     */
#if (USE_OF_RESERVED_MEM == 1)
    /* If "memory-region" property is spsecified, of_dma_configure() will not be executed.
     * Because in that case, it is already executed in of_reserved_mem_device_init().
     */
    if (device_data->of_reserved_mem == 0)
#endif
    {
#if (LINUX_VERSION_CODE >= 0x040C00)
        retval = of_dma_configure(&pdev->dev, pdev->dev.of_node, false);
        if (retval != 0) {
            dev_err(&pdev->dev, "of_dma_configure failed. return=%d\n", retval);
            goto failed;
        }
#else
        of_dma_configure(&pdev->dev, pdev->dev.of_node, false);
#endif
    }
#endif
```

ikwzm, 2019-05-31:
Thank you for the issue.

I am in a place I can not try now. Please wait a couple of days.

This is a guess, but force_dma = true may be better.

oguzhancik, 2019-05-31:
After the changes i made with "false" statement, it looks working.

```
[    6.932847] udmabuf dma_tx_data_buffer: driver version = 1.3.2
[    6.938724] udmabuf dma_tx_data_buffer: major number   = 241
[    6.944389] udmabuf dma_tx_data_buffer: minor number   = 0
[    6.949875] udmabuf dma_tx_data_buffer: phys address   = 0x000000006fd00000
[    6.956839] udmabuf dma_tx_data_buffer: buffer size    = 2097152
[    6.962852] udmabuf dma_tx_data_buffer: dma coherent   = 1
[    6.968341] udmabuf dma_tx_data_buffer@0: driver installed. 
```

ikwzm, 2019-06-02:
I have released v1.4.1-rc1 to fix this problem. try it.

ikwzm, 2019-06-05:
It has been confirmed that it works properly on Linux Kernel 4.19.
If you still have problems, please report.

closed 2019-06-05 by ikwzm

oguzhancik, 2019-06-05:
i tested it and there is no problem now. I'll report in case of a problem. Thanks for the support.

honorpeter, 2019-06-13:
> After the changes i made with "false" statement, it looks working.
>
> ```
> [    6.932847] udmabuf dma_tx_data_buffer: driver version = 1.3.2
> [    6.938724] udmabuf dma_tx_data_buffer: major number   = [...]
I also  use it with  zcu102.but can't  succeed.
How to do in your PL hw bd design?  and how to add these file to petalinux? only with the udmabuf.c?

oguzhancik, 2019-06-14:
You can create kernel module according to UG1144 and change the udmabuf.c content with the udmabuf.c from @ikwzm. You should use the latest version 1.4.1 to build it properly for Petalinux 2019.1 because Petalinux 2019.1 comes with Linux 4.19 kernel.

You need to do some modifications in system-user.dtsi file in petalinux. The modifications are explained really good in README.md.

I'm also writing C++ library to use and manage DMA memory buffers easily from Linux userspace. It works well with AXI DMA without cache flush or invalidate on ZCU 102. I will post it soon.