ikwzm/udmabuf #19: make error for zynq debian9 rootfs

kimanha, 2019-06-05:
Hello,

I am trying to make your udmabuf module, but I found errors.
I did not find 4.19.0-xilinx-v2019.1/build directory form /lib/modules.
How do I install or copy the 4.19.0-xilinx-v2019.1/build directory at debian9 rootfs ?

My test environment is petalinux 2019.1 and Debian9 rootfs.

root@zdfe-bxb:/home/kha/udmabuf-kmod-dpkg/udmabuf# make
make -C /lib/modules/4.19.0-xilinx-v2019.1/build ARCH=arm CROSS_COMPILE= M=/home/kha/udmabuf-kmod-dpkg/udmabuf modules
make[1]: *** /lib/modules/4.19.0-xilinx-v2019.1/build: No such file or directory.  Stop.
Makefile:19: recipe for target 'all' failed
make: *** [all] Error 2

Thanks,
Kiman

ikwzm, 2019-06-05:
Thank you for the issue.

If you are using Debian, you can use https://github.com/ikwzm/udmabuf-kmod-dpkg.git.

Make a Debian package referring to the readme.md of udmabuf-kmod-dpkg.
Start Zynq and install the Debian package using the dpkg -i command.

kimanha, 2019-06-06:
Hello,

I want to compile with Debian9 for my test.
I copied build directory from the petalinux project, but found different errors.

build directory from my petalinux project:
~/xilinx/kiman/bxb/zdfe_bxb.sdk/bxb_petalinux/build $ ls
bitbake-cookerdaemon.log  build.log  cache  conf  config.log  config.log.old  downloads  misc  sstate-cache  tmp

Can you advice a little more details about the linux kernel header file and where to copy for compile?

root@zdfe-bxb:/home/kha/test/udmabuf-kmod-dpkg/udmabuf# make
make -C /lib/modules/4.19.0-xilinx-v2019.1/build ARCH=arm CROSS_COMPILE= M=/home/kha/test/udmabuf-kmod-dpkg/udmabuf modules
make[1]: Entering directory '/lib/modules/4.19.0-xilinx-v2019.1/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/lib/modules/4.19.0-xilinx-v2019.1/build'
Makefile:19: recipe for target 'all' failed
make: *** [all] Error 2

Thanks,
Kiman

ikwzm, 2019-06-06:
I'm sorry. I do not understand petalinux well.

but...

In order to self-compile the kernel module on Zynq + Debian, you need the include files and scripts used when building Linux. And those files are placed in /usr/src/linux-headers-XXX/ and symlinked from /lib/modules/XXX/build. When building a Linux Kernel with petalinux, you may need to find the complete source code from your project and copy it to /lib/modules/XXX/build.

If you do not need to self-compile on Zynq + Debian, consider cross-compiling on your PC. It shows how to cross-compile to udmabuf-kmod-dpkg. 
It is easy to copy cross-compiled udmabuf-XXX-YYY.deb to Zynq + Debian and install with dpkg -i.

kimanha, 2019-06-06:
Hello kwzm,

We also have Debian7 rootfs for the old version zynq board.
This version included /lib/modules/3.17.0-xilinx source code and works a good your udmabuf driver module.

Thanks for your comment and I will find out the latest petalinux kernel headers source code for the Debian9 rootfs.

I still prefer self-compile for the development stage.

here is the compile message from Debian7.

root@rfbpm-zynq:/home/kha/udmabuf-kmod-dpkg# make
cd ./udmabuf ; make ARCH=arm KERNEL_SRC_DIR=/lib/modules/3.17.0-xilinx/build all   ; cd /home/kha/udmabuf-kmod-dpkg
make[1]: Entering directory `/home/kha/udmabuf-kmod-dpkg/udmabuf'
make -C /lib/modules/3.17.0-xilinx/build ARCH=arm CROSS_COMPILE= M=/home/kha/udmabuf-kmod-dpkg/udmabuf modules
make[2]: Entering directory `/opt/devel/kernel/linux-xlnx'
  CC [M]  /home/kha/udmabuf-kmod-dpkg/udmabuf/udmabuf.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/kha/udmabuf-kmod-dpkg/udmabuf/udmabuf.mod.o
  LD [M]  /home/kha/udmabuf-kmod-dpkg/udmabuf/udmabuf.ko
make[2]: Leaving directory `/opt/devel/kernel/linux-xlnx'
make[1]: Leaving directory `/home/kha/udmabuf-kmod-dpkg/udmabuf'

closed 2019-06-20 by kimanha