ikwzm/udmabuf #20: Zcu102 Kernel 4.14 petalinux problem by the udmabuf-master

honorpeter, 2019-06-13:
There is a compilation problem while it is being used with Petalinux 20118.2. This version includes kernel version 4.14. I want to use it with Zynq Ultrascale+ (ARM A-53) and Linux 4.14 kernel.

I copy the udmabuf.c to the   /project-spec/meta-user/recipes-modules/udmabuf/files

and  add the device tree in       /project-spec/meta-user/recipes-bsp/device-tree/files/system-user.dtsi like this
/{
reserved-memory {
		#address-cells = <1>;
		#size-cells = <1>;
		ranges;
		image_buf0: image_buf@0 {
			compatible = "shared-dma-pool";
			reusable;
			reg = <0x3C000000 0x04000000>; 
			label = "image_buf0";
		};
	};
	udmabuf@0 {
		compatible = "ikwzm,udmabuf-0.10.a";
		device-name = "udmabuf0";
		size = <0x04000000>; // 64MiB
		memory-region = <&image_buf0>;
	};
}

and  compile success. but when I boot on zcu102 there was some error.

Here is the boot error.

Starting udev
[    5.476400] [drm] Cannot find any crtc or sizes
[    5.501104] udevd[1875]: starting version 3.2.2
[    5.542714] udevd[1876]: starting eudev-3.2.2
[    5.674402] udmabuf: loading out-of-tree module taints kernel.
[    5.683052] udmabuf udmabuf@0: of_reserved_mem_device_init failed. return=-22
[    5.700321] udmabuf: probe of udmabuf@0 failed with error -22

I may need some support for this. Thank you.

ikwzm, 2019-06-13:
Thank you for the issue.

Please check only one. Did you see the following line in the boot log?

```
Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.0-xlnx-v2018.2-zynqmp-fpga (ichiro@sphinx-vm-ubuntu) (gcc version 5.4.0 20160609 (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.9)) #1 SMP Fri Aug 10 11:05:02 JST 2018
[    0.000000] Boot CPU: AArch64 Processor [410fd034]
[    0.000000] Machine model: ZynqMP Ultra96
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] OF: fdt: Reserved memory: unsupported node format, ignoring
```

Notice the last line. If such a message appears, it has failed to secure the memory area of ​​reserved-memory. At that time, udmabuf also fails.

For arm64, the reserved-memory address and size must be 2 words each. Try using the reserved-memory node as follows:

```
        reserved-memory {
                #address-cells = <2>;
                #size-cells = <2>;
                ranges;
                image_buf0: image_buf@0 {
                        compatible = "shared-dma-pool";
                        reusable;
                        reg = <0x0 0x3c000000 0x0 0x04000000>;
                        label = "image_buf0";
                };
        };
```

The following message will be output if space reservation for reserved-memory is successful.

```
Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.0-xlnx-v2018.2-zynqmp-fpga (ichiro@sphinx-vm-ubuntu) (gcc version 5.4.0 20160609 (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.9)) #1 SMP Fri Aug 10 11:05:02 JST 2018
[    0.000000] Boot CPU: AArch64 Processor [410fd034]
[    0.000000] Machine model: ZynqMP Ultra96
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000003c000000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node image_buf@0, compatible id shared-dma-pool
```

I think udmabuf will also succeed.

```
[    6.311885] udmabuf: loading out-of-tree module taints kernel.
[    6.319927] udmabuf udmabuf@0: assigned reserved memory node image_buf@0
[    6.365631] udmabuf udmabuf0: driver version = 1.3.2
[    6.370539] udmabuf udmabuf0: major number   = 243
[    6.375311] udmabuf udmabuf0: minor number   = 0
[    6.379915] udmabuf udmabuf0: phys address   = 0x000000003c000000
[    6.385981] udmabuf udmabuf0: buffer size    = 67108864
[    6.391201] udmabuf udmabuf0: dma coherent   = 0
[    6.395791] udmabuf udmabuf@0: driver installed.
```

honorpeter, 2019-06-13:
> Thank you for the issue.
>
> Please check only one. Did you see the following line in the boot log?
>
> ```
> Starting kernel ...
>
> [    0.000000] Booting Linux on physical CPU 0x0
> [    0.000000] Linux [...]

I will try this  tomorrow.Thank you for you kind help.

honorpeter, 2019-06-14:
> Thank you for the issue.
>
> Please check only one. Did you see the following line in the boot log?
>
> ```
> Starting kernel ...
>
> [    0.000000] Booting Linux on physical CPU 0x0
> [    0.000000] Linux [...]

I try as your suggestion and succeed on boot . but when I rmmod it .and insmod again. There was another error as follows:
 cd /lib/modules/4.14.0-xilinx-v2018.2/extra
# insmod udmabuf.ko udmabf
[  673.099906] udmabuf udmabuf@0: assigned reserved memory node image_buf@0     
[  673.121555] udmabuf udmabuf0: driver version = 1.4.1                         
[  673.126457] udmabuf udmabuf0: major number   = 241                           
[  673.131229] udmabuf udmabuf0: minor number   = 0                             
[  673.135822] udmabuf udmabuf0: phys address   = 0x000000003c000000            
[  673.141900] udmabuf udmabuf0: buffer size    = 67108864                      
[  673.147106] udmabuf udmabuf0: dma coherent   = 0                             
[  673.151708] udmabuf udmabuf@0: driver installed.                             
[  673.156429] couldn't allocate minor number(=0).                              
[  673.160881] udmabuf udmabuf.0: driver create failed. return=0.               
[  673.166699] udmabuf: probe of udmabuf.0 failed with error -22

honorpeter, 2019-06-14:
@ikwzm  Thank you for you reply.
In addition. when boot:

[    0.000000] Booting Linux on physical CPU 0x0                                
[    0.000000] Linux version 4.14.0-xilinx-v2018.2 (oe-user@oe-host) (gcc versi9
[    0.000000] Boot CPU: AArch64 Processor [410fd034]                           
[    0.000000] Machine model: ZynqMP ZCU102 Rev1.0                              
[    0.000000] earlycon: cdns0 at MMIO 0x00000000ff000000 (options '115200n8')  
[    0.000000] bootconsole [cdns0] enabled                                      
[    0.000000] efi: Getting EFI parameters from FDT:                            
[    0.000000] efi: UEFI not found.                                             
[    0.000000] Reserved memory: created CMA memory pool at 0x000000003c000000, B
[    0.000000] OF: reserved mem: initialized node image_buf@0, compatible id shl
[    0.000000] cma: Reserved 256 MiB at 0x000000006fc00000   
;
INIT: version 2.88 booting                                                      
Starting udev                                                                   
[    5.488351] [drm] Cannot find any crtc or sizes                              
[    5.518555] udevd[1875]: starting version 3.2.2                              
[    5.560471] udevd[1876]: starting eudev-3.2.2                                
[    5.675792] udmabuf: loading out-of-tree module taints kernel.               
[    5.684855] udmabuf udmabuf@0: assigned reserved memory node image_buf@0     
[    5.710357] udmabuf udmabuf0: driver version = 1.4.1                         
[    5.715282] udmabuf udmabuf0: major number   = 243                           
[    5.720059] udmabuf udmabuf0: minor number   = 0                             
[    5.724646] udmabuf udmabuf0: phys address   = 0x000000003c000000            
[    5.730709] udmabuf udmabuf0: buffer size    = 67108864                      
[    5.735933] udmabuf udmabuf0: dma coherent   = 0                             
[    5.740518] udmabuf udmabuf@0: driver installed.  

root@xdmatest:~# rmmod udmabuf    
[  338.843912] udmabuf udmabuf@0: driver removed.
root@xdmatest:~# insmod /lib/modules/4.14.0-xilinx-v2018.2/extra/udmabuf.ko
[  401.571302] udmabuf udmabuf@0: assigned reserved memory node image_buf@0
[  401.593201] udmabuf udmabuf0: driver version = 1.4.1
[  401.598099] udmabuf udmabuf0: major number   = 242
[  401.602870] udmabuf udmabuf0: minor number   = 0
[  401.607465] udmabuf udmabuf0: phys address   = 0x000000003c000000
[  401.613546] udmabuf udmabuf0: buffer size    = 67108864
[  401.618750] udmabuf udmabuf0: dma coherent   = 0
[  401.623354] udmabuf udmabuf@0: driver installed.

root@xdmatest:~# rmmod udmabuf
[  427.096807] udmabuf udmabuf@0: driver removed.

# insmod /lib/modules/4.14.0-xilinx-v2018.2/extra/udmabuf.ko udmabuf0=67108864
[  458.807091] udmabuf udmabuf@0: assigned reserved memory node image_buf@0
[  458.828729] udmabuf udmabuf0: driver version = 1.4.1
[  458.833623] udmabuf udmabuf0: major number   = 241
[  458.838389] udmabuf udmabuf0: minor number   = 0
[  458.842988] udmabuf udmabuf0: phys address   = 0x000000003c000000
[  458.849071] udmabuf udmabuf0: buffer size    = 67108864                               
[  458.854275] udmabuf udmabuf0: dma coherent   = 0                                      
[  458.858878] udmabuf udmabuf@0: driver installed.                                      
[  458.863594] couldn't allocate minor number(=0).                                       
[  458.868048] udmabuf udmabuf.0: driver create failed. return=0.                        
[  458.873866] udmabuf: probe of udmabuf.0 failed with error -22  

why? if  the udmabuf0=67108864  was wrong?

ikwzm, 2019-06-14:
The device-tree node remains in the kernel after removing the device driver. And when you install the device driver again, the device-tree node is probed again.

In your example, the udmabuf@0 node of device-tree that was left in the kernel when you installed udmabuf.ko with insmod will be probed again to generate udmabuf0.
At the same time, trying to generate udmabuf0=67108864 specified by the argument of insmod, it fails in conflict with udmabuf0 whose name and minor number already exist.

Change the device driver name and minor number to avoid conflicts.

honorpeter, 2019-06-14:
@

> The device-tree node remains in the kernel after removing the device driver. And when you install the device driver again, the device-tree node is probed again.
>
> In your example, the udmabuf@0 node [...]

Thank you ,I  see .

closed 2019-06-14 by honorpeter

Scriabing, 2020-05-30:
@honorpeter 可以问您nvdla在fpga相关的问题吗？我的邮箱是z1403154226@163.com，希望能和您交流下