ikwzm/udmabuf #22: Suggest use pfn instead of page in mmap page fault handler

litingqianintel, 2019-06-27:
when dma_alloc_xxx API return pages from buddy rather than CMA(for example, CMA reserve is exhausted), only first page is refcounted, the pages after it will have refcount zero, this is not problem usually, but when you map them one by one in page fault handler, the munmap() call will decrease refcount for each page and free some of them if it's not refcounted, the first page is OK, all pages after it will be free after munmap, this is a bug.

I suggest to use pfn mapping (using vmf_insert_pfn()) in that case. and set VM_PFNMAP flag in vma->vm_flags, so munmap will not try to manipulate page struct.

ikwzm, 2019-06-27:
Thank you for the issue.

I respect your suggestion. 
Please wait a little more for this.
I need to learn more.

ikwzm, 2019-07-05:
According to your suggest, I changed it to use vm_insert_pfn () in mmap page fault handler.

I released it temporarily with v1.4.2-rc1.
I will confirm the operation as it is for a while.

Thank you for your suggestion.

usstq, 2019-07-08:
No problem, I'm glad to contribute to this wonderful kernel module!

ikwzm, 2019-07-18:
I have released udmabuf v1.4.2 which adopted this proposal.
Thank you.

closed 2019-07-18 by ikwzm