ikwzm/udmabuf #24: How to make a DMA transfer?

itzikhai, 2019-07-18:
What is the purpose of the driver ?
How it can be used to really transfer the buffer using a DMA ? Isn't it an architecture dependent ?
Why it is depended only at ARM ?
For example, using PPC chip with a PCIe device.

ikwzm, 2019-07-18:
Thank you for the issue.

The purpose of this driver is to provide a DMA buffer that can be accessed by the CPU.
udmabuf does not control DMA. It is the job of the user program.

The architecture is limited to ARM / ARM64 because I have only verified operation on those architectures. I have not checked if it works on other architectures.

itzikhai, 2019-07-18:
Hello ikwzm.

**Thanks for your replies.**

You mean that control DMA is the part of driver - right?
Is is possible to control DMA at userspace program?

Regarding the arch. - I may be able modify, compile and build your device driver at PPC64 based on NXP B4860.

ikwzm, 2019-07-18:
You can use uio and udmabuf to do DMA transfers without writing a kernel driver.

For example, in the following example, DMA transfer is performed with a program written in Python.

https://github.com/ikwzm/ZynqMP-FPGA-Linux-Example-2-Ultra96

noelpedro, 2020-02-11:
Hi guys I'm trying to use udmabuf on a simple loopback design here is my device tree that I plan to use:

Do I need to define memery region for RX and TX dma channels?
`/ {
    /* Loopback DMA setup */
    
    loopback_dma: axidma@40400000 {
        compatible = "generic-uio";
        #dma-cells = <1>;
        reg = < 0x40400000 0x10000 >;
        clocks = <&clkc 15>, <&clkc 15>, <&clkc 15>, <&clkc 15>; // fclk0 from clock controller
        clock-names = "s_axi_lite_aclk", "m_axi_mm2s_aclk", "m_axi_s2mm_aclk", "m_axi_sg_aclk";
        xlnx,include-sg;

        loopback_dma_mm2s_chan: dma-channel@40400000 {
            compatible = "xlnx,axi-dma-mm2s-channel";
            interrupt-parent = <&intc>;
            interrupts = <0 31 4>; 
            xlnx,datawidth = <0x20>;        
            xlnx,sg-length-width = <14>;    

            xlnx,device-id = <0x1>;     
        };

        loopback_dma_s2mm_chan: dma-channel@40400030 {
            compatible = "xlnx,axi-dma-s2mm-channel";
            interrupt-parent = <&intc>;
            interrupts = <0 32 4>; 
            xlnx,datawidth = <0x20>;       
            xlnx,sg-length-width = <14>;    
            xlnx,device-id = <0x1>;     
        };
    };

udmabuf@0x00 {
			compatible = "ikwzm,udmabuf-0.10.a";
			device-name = "udmabuf0";
			minor-number = <0>;
			size = <0x00100000>;
                        sync-direction = <1>; //TX
		};
udmabuf@0x00 {
			compatible = "ikwzm,udmabuf-0.10.a";
			device-name = "udmabuf1";
			minor-number = <0>;
			size = <0x00100000>;
                        sync-direction = <2>; //RX
		};

};  '