ikwzm/udmabuf #30: udmabuf udmabuf@1: driver setup failed

gi4you, 2019-12-11:
Hello, 
I have udmabuf.ko loading error with my zcu102 board. Can you please see error message?

 system-user.dtsi device tree configuration.

        udmabuf@0 {
                compatible = "ikwzm,udmabuf-0.10.a";
                device-name = "udmabuf0";
                minor-number = <0>;
                size = <0x8000000>;
        };
        udmabuf@1 {
                compatible = "ikwzm,udmabuf-0.10.a";
                device-name = "udmabuf1";
                minor-number = <1>;
                size = <0x8000000>;
        };

# sudo insmod udmabuf.ko
[  675.848739] udmabuf udmabuf0: driver version = 1.4.2
[  675.853703] udmabuf udmabuf0: major number   = 241
[  675.858491] udmabuf udmabuf0: minor number   = 0
[  675.863108] udmabuf udmabuf0: phys address   = 0x000000006fd00000
[  675.869201] udmabuf udmabuf0: buffer size    = 134217728
[  675.874511] udmabuf udmabuf0: dma coherent   = 0
[  675.879124] udmabuf udmabuf@0: driver installed.
[  675.884167] udmabuf udmabuf@1: swiotlb buffer is full (sz: 134217728 bytes)
[  675.891138] udmabuf udmabuf@1: swiotlb: coherent allocation failed, size=134217728
[  675.898709] CPU: 3 PID: 3722 Comm: insmod Tainted: G           O      4.19.0 #1
[  675.906004] Hardware name: xlnx,zynqmp (DT)
[  675.910171] Call trace:
[  675.912609]  dump_backtrace+0x0/0x148
[  675.916259]  show_stack+0x14/0x20
[  675.919566]  dump_stack+0x90/0xb4
[  675.922865]  swiotlb_alloc+0x160/0x168
[  675.926606]  __dma_alloc+0xa8/0x1e0
[  675.930085]  udmabuf_platform_driver_probe+0x4b4/0x950 [udmabuf]
[  675.936078]  platform_drv_probe+0x50/0xa0
[  675.940078]  really_probe+0x1c8/0x280
[  675.943724]  driver_probe_device+0x54/0xe8
[  675.947803]  __driver_attach+0xe4/0xe8
[  675.951538]  bus_for_each_dev+0x70/0xc0
[  675.955364]  driver_attach+0x20/0x28
[  675.958924]  bus_add_driver+0x1dc/0x208
[  675.962743]  driver_register+0x60/0x110
[  675.966563]  __platform_driver_register+0x44/0x50
[  675.971255]  udmabuf_module_init+0x214/0x1000 [udmabuf]
[  675.976468]  do_one_initcall+0x74/0x178
[  675.980288]  do_init_module+0x54/0x1c8
[  675.984020]  load_module+0x1b5c/0x20e0
[  675.987753]  __se_sys_finit_module+0xb8/0xc8
[  675.992007]  __arm64_sys_finit_module+0x18/0x20
[  675.996522]  el0_svc_common+0x84/0xd8
[  676.000167]  el0_svc_handler+0x68/0x80
[  676.003899]  el0_svc+0x8/0xc
[  676.006787] dma_alloc_coherent() failed. return(0)
[  676.011573] udmabuf udmabuf@1: driver setup failed. return=-12
[  676.017759] udmabuf: probe of udmabuf@1 failed with error -12

ikwzm, 2019-12-12:
Thank you for the issue.

Please tell me how the cma area is secured. When you boot Linux Kernel, you will see the following message first.

```console
Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.0-xlnx-v2018.2-zynqmp-fpga (ichiro@sphinx-vm-ubuntu) (gcc version 5.4.0 20160609 (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.9)) #1 SMP Fri Aug 10 11:05:02 JST 2018
[    0.000000] Boot CPU: AArch64 Processor [410fd034]
[    0.000000] Machine model: ZynqMP Ultra96
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] cma: Reserved 256 MiB at 0x0000000070000000
```

udmabuf uses dma_alloc_coherent () to allocate memory.
dma_alloc_coherent () allocates memory from kernel space when there is no CMA.

Since the CMA area uses other devices, if udmabuf reserves 128MiB * 2, the CMA area will not be enough for 256MiB.

gi4you, 2019-12-12:
Thank you for your response.

Here is my system boot up message.

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 4.19.0 (oe-user@oe-host) (gcc version 8.2.0
(GCC)) #1 SMP Tue Dec 10 17:00:30 UTC 2019
[    0.000000] Machine model: xlnx,zynqmp
[    0.000000] earlycon: cdns0 at MMIO 0x00000000ff000000 (options
'115200n8')
[    0.000000] bootconsole [cdns0] enabled
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] cma: Reserved 256 MiB at 0x000000006fc00000

ikwzm, 2019-12-12:
I think you should increase the cma area. Try adding ```cma=320M``` to bootargs.

gi4you, 2019-12-12:
Thank you, after increased to 320M CAM size and works fine.

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 4.19.0 (oe-user@oe-host) (gcc version 8.2.0
(GCC)) #1 SMP Thu Dec 12 14:29:04 UTC 2019
[    0.000000] Machine model: xlnx,zynqmp
[    0.000000] earlycon: cdns0 at MMIO 0x00000000ff000000 (options
'115200n8')
[    0.000000] bootconsole [cdns0] enabled
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] cma: Reserved 320 MiB at 0x000000006bc00000

test/udmabuf# sudo insmod udmabuf.ko
[  117.733836] udmabuf: loading out-of-tree module taints kernel.
[  117.775766] udmabuf udmabuf0: driver version = 1.4.2
[  117.780733] udmabuf udmabuf0: major number   = 242
[  117.785528] udmabuf udmabuf0: minor number   = 0
[  117.790150] udmabuf udmabuf0: phys address   = 0x000000006bd00000
[  117.796247] udmabuf udmabuf0: buffer size    = 134217728
[  117.801555] udmabuf udmabuf0: dma coherent   = 0
[  117.806169] udmabuf udmabuf@0: driver installed.
[  117.844553] udmabuf udmabuf1: driver version = 1.4.2
[  117.849522] udmabuf udmabuf1: major number   = 242
[  117.854316] udmabuf udmabuf1: minor number   = 1
[  117.858937] udmabuf udmabuf1: phys address   = 0x0000000073d00000
[  117.865032] udmabuf udmabuf1: buffer size    = 134217728
[  117.870343] udmabuf udmabuf1: dma coherent   = 0
[  117.874953] udmabuf udmabuf@1: driver installed.

ikwzm, 2020-02-18:
Closed because it seems to be solved.

closed 2020-02-18 by ikwzm