ikwzm/udmabuf #34: will udmabuf be bottleneck

lifang2012, 2020-02-07:
hi, i want to use udmabuf in my axi dma userspace driver, according xilinx pg021 axi dma.pdf, axi dma throughput is about 300MB/s, in your README.md, 
zynq# dd if=/dev/udmabuf0 of=udmabuf0.bin bs=8388608
get about 55MB/s, will udmabuf be bottleneck of axi dma userspace driver? thank you.

ikwzm, 2020-02-07:
Thank you for the issue.

In the first place, what is compared is different. The performance indicated by PG021 is the transfer performance between PL-PS (SDRAM). What is measured by the dd command is the transfer performance between the CPU-SDRAM.

The dd command is a command to transfer from file to file. There is programmatic overhead internally.
For example, if the output file is set to /dev/null as follows, the performance will be improved.

```console
zynq# dd if=/dev/udmabuf0 of=/dev/null bs=8388608
1+0 records in
1+0 records out
8388608 bytes (8.4 MB, 8.0 MiB) copied, 0.0457842 s, 183 MB/s
```

lifang2012, 2020-02-08:
thank you, i will try later.

closed 2020-02-08 by lifang2012