ikwzm/udmabuf #37: Buffer for BRAM

noelpedro, 2020-02-26:
Hi Ichiro,

If I want to use udmabuf for vivado’s BRAM IP core. 

Do I need to set the bram cntrl node as a uio node in device tree or do I follow the memory region example below?

Thanks man

ikwzm, 2020-02-26:
Thank you for the issue.

However, I have never used BRAM IP core.

I'm sorry I can't be of any help.

noelpedro, 2020-02-26:
hmmm I was thinking somewhere along the lined of this :
[](https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18842412/Accessing+BRAM+In+Linux)
"section 3:  Acessing BRAM using /dev/mem

In that example that look a lot like the "memory-region" on your README thanks.

closed 2020-02-26 by noelpedro

ikwzm, 2020-02-26:
By the way, the person who gave me this PR(https://github.com/ikwzm/udmabuf/pull/8) said that I would use internal SRAM.