ikwzm/udmabuf #42: [QUESTION]Performance comparted to UIO?

ilpiero, 2020-04-10:
I looked at this project since I'm trying to optimize a RAM - to - reseved RAM transfer on a 7 series Zynq device.
It looks you did really a great work! It's explained very clearly and seem easy to use.
My question is the following: what is the performance compared to the uio driver?
Based on some benchmark I did, I could get 87.00 MBytes/s with the standard UIO implementation.
(https://forums.xilinx.com/t5/Embedded-Linux/User-IO-perfmance-on-Zynq-UIO/m-p/1094560/highlight/false#M41587), while at the bottom of the help page i could see a 55.4 MB/s using this driver.
Can you share some information about the system you used to get this data? Am I looking at the "right number"?

Thanks!!

Marco

ikwzm, 2020-04-10:
Thank you for the issue.

I measured it with ZYBO-Z7.
The result is as follows:

```console
root@debian-fpga:/home/fpga/examples/udmabuf_test# insmod /lib/modules/4.19.57-armv7-fpga/ikwzm/udmabuf.ko udmabuf0=42336000
[ 2295.399377] udmabuf udmabuf.0: DMA mask not set
[ 2295.522079] udmabuf udmabuf0: driver version = 1.4.6
[ 2295.527075] udmabuf udmabuf0: major number   = 242
[ 2295.531860] udmabuf udmabuf0: minor number   = 0
[ 2295.536517] udmabuf udmabuf0: phys address   = 0x30100000
[ 2295.541913] udmabuf udmabuf0: buffer size    = 42336256
[ 2295.547183] udmabuf udmabuf0: dma device     = udmabuf.0
[ 2295.552493] udmabuf udmabuf0: dma coherent   = 0
[ 2295.557156] udmabuf udmabuf.0: driver installed.

root@debian-fpga:/home/fpga/examples/udmabuf_test# ./test
0 ms to map the udmabuf device
154 ms to write 42336000 bytes
274.00 MBytes/s
86 ms to sync  42336000 bytes
492.00 MBytes/s
1 ms to unmap the udmabuf device
```

The source code of the program used for the measurement is at the following URL.

https://gist.github.com/ikwzm/c42bc209f5df1c4076d35c59de36bf1b

ilpiero, 2020-04-10:
Sounds very promising! I will try it on my design and I will share the results in the next days.
Thanks!