ikwzm/udmabuf #47: Kernel oops when syncing (and can't use reusable)

iabdalkader, 2020-05-27:
First I wanted to thank you for this module I find it very useful and exactly what I need for my project. That said, I'm seeing the exact same issue reported in #10 (kernel oops when sync'ing with reserved memory and `no-map;`). I tired using `reusable;` as you suggested, but I get an error when loading udmabuf.

device tree:
```
reserved-memory {
	#address-cells = <1>;
	#size-cells = <1>;
	ranges;

	fb: fb@0 {
		compatible = "shared-dma-pool";
		reg = <0xd8000000 0x100000>;
		label = "fb";
		reusable;
	};
};

udmabuf@0 {
	compatible = "ikwzm,u-dma-buf";
	device-name = "fb";
	minor-number = <0>;
	memory-region = <&fb>;
	size = <0x100000>; // 1MiB
};
```
When doing `modprobe u-dma-buf`:
```
[   16.367839] u_dma_buf: loading out-of-tree module taints kernel.
[   16.370110] u-dma-buf udmabuf@0: of_reserved_mem_device_init failed. return=-22
[   16.370380] u-dma-buf udmabuf@0: driver installed.
[   16.370403] u-dma-buf: probe of udmabuf@0 failed with error -22
```

I also noticed the memory is not actually reserved with `reusable` :
```
cat /proc/iomem
.....
d8000000-dfffffff : System RAM
```

It must be part of some memory pool. When using no-map I can see it's reserved:
```
cat /pro/iomem
....
d8100000-dfffffff : System RAM
```
I'm testing on ARM + kernel 4.19.49... Any suggestions to fix this ? I need to use reserve memory so the address is known, and be able to invalidate the cache after a DMA write...

ikwzm, 2020-05-28:
Thank you for the issue.

The Linux CMA region has alignment constraints.
For MPSoC, the address and size that can be specified in the regs property of reserved-memory must be in units of 0x00400000 (4MiB).

Please also refer to this issue.

https://github.com/ikwzm/udmabuf/issues/29

iabdalkader, 2020-05-29:
You were right, I fixed the alignment issue (in my case had to reserve 8MBytes) and it's now working. Thanks for the help.

closed 2020-05-29 by iabdalkader