ikwzm/udmabuf #49: Not working for kernel 5.4.0

p00chie, 2020-06-10:
The kernel module won't compile with petalinux 2020.1 with kernel version 5.4.0.
It was working with petalinux 2019.2 and kernel version 4.19.x

I have created a new module with petalinux 2020.1 just like i did with 2019.2 but changed the module name from udmabuf to u-dma-buf because it was mentioned that after kernel 5.0 an udmabuf module exists.

I will attach the log file from the petalinux output with udmabuf-2.1.5. and udmabuf-2.2.0-rc3.
[udmabuf-2.1.5-log.txt](https://github.com/ikwzm/udmabuf/files/4757972/udmabuf-2.1.5-log.txt)
[udmabuf-2.2.0-rc3-log.txt](https://github.com/ikwzm/udmabuf/files/4757973/udmabuf-2.2.0-rc3-log.txt)

Am i doing something wrong or do i miss something?

ikwzm, 2020-06-10:
Thank you for the issue.

I have not confirmed the operation on Linux Kernel 5.4 yet. 
Please wait for a while.

After a quick inspection, the is_device_dma_coherent() macro seems to have been removed since Linux 5.4.
It seems that dev_is_dma_coherent() has been added instead, but it is necessary to investigate a bit more to see if it has the same functionality.

ikwzm, 2020-06-15:
u-dma-buf v2.1.6-rc1 corresponding to Linux Kernel 5.4 has been released.
It's still a Release Candidate, but give it a try if you have the opportunity.

```
shell$ git clone -b v2.1.6-rc1 https://github.com/ikwzm/udmabuf.git u-dma-buf-2.1.6-rc1
```

p00chie, 2020-06-15:
Thanks for your support. It's working again :)
dtsi:
```
	reserved-memory {
		#address-cells = <2>;
		#size-cells = <2>;
		ranges;
		image_buf0: image_buf@0 {
			compatible = "shared-dma-pool";
			reusable;
			reg = <0x0 0x60000000 0x0 0x10000000>; 
			label = "image_buf0";
		};
	};

	udmabuf@0x00 {
		compatible = "ikwzm,u-dma-buf";
		device-name = "udmabuf0";
		minor-number = <0>;
		size = <0x01000000>;
		memory-region = <&image_buf0>;
	};
```
dmesg:
```
[    9.000251] u-dma-buf udmabuf@0x00: driver probe start.
[    9.001663] u-dma-buf udmabuf@0x00: assigned reserved memory node image_buf@0
[    9.019227] u-dma-buf udmabuf0: driver version = 3.0.0
[    9.032296] u-dma-buf udmabuf0: major number   = 243
[    9.041597] u-dma-buf udmabuf0: minor number   = 0
[    9.055035] u-dma-buf udmabuf0: phys address   = 0x0000000060000000
[    9.069946] u-dma-buf udmabuf0: buffer size    = 16777216
[    9.076150] u-dma-buf udmabuf0: dma device     = udmabuf@0x00
[    9.086192] u-dma-buf udmabuf0: dma coherent   = 0
[    9.092790] u-dma-buf udmabuf@0x00: driver installed.
```
u-dma-buf_test:
```
phys_addr=0x60000000
size=16777216
check_buf()
sync_mode=0, O_SYNC=0, time = 6.140172 sec
sync_mode=0, O_SYNC=1, time = 6.138564 sec
sync_mode=1, O_SYNC=0, time = 6.138256 sec
sync_mode=1, O_SYNC=1, time = 24.418203 sec
sync_mode=2, O_SYNC=0, time = 6.139768 sec
sync_mode=2, O_SYNC=1, time = 24.407102 sec
sync_mode=3, O_SYNC=0, time = 6.137998 sec
sync_mode=3, O_SYNC=1, time = 24.404777 sec
sync_mode=4, O_SYNC=0, time = 6.138019 sec
sync_mode=4, O_SYNC=1, time = 6.137971 sec
sync_mode=5, O_SYNC=0, time = 21.921401 sec
sync_mode=5, O_SYNC=1, time = 21.777650 sec
sync_mode=6, O_SYNC=0, time = 21.772914 sec
sync_mode=6, O_SYNC=1, time = 21.809481 sec
sync_mode=7, O_SYNC=0, time = 21.816524 sec
sync_mode=7, O_SYNC=1, time = 21.770559 sec
clear_buf()
sync_mode=0, O_SYNC=0, time = 0.207043 sec
sync_mode=0, O_SYNC=1, time = 0.206827 sec
sync_mode=1, O_SYNC=0, time = 0.206901 sec
sync_mode=1, O_SYNC=1, time = 0.877095 sec
sync_mode=2, O_SYNC=0, time = 0.206663 sec
sync_mode=2, O_SYNC=1, time = 0.211102 sec
sync_mode=3, O_SYNC=0, time = 0.206619 sec
sync_mode=3, O_SYNC=1, time = 0.211111 sec
sync_mode=4, O_SYNC=0, time = 0.206765 sec
sync_mode=4, O_SYNC=1, time = 0.206919 sec
sync_mode=5, O_SYNC=0, time = 0.877091 sec
sync_mode=5, O_SYNC=1, time = 0.877015 sec
sync_mode=6, O_SYNC=0, time = 0.211037 sec
sync_mode=6, O_SYNC=1, time = 0.211010 sec
sync_mode=7, O_SYNC=0, time = 0.211022 sec
sync_mode=7, O_SYNC=1, time = 0.211034 sec
```

closed 2020-06-15 by p00chie

SushiTee, 2021-03-03:
I'm having and issue, too.

I'm building with petalinux 2020.2 and I got it to compile. I had to change the Makefile to make it build.

I did the following steps:

* petalinux-create -t modules -n u-dma-buf --force --enable
* remove the stuff from the _files_ directory and cloned the master branch into the _files_ directory.
* Then I changed the make file like so:
```
HOST_ARCH       ?= $(shell uname -m | sed -e s/arm.*/arm/ -e s/aarch64.*/arm64/)
ARCH            ?= $(shell uname -m | sed -e s/arm.*/arm/ -e s/aarch64.*/arm64/)

ifeq ($(ARCH), arm)
 ifneq ($(HOST_ARCH), arm)
   CROSS_COMPILE  ?= arm-linux-gnueabihf-
 endif
endif
ifeq ($(ARCH), arm64)
 ifneq ($(HOST_ARCH), arm64)
   CROSS_COMPILE  ?= aarch64-linux-gnu-
 endif
endif

u-dma-buf-obj           := u-dma-buf.o
obj-$(CONFIG_U_DMA_BUF) += $(u-dma-buf-obj)

ifndef UDMABUF_MAKE_TARGET
  KERNEL_VERSION_LT_5 ?= $(shell awk '/^VERSION/{print int($$3) < 5}' $(KERNEL_SRC)/Makefile)
  ifeq ($(KERNEL_VERSION_LT_5), 1)
    UDMABUF_MAKE_TARGET ?= modules
  else
    UDMABUF_MAKE_TARGET ?= u-dma-buf.ko
  endif
endif

all:
	make -C $(KERNEL_SRC) ARCH=$(ARCH) CROSS_COMPILE=$(CROSS_COMPILE) M=$(PWD) obj-m=$(u-dma-buf-obj) $(UDMABUF_MAKE_TARGET)

modules_install:
	make -C $(KERNEL_SRC) ARCH=$(ARCH) CROSS_COMPILE=$(CROSS_COMPILE) M=$(PWD) obj-m=$(u-dma-buf-obj) modules_install

clean:
	make -C $(KERNEL_SRC) ARCH=$(ARCH) CROSS_COMPILE=$(CROSS_COMPILE) M=$(PWD) clean
```
I removed the _KERNEL_SRC_DIR  ?= /lib/modules/$(shell uname -r)/build_ part. and added the _modules_install_ section as I don't want to load it with insmod. It should be directly integrated into the kernel.

It worked just fine when compiling with petalinux 2019.2.

It is simply not creating the deice files.

The device tree looks like this:

```
reserved-memory {
    #address-cells = <1>;
    #size-cells = <1>;
    ranges;
    sensor_buf0: sensor_buf@0 {
      compatible = "shared-dma-pool";
      reusable;
      reg = <0x10000000 0x2000000>;
      label = "sensor_buf0";
    };
    sensor_buf1: sensor_buf@1 {
      compatible = "shared-dma-pool";
      reusable;
      reg = <0x12000000 0x2000000>;
      label = "sensor_buf1";
    };
    sensor_buf2: sensor_buf@2 {
      compatible = "shared-dma-pool";
      reusable;
      reg = <0x14000000 0x2000000>;
      label = "sensor_buf2";
    };
    sensor_buf3: sensor_buf@3 {
      compatible = "shared-dma-pool";
      reusable;
      reg = <0x16000000 0x2000000>;
      label = "sensor_buf3";
    };
    sensor_buf4: sensor_buf@4 {
      compatible = "shared-dma-pool";
      reusable;
      reg = <0x18000000 0x2000000>;
      label = "sensor_buf4";
    };
    sensor_buf5: sensor_buf@5 {
      compatible = "shared-dma-pool";
      reusable;
      reg = <0x1a000000 0x2000000>;
      label = "sensor_buf5";
    };
    sensor_buf6: sensor_buf@6 {
      compatible = "shared-dma-pool";
      reusable;
      reg = <0x1c000000 0x2000000>;
      label = "sensor_buf6";
    };
    sensor_buf7: sensor_buf@7 {
      compatible = "shared-dma-pool";
      reusable;
      reg = <0x1e000000 0x2000000>;
      label = "sensor_buf7";
    };
  };

  udmabuf@0 {
    compatible = "ikwzm,u-dma-buf";
    device-name = "udmabuf0";
    size = <0x2000000>; // 32MiB
    memory-region = <&sensor_buf0>;
  };
  udmabuf@1 {
    compatible = "ikwzm,u-dma-buf";
    device-name = "udmabuf1";
    size = <0x2000000>; // 32MiB
    memory-region = <&sensor_buf1>;
  };
  udmabuf@2 {
    compatible = "ikwzm,u-dma-buf";
    device-name = "udmabuf2";
    size = <0x2000000>; // 32MiB
    memory-region = <&sensor_buf2>;
  };
  udmabuf@3 {
    compatible = "ikwzm,u-dma-buf";
    device-name = "udmabuf3";
    size = <0x2000000>; // 32MiB
    memory-region = <&sensor_buf3>;
  };
  udmabuf@4 {
    compatible = "ikwzm,u-dma-buf";
    device-name = "udmabuf4";
    size = <0x2000000>; // 32MiB
    memory-region = <&sensor_buf4>;
  };
  udmabuf@5 {
    compatible = "ikwzm,u-dma-buf";
    device-name = "udmabuf5";
    size = <0x2000000>; // 32MiB
    memory-region = <&sensor_buf5>;
  };
  udmabuf@6 {
    compatible = "ikwzm,u-dma-buf";
    device-name = "udmabuf6";
    size = <0x2000000>; // 32MiB
    memory-region = <&sensor_buf6>;
  };
  udmabuf@7 {
    compatible = "ikwzm,u-dma-buf";
    device-name = "udmabuf7";
    size = <0x2000000>; // 32MiB
    memory-region = <&sensor_buf7>;
  };
```

ikwzm, 2021-03-04:
Thank you very much for the valuable information. 

I modified the Makefile. 
I'm not using PetaLinux so I can't try it.  If you like, check if it works with PetaLinux.

Thank you.

SushiTee, 2021-03-04:
Thank you. I still had to add the modules_install target.

```
modules_install:
	make -C $(KERNEL_SRC_DIR) ARCH=$(ARCH) CROSS_COMPILE=$(CROSS_COMPILE) M=$(PWD) obj-m=$(u-dma-buf-obj) modules_install
```

It is working as expected now. I forgot to copy the rootfs including the /lib/modules/... directory 😅

ikwzm, 2021-03-04:
Thank you.

I modified the Makefile.