ikwzm/udmabuf #50: Ubuntu 16 Compile Error

ollietechnology, 2020-06-16:
I am trying to compile this for Petalinux 2019.2 using a Ubuntu 16.04 development environment on kernel version 4.15. Petalinux fails compilation during the compilation of this module throwing error, 
include/linux/refcount.h:58:12: fatal error: asm/refcount.h: No such file or directory. I wouldn't expect that my version of the OS is too old to support this. Do you have any experience with this error mode? I can't seem to find any references to an asm/refcount.h anywhere online. Any help would be appreciated.

ikwzm, 2020-06-17:
Thank you for the issue.

I'm sorry I can't be of any help, because I don't use Petalinux.

However, using include/asm/refcount.h is when building a module for x86. 
Do you want udmabuf for x86?

ollietechnology, 2020-06-17:
oh that would change it. Good call. I'm trying to cross compile for ARM64 from an x86 Ubuntu system. The Petalinux tools handle most of the work for me in this case so I don't need to specify any architecture anywhere. Do I need to modify the makefile or provide some other magic to cross compile for this case?

ikwzm, 2020-06-17:
I'm sorry I can't be of any help, because I don't use Petalinux.

formalism, 2020-06-17:
Hello @ollietechnology ,

My build environment is somewhat different from yours (PetaLinux 2019.1, Ubuntu 18.04 host machine), but I can build the udmabuf without any hassles.

The method is written in UG1144 from Xilinx, Chapter 8, "Steps to Add Custom Modules".

1. `$ petalinux-create -t modules --name udmabuf --enable`
2. Replace <project-dir>/project-spec/meta-user/recipes-modules/udmabuf/files/udmabuf.c with the one obtained from GitHub.
3. `$ petalinux-build`

The above procedure builds udmabuf.ko and puts it in the root file system. You can load the module in the target by modprobe command.

Hope this helps,

akira-nishiyama, 2020-06-17:
I just tried Petalinux 2019.2 and Ubuntu 18.04 host and I could build the udmabuf with the method @formalism represent.
Perhaps the Makefile provided by this repository is not suite for Petalinux project.

ollietechnology, 2020-06-17:
FELLAS! That is the magic sauce that I needed. I was copying over both the makefile and udmabuf.c and that was the source of my error. Using the petalinux generated makefile was the key. Thanks for your help!

closed 2021-01-15 by ikwzm