ikwzm/udmabuf #54: Q: is it possible to use same /dev/udmabuf0 device from multiple programms?

h4tr3d, 2020-08-05:
Scenario is simple: shared memory implementation in case, when hardware blocks is used.

Application1 open /dev/udmabuf0, mmap it, feed buffer into V4L2 and signal Application 2.

Application 2 open /dev/udmabuf0, take some data and feed buffer into... VCU codec, for example. 

In such case, we can omit extra memory coping.

ikwzm, 2020-08-05:
Thank you for the issue.

u-dma-buf can be open()&mmap() simultaneously in multiple applications.
u-dma-buf does not perform exclusive control in particular. 
Please perform exclusive control between applications.

h4tr3d, 2020-08-07:
@ikwzm Thanks a lot for clarification. I understand that external inter-process locking interface is needed.

closed 2021-01-15 by ikwzm