ikwzm/udmabuf #55: data reset during u-dma-buf creation

jowoeber, 2020-08-11:
Hi,
I would like to use u-dma-buf to transfer data between two processors in an Ultrascale+ platform. The problem is, that one Processor is running much earlier and producing data much earlier than the other processor which is running the Linux OS and uses the u-dma-buf module.

So there is data in the specified memory but during the boot of the Linux OS and the creation of the u-dma-buf it is reset. I am creating the u-dma-buf via a device tree entry nearly identical to the one in the documentation. If i remove the udmabuf node and only leave the reserved-memory node the data remains in the buffer.

Is there a simple solution to this anybody can think of? I tried removing the reusable option but this only results in a kernel panic.

Thanks

```
reserved-memory {
	#address-cells = <1>;
	#size-cells = <1>;
	ranges;
	image_buf0: image_buf@0 {
		compatible = "shared-dma-pool";
		reusable;
		reg = <0x3C000000 0x04000000>; 
		label = "image_buf0";
	};
};
udmabuf@0 {
	compatible = "ikwzm,u-dma-buf";
	device-name = "udmabuf0";
	size = <0x04000000>; // 64MiB
	memory-region = <&image_buf0>;
};
```

ikwzm, 2020-08-11:
Thank you for the issue.

u-dma-buf will clear the buffer when it is created. 
Unfortunately, there is no way to stop it.

What is the purpose of using u-dma-buf? 
Is /dev/mem or /dev/uio no good?

jowoeber, 2020-08-11:
Thanks.
I will try using /dev/mem of uio for that purpose.

jowoeber, 2020-08-11:
I am using u-dma-buf because I had no caching problems using it. /dev/mem surely is the better fit for my problem but I always have caching issues using it. But if the ram is reset on purpose using udmabuf I will try /dev/mem again.

ikwzm, 2020-08-11:
Thank you for your reply.

If you want to enable caching and access, try uiomem.
uiomem is a derivative of u-dma-buf.
uiomem is still in alpha version.
uiomem is located at the following URL.

https://github.com/ikwzm/uiomem/tree/v1.0.0-alpha.1

Prepare the following device tree.

```
        reserved-memory {
                #address-cells = <2>;
                #size-cells = <2>;
                ranges;
                image_buf0: image_buf@0 {
                        compatible = "shared-dma-pool";
                        reg = <0x0 0x3C000000 0x0 0x04000000>;
                        no-map;
                };
        };
        uiomem@0 {
                #address-cells = <2>;
                #size-cells = <2>;
                ranges;
                compatible = "ikwzm,uiomem";
                reg = <0x0 0x3C000000 0x0 0x04000000>;
        };
```