ikwzm/udmabuf #60: of_reserved_mem_device_init failed. return=-22

sombohan, 2020-10-06:
Hi,

I'm trying to include the module to my Kernel 5.4.54 the cma module is enabled in the menuconfig my device tree is configured as following.

        reserved-memory {
                #address-cells = <1>;
                #size-cells = <1>;
                ranges;
                image_buf0: image_buf@0 {
                        compatible = "shared-dma-pool";
                        reusable;
                        reg = <0x30000000 0x08000000>;
                        label = "image_buf0";
                };
                image_buf1: image_buf@1 {
                        compatible = "shared-dma-pool";
                        reusable;
                        reg = <0x38000000 0x08000000>;
                        label = "image_buf1";
                };
        };

        udmabuf@0x00 {
                compatible = "ikwzm,u-dma-buf";
                device-name = "udmabuf0";
                size = <0x08000000>; //128MiB
                memory-region = <&image_buf0>;
        };

        udmabuf@0x01 {
                compatible = "ikwzm,u-dma-buf";
                device-name = "udmabuf0";
                size = <0x08000000>; //128MiB
                memory-region = <&image_buf1>;
        };

The output of insmod:
> root@cyclone5:~# insmod /lib/modules/5.4.54-altera/extra/u-dma-buf.ko udmabuf0=1
048576
[  351.542648] u-dma-buf udmabuf@0x00: of_reserved_mem_device_init failed. return=-22
[  351.550566] u-dma-buf udmabuf@0x00: driver installed.
[  351.555651] u-dma-buf: probe of udmabuf@0x00 failed with error -22
[  351.562062] u-dma-buf udmabuf@0x01: of_reserved_mem_device_init failed. return=-22
[  351.569855] u-dma-buf udmabuf@0x01: driver installed.
[  351.574927] u-dma-buf: probe of udmabuf@0x01 failed with error -22
[  351.585628] u-dma-buf udmabuf0: driver version = 3.2.2
[  351.590864] u-dma-buf udmabuf0: major number   = 246
[  351.595843] u-dma-buf udmabuf0: minor number   = 0
[  351.600616] u-dma-buf udmabuf0: phys address   = 0x2dd00000
[  351.606177] u-dma-buf udmabuf0: buffer size    = 1048576
[  351.611479] u-dma-buf u-dma-buf.0: driver installed.

How to resolve the error? Am I missing some extra configuration step in the kernel menuconfig?

What I want to accomplish: Reserving 2x 128MB of space for DMA Transfers.

Regards,

Sombohan

ikwzm, 2020-10-06:
Thank you for the issue

Check the boot log.
Is the CMA area reserved?
If so, you will see a message similar to the following in the boot log:

```
[    0.000000] Reserved memory: created CMA memory pool at 0x38000000, size 128 MiB
[    0.000000] OF: reserved mem: initialized node image_buf@1, compatible id shared-dma-pool
[    0.000000] Reserved memory: created CMA memory pool at 0x30000000, size 128 MiB
[    0.000000] OF: reserved mem: initialized node image_buf@0, compatible id shared-dma-pool
[    0.000000] cma: Reserved 16 MiB at 0x2f000000
```

Please also refer to #45.

sombohan, 2020-10-07:
Hi thank you for the quick response. Unfortunately I've tried the examples in your referenced thread. Without success, I've tried to use instead the dynamic variant of cma by supplying the size parameter but the only output in the log is: `node linux,cma compatible matching fail`

So I don't pass the cma stuff to get the u-dma-buf running.

Regards,

Sombohan

sombohan, 2020-10-16:
I've found thr reason for the -22. The CONFIG_DMA_CMA has to be set too.

Now after trying around. I can allocate 256MiB by CMA but if I try to use the complete chunk for one buffer its getting error -12 is there any extra space needed in u-dma-buf?

    reserved-memory {
        #address-cells = <1>;
        #size-cells = <1>;
        ranges;
        image_buf0: image_buf@0 {
            compatible = "shared-dma-pool";
            reusable;
            size = <0x10000000>;
            label = "image_buf0";
        };
    };

    udmabuf@0 {
        compatible = "ikwzm,u-dma-buf";
        device-name = "udmabuf0";
        size = <0x10000000>; // 256MiB
        memory-region = <&image_buf0>;
    };

Regards,

Sombohan