ikwzm/udmabuf #61: Delete on abnormal termination.

ldalessa, 2020-10-26:
I'd like to be able to have buffers `deleted` up on termination. Normal `atexit/on_exit` is fine for normal termination, but abnormal termination is much more complex, requiring signal chaining or some form of extra-process monitoring.

I'm not familiar with the kernel internals. Is this functionality that the `u-dma-buf` driver could support through flags during `create`? Would it require some sort of daemon, or could it append the buffers as resources that are already cleaned up by the kernel?

ikwzm, 2020-10-26:
Thank you for the issue.

With u-dma-buf, buffers can only be created/deleted by adding/deleting device tree or by using u-dma-buf-mgr. I wasn't thinking of removing it automatically when the user application was closed. If you want to remove it automatically, you will need a daemon.

ldalessa, 2020-10-26:
Okay, thanks. I didn't know if there was `atexit` kernel equivalents that you could already hook up to that I don't have access to in user-space. Oh well.

closed 2020-10-26 by ldalessa