ikwzm/udmabuf #64: Trying to perform DMA reads

daloisio, 2020-11-12:
HI,

I have no problems with your driver.
I am only using it to perform 64 reads to DMA pages, i need to read 64 unchaced pages on CORTEX-A72 ARMv8.
I edited the u-dma-buf_test.c a little bit. I simply added the following code:

	unsigned char* mem;
	if ((fd  = open("/dev/udmabuf0", O_RDWR | O_SYNC)) != -1) {
		mem = mmap(NULL,buf_size, PROT_READ|PROT_WRITE, MAP_SHARED, fd, 0);
	}
	close(fd);
	unsigned char*temp=mem;
	unsigned char *dma_page_buffer[64];
	for(int i=0;i<64;i++){
		printf("addr: %p\n", temp+(i*PAGE_SIZE));
		dma_page_buffer[i] = temp+(i*PAGE_SIZE);
	}

and then i perform the reads of the pages with dma_page_buffer. To perform uncached reads, should i use the instructions to flush the cache or i should not because the CPU cache of allocated DMA buffer is disabled ? I specify the O_SYNC flag in
mmap function.
 I wanted to make sure i am doing things in the right way.

Thank you.

ikwzm, 2020-11-12:
Thank you for the issue.

> To perform uncached reads, should i use the instructions to flush the cache or i should not because the CPU cache of allocated DMA buffer is disabled ? I specify the O_SYNC flag in
mmap function.

Opening the device file with O_SYNC disables the CPU cache.
When the CPU cache is disabled, there is no need to manually specify the cache flush & invalidate.

daloisio, 2020-11-12:
Thank you for your answer and help.

Best.

Inviato da Outlook Mobile<https://aka.ms/blhgte>

closed 2021-01-15 by ikwzm