ikwzm/udmabuf #65: delete hang forever

florisch, 2020-11-18:
Hi,

I am trying to use u-dma-buf-mgr to create and delete my buffers on demand. Creation seems to work but delete hang forever.

```
root@hw1:~# uname -a
Linux hw1 4.4.0-xilinx-00003-g5d53636e #44 SMP PREEMPT Wed Nov 18 09:13:00 CET 2020 armv7l GNU/Linux
root@hw1:~# modprobe udmabuf
root@hw1:~# echo "create udmabuf0 4096" > /dev/u-dma-buf-mgr
u-dma-buf-mgr : create udmabuf0 4096
u-dma-buf udmabuf0: driver version = 3.2.2
u-dma-buf udmabuf0: major number   = 247
u-dma-buf udmabuf0: minor number   = 0
u-dma-buf udmabuf0: phys address   = 0x1f043000
u-dma-buf udmabuf0: buffer size    = 4096
u-dma-buf u-dma-buf.0.auto: driver installed.
root@hw1:~# cat /sys/class/u-dma-buf/udmabuf0/phys_addr
0x1f043000
root@hw1:~# cat /sys/class/u-dma-buf/udmabuf0/size
4096
root@hw1:~# echo "delete udmabuf0" > /dev/u-dma-buf-mgr
u-dma-buf-mgr : delete udmabuf0
```

The ``echo "delete udmabuf0" > /dev/u-dma-buf-mgr`` never returns. Then after 120sec, the hang is detected by kernel with the log below.

````
INFO: task sh:1022 blocked for more than 120 seconds.
      Tainted: G           O    4.4.0-xilinx-00003-g5d53636e #44
"echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
sh              D 4047e004     0  1022   1021 0x00000001
[<4047e004>] (__schedule) from [<4047e324>] (schedule+0x50/0xc0)
[<4047e324>] (schedule) from [<4047e758>] (schedule_preempt_disabled+0x24/0x40)
[<4047e758>] (schedule_preempt_disabled) from [<4047fc2c>] (__mutex_lock_slowpath+0xc8/0x168)
[<4047fc2c>] (__mutex_lock_slowpath) from [<4047fd14>] (mutex_lock+0x48/0x4c)
[<4047fd14>] (mutex_lock) from [<3f0750f4>] (udmabuf_get_device_name_property+0x1c/0x7c [udmabuf])
[<3f0750f4>] (udmabuf_get_device_name_property [udmabuf]) from [<3f075858>] (udmabuf_manager_file_write+0x364/0x4dc [udmabuf])
[<3f075858>] (udmabuf_manager_file_write [udmabuf]) from [<400d550c>] (__vfs_write+0x20/0xc8)
[<400d550c>] (__vfs_write) from [<400d5d00>] (vfs_write+0x90/0x164)
[<400d5d00>] (vfs_write) from [<400d651c>] (SyS_write+0x44/0x9c)
[<400d651c>] (SyS_write) from [<4000f6c0>] (ret_fast_syscall+0x0/0x3c)
````

ikwzm, 2020-11-18:
Thank you for the issue.

This has been fixed in u-dma-buf v3.2.3.

Thank you again for everything you’ve done

florisch, 2020-11-18:
Thanks you for fixing this issue so quickly.

Work fine now.

closed 2020-11-18 by florisch