ikwzm/udmabuf #70: petalinux2020.2 do not work with u-dma-buf

roundchen, 2021-03-05:
Hi ikwzm

I'm using u-dma-buf (master branch)on a Zynq MPSoC Ultrascale+ board (ZCU111),and I use xilinx offer petalinux bsp for zcu111 with the petalinux version is 2020.2.
I compile u-dma-buf.c and insmod it install well. then I run the test program a.out ( u-dma-buf_test.c) ,there is a Segmentation fault .
It seems u-dma-buf can not work with petalinux 2020.2

then I compile u-dma-buf.c in petalinux 2019.1,it work well

thanks

root@xilinx-zcu111-2020_2:~# mount /dev/mmcblk0p1 /mnt                                                                              
[   54.788933] /dev/mmcblk0p1: Can't open blockdev                                                                                  
root@xilinx-zcu111-2020_2:~#                                                                                                        
root@xilinx-zcu111-2020_2:~# cd  /mnt                                                                                               
root@xilinx-zcu111-2020_2:/mnt#                                                                                                     
048576ilinx-zcu111-2020_2:/mnt# insmod u-dma-buf.ko udmabuf0=78643200 udmabuf1=1048576                                                  
[   54.867568] u_dma_buf: loading out-of-tree module taints kernel.                                                                 
[   54.896593] u-dma-buf udmabuf0: driver version = 3.2.4                                                                           
[   54.901732] u-dma-buf udmabuf0: major number   = 242                                                                             
[   54.906689] u-dma-buf udmabuf0: minor number   = 0                                                                               
[   54.911482] u-dma-buf udmabuf0: phys address   = 0x0000000068500000                                                              
[   54.917747] u-dma-buf udmabuf0: buffer size    = 78643200                                                                        
[   54.923144] u-dma-buf u-dma-buf.0: driver installed.                                                                             
[   54.929525] u-dma-buf udmabuf1: driver version = 3.2.4                                                                           
[   54.934669] u-dma-buf udmabuf1: major number   = 242                                                                             
[   54.939632] u-dma-buf udmabuf1: minor number   = 1                                                                               
[   54.944418] u-dma-buf udmabuf1: phys address   = 0x000000006d000000                                                              
[   54.950680] u-dma-buf udmabuf1: buffer size    = 1048576                                                                         
[   54.955987] u-dma-buf u-dma-buf.1: driver installed.                                                                             
root@xilinx-zcu111-2020_2:/mnt#                                                                                                     
root@xilinx-zcu111-2020_2:/mnt# ./a.out                                                                                             
phys_addr=0x68500000                                                                                                                
size=78643200                                                                                                                       
check_buf()                                                                                                                         
Segmentation fault

roundchen, 2021-03-05:
I solve this problem using petalinux generated kernel  compile  u-dma-buf.c 
when I download linux-xlnx and compiled as the kernel to compile  u-dma-buf.c ,this problem appear.
thank you for your contribution to open Source

ikwzm, 2021-03-05:
Thank you for the issue.

This issue did not occur with 5.4.0-xlnx-v2020.2-zynqmp-fpga provided by ZynqMP-FPGA-Linux. 5.4.0-xlnx-v2020.2-zynqmp-fpga is not built using PetaLinux.

 * https://github.com/ikwzm/ZynqMP-FPGA-Linux

I'm not using PetaLinux, so it's not easy to solve this problem. 
Please provide any information.