ikwzm/udmabuf #74: Zynq u-dma-buf with DMA

gtortone, 2021-03-31:
Hi,
do you have an example/tutorial to use DMA on Zynq with u-dma-buf ?

I already have a Vivado design with a PL FIFO filled by a counter for Microzed (Zynq-7000) that work 
with dma-proxy (Xilinx demo) but I'did not find any example on how to use your implementation to receive 
data from DMA.

Thanks in advance.

ikwzm, 2021-03-31:
Thank you for the issue.

I haven't prepared a tutorial in particular, but the following repositories may be helpful.

 * https://github.com/ikwzm/FPGA-SoC-Linux-Example-1-ZYBO-Z7
 * https://github.com/ikwzm/FPGA-SoC-Linux-Example-2-ZYBO-Z7

gtortone, 2021-04-01:
Hi,
thanks a lot for the links ! They are very helpful for me to understand DMA on Zynq.
I realized that in your C code you are using DMA in direct-mode... do you have an example
to use DMA scatter-gather mode ?

Thanks a lot !

ikwzm, 2021-04-01:
Excuse me. There is no scatter-gather DMA sample. u-dma-buf reserves a "contiguous" memory space in the kernel as a buffer, so we don't use scatter-gather mode.

gtortone, 2021-04-01:
Hi,
thanks for the explanation... but I'm asking if it is possible to use scatter-gather DMA with multiple u-dma-buf kernel buffers... or inside a single u-dma-buf using different sub-buffers...

ikwzm, 2021-04-01:
I have no experience with it, but I think it is possible.

balaji-ch, 2021-04-01:
> 
> 
> I have no experience with it, but I think it is possible.

Could you suggest how can that be done. I am working on a platform that doesn't have CMA and at max I am able to allocate only 4MiB . Lets say I want a 16MiB buffer, do I need to create a 4 instances of this device ?