ikwzm/udmabuf #75: Check for contiguity

balaji-ch, 2021-04-12:
Hi,

Is there a way to check that memory allocated is contagious?

For example, in Ubuntu VM (CMA enabled and disabled CONFIG_STRICT_DEVMEM) 
```
dmesg | cma
[ 0.068544] cma: Reserved 64 MiB at 0x000000021bc00000

sudo insmod u-dma-buf.ko dma_mask_bit=64 udmabuf0=8388608
u-dma-buf udmabuf0: driver installed
u-dma-buf udmabuf0: major number   = 248
u-dma-buf udmabuf0: minor number   = 0
u-dma-buf udmabuf0: phys address   = 0x000000021be00000
u-dma-buf udmabuf0: buffer size    = 8388608
u-dma-buf udmabuf0: dma coherent  = 0
```
I could test this by  dumping the physical memory  dd if=/dev/mem bs=1 count=8388608 offset=0x21be000000

But on an embedded board where I reserved the memory via DTS (instead of CMA) and when tested in the similar fashion it says unreachable offset.

ikwzm, 2021-04-12:
Thank you for the issue.

What kind of DTS do you use?

What arguments does the dd command specify?

balaji-ch, 2021-04-13:
The DTS is as follows
```
		 memory@80000000 {
			device_type = "memory";
			reg = <0x0 0x80000000 0x0 0x40000000>;

		};
 

        reserved-memory {
            #address-cells = <2>;
            #size-cells = <2>;
            ranges;
            image_buf0: img_buf@0xbc000000 {
                compatible = "shared-dma-pool";
                no-map;
                reg = <0x0 0xbc000000 0x0 0x4000000>;
            };
			
			
	  udmabuf@0 {
		compatible = "ikwzm,u-dma-buf";
		device-name = "udmabuf0";
		size = <0x04000000>; // 64MiB
		memory-region = <&image_buf0>;

	 };
```

The command is as follows
```
dd if=/dev/udmabuf0 bs=1 count=1024 skip=3154116608
```

balaji-ch, 2021-04-13:
Hey I am able to resolve it by 
```dd if=/dev/udmabuf0 bs=1 count=1024 skip=0```
But when I use it like this
```dd if=/dev/mem bs=1 count=1024 skip=0 ```
or
```dd if=/dev/mem bs=1 count=1024 skip=315411608```
I get **error reading /dev/mem/ : Bad address**

ikwzm, 2021-04-13:
/dev/udmabuf is treated as a file of the size specified by the size property.  An error will occur if you specify an area that exceeds the size.

I don't know why /dev/mem fails.