ikwzm/udmabuf #76: Regarding Manual cache management with the CPU cache still being enabled

balaji-ch, 2021-04-25:
Do we need to call 
fd  = open("/sys/class/u-dma-buf/udmabuf0/sync_for_device", O_WRONLY)) != -1)
once before mmap ? or any special care need to be taken for manual cache management ?

ikwzm, 2021-04-25:
Thank you for the issue.

> Do we need to call
fd = open("/sys/class/u-dma-buf/udmabuf0/sync_for_device", O_WRONLY)) != -1)
once before mmap ?

there is no need. 
It's the same whether you open the file before mmap() or after mmap().

> or any special care need to be taken for manual cache management ?

Knowledge of general CPU Cache and Linux Kernel is required.

balaji-ch, 2021-04-26:
But here in this example
https://github.com/ikwzm/ZynqMP-FPGA-Linux-Example-2-UltraZed/blob/master/negative.py
        udmabuf4.sync_for_device()
        udmabuf5.sync_for_device()
is called in the for loop ?

ikwzm, 2021-04-26:
In this example, sync_for_device()/sync_for_cpu() is inside the loop because it runs multiple times to calculate the average time it takes to process it.