ikwzm/udmabuf #80: kernel 5.10 support

Tobinjh, 2021-09-08:
I've recently upgraded from petalinux 2018.3 to 2021.1 which uses the 5.10.0 kernel.  on 2018.3 the driver seems to be working well. Furthermore, I've been using cache operations available to el0 (SCTLR_EL1.UCI= 1 from what I see) to avoid the expensive (for me) kernel calls in my use case.  not sure if this is fully legal, but it seems to work.  However when upgrading to 5.10.0 the memory space allocated by the driver seems to cause bus faults using O_SYNC, and without O_SYNC  I have other failures which lead me to suspect the driver.

Are there any obvious issues with this?

ikwzm, 2021-09-09:
Thank you for the issue.

I have not confirmed the operation on Linux Kernel 5.10 yet.
Please wait for a while.

ikwzm, 2021-11-10:
Thank you for the issue.

I tried it with ZynqMP-FPGA-Linux v2021.1.1-rc3 (linux 5.10.0-xlnx-v2021.1-zynqmp-fpga). When I ran u-dma-buf_test, it ended normally as follows.

```console
shell# insmod u-dma-buf udmabuf0=1000000
shell# ./u-dma-buf_test
phys_addr=0x70100000
size=1000000
check_buf()
sync_mode=0, O_SYNC=0, time = 0.292597 sec
sync_mode=0, O_SYNC=1, time = 0.292479 sec
sync_mode=1, O_SYNC=0, time = 0.292482 sec
sync_mode=1, O_SYNC=1, time = 1.249395 sec
sync_mode=2, O_SYNC=0, time = 0.292544 sec
sync_mode=2, O_SYNC=1, time = 1.248974 sec
sync_mode=3, O_SYNC=0, time = 0.292435 sec
sync_mode=3, O_SYNC=1, time = 1.249206 sec
sync_mode=4, O_SYNC=0, time = 0.292503 sec
sync_mode=4, O_SYNC=1, time = 0.292480 sec
sync_mode=5, O_SYNC=0, time = 1.249416 sec
sync_mode=5, O_SYNC=1, time = 1.249356 sec
sync_mode=6, O_SYNC=0, time = 1.248993 sec
sync_mode=6, O_SYNC=1, time = 1.248952 sec
sync_mode=7, O_SYNC=0, time = 1.248992 sec
sync_mode=7, O_SYNC=1, time = 1.249032 sec
clear_buf()
sync_mode=0, O_SYNC=0, time = 0.008233 sec
sync_mode=0, O_SYNC=1, time = 0.008180 sec
sync_mode=1, O_SYNC=0, time = 0.008148 sec
sync_mode=1, O_SYNC=1, time = 0.046856 sec
sync_mode=2, O_SYNC=0, time = 0.008132 sec
sync_mode=2, O_SYNC=1, time = 0.012274 sec
sync_mode=3, O_SYNC=0, time = 0.008138 sec
sync_mode=3, O_SYNC=1, time = 0.012286 sec
sync_mode=4, O_SYNC=0, time = 0.008134 sec
sync_mode=4, O_SYNC=1, time = 0.008189 sec
sync_mode=5, O_SYNC=0, time = 0.046847 sec
sync_mode=5, O_SYNC=1, time = 0.046813 sec
sync_mode=6, O_SYNC=0, time = 0.012257 sec
sync_mode=6, O_SYNC=1, time = 0.012236 sec
sync_mode=7, O_SYNC=0, time = 0.012250 sec
sync_mode=7, O_SYNC=1, time = 0.012237 sec
```

closed 2022-05-15 by ikwzm