ikwzm/udmabuf #81: Allocating multiple DMA buffers but not in a single parse

rajatkha, 2021-11-10:
What if I want to allocate two buffers of size 256 bytes but not in a single go?
insmod u-dma-buf.ko udmabuf0=256 udmabuf1=256 --> this is correct
but what if I want to allocate them separately?
insmod u-dma-buf.ko udmabuf0=256
now what about the udmabuf1? How do I allocate this now?

ikwzm, 2021-11-10:
Thank you for the issue.

How about using ```/dev/u-dma-buf/mgr```?
Allocate the buffer as follows:

```console
shell# insmod u-dma-buf.ko udmabuf0=256
shell# echo 'create udmabuf1 256 > /dev/u-dma-buf-mgr
```

Alternatively, use device tree overlay to allocate the buffer. See the Readme.md for more details.

rajatkha, 2021-11-17:
Thanks for replying.
I tried with the above method but facing this error on executing echo 'create udmabuf1 256' > /dev/u-dma-buf-mgr
-bash: /dev/u-dma-buf-mgr: Permission denied

Any solutions to this?

ikwzm, 2021-11-17:
Does ```/dev/u-dma-buf-mrg``` exist?

What is the permission for ```/dev/u-dma-buf-mgr```