ikwzm/udmabuf #82: bug in sync_for_cpu sample in documentation?

szymonkowalewski, 2022-01-10:
Hi,

In Readme.md in section https://github.com/ikwzm/udmabuf#sync_for_cpu it states that:

`If '1' is written to device file, if sync_direction is 2(=DMA_FROM_DEVICE) or 0(=DMA_BIDIRECTIONAL), the write to the device file invalidates a cache specified by sync_offset and sync_size.`

but in both code samples

**unsigned long  sync_for_cpu   = 1;**

I expect this is just a type in the code sample right?

ikwzm, 2022-01-10:
No, it's not a typo.
Whenever you run sync_for_cpu, you must write a non-zero value. 
Therefore, 1 is set in sync_for_cpu to make the least significant bit 1.

szymonkowalewski, 2022-01-10:
Sorry I actually copied wrong line, I meant:

**unsigned int   sync_direction = 1;**

I'd expect direction to be either 2 or 0 based on the description.

ikwzm, 2022-01-10:
As you pointed out, it is confusing as a sample code, so I corrected it.

closed 2022-01-10 by ikwzm