ikwzm/udmabuf #83: include file error

njsys, 2022-01-14:
I just got the latest source and unzipped it and tried to build it. I get the error below. 
What is the correct way of building this? Thanks

[root@localhost tools]# cd udmabuf-3.2.4/
[root@localhost udmabuf-3.2.4]# make u-dma-buf
cc     u-dma-buf.c   -o u-dma-buf
u-dma-buf.c:31:24: fatal error: linux/cdev.h: No such file or directory
 #include <linux/cdev.h>
                        ^

ikwzm, 2022-01-14:
Thank you for the issue.

u-dma-buf is a kernel module, so you need linux-headers to compile it.
Please install linux-headers.

njsys, 2022-01-14:
thanks for your rapid response.

I have previously used udmabuf1.4* without build or runtime issues.
but I notice this build issue only with the latest udmabuf.

the machine does have /linux5*/include/linux/cdev.h

but somehow the makefile needs to be told where to find it?

ikwzm, 2022-01-14:
Please try ``` make``` instead of ```make u-dma-buf.o```.

njsys, 2022-01-14:
Then I get this error 

[root@cotsicp1 udmabuf-3.2.4]# make
make -C /lib/modules/5.6.9/build ARCH=x86_64 CROSS_COMPILE= M=/home/src/udmabuf-3.2.4 obj-m=u-dma-buf.o u-dma-buf.ko
make[1]: Entering directory `/home/src/linux-5.6.9'
  AR      /home/src/udmabuf-3.2.4/built-in.a
make[3]: *** No rule to make target `u-dma-buf.o', needed by `__build'.  Stop.
make[2]: *** [/home/src/udmabuf-3.2.4] Error 2
make[1]: *** [sub-make] Error 2
make[1]: Leaving directory `/home/src/linux-5.6.9'
make: *** [all] Error 2

ikwzm, 2022-01-14:
I guess /home/src/linux-5.6.9 is wrong. 
What directory is /home/src/linux-5.6.9?

njsys, 2022-01-16:
linux kernel 5.6.9 was built from source tar file(in folder /home/src/linux-5.6.9) and installed on this computer.
cdev.h does exist in some subdirectory of /home/src/linux-5.6.9

the same folder has also been copied to /usr/src/kernels. But no difference to the udmabuf build errors.

njsys, 2022-01-16:
i had to change makefile content to this and then it worked....

HOST_ARCH       ?= $(shell uname -m | sed -e s/arm.*/arm/ -e s/aarch64.*/arm64/)
ARCH            ?= $(shell uname -m | sed -e s/arm.*/arm/ -e s/aarch64.*/arm64/)
KERNEL_SRC_DIR  ?= /lib/modules/$(shell uname -r)/build

ifeq ($(ARCH), arm)
 ifneq ($(HOST_ARCH), arm)
   CROSS_COMPILE  ?= arm-linux-gnueabihf-
 endif
endif
ifeq ($(ARCH), arm64)
 ifneq ($(HOST_ARCH), arm64)
   CROSS_COMPILE  ?= aarch64-linux-gnu-
 endif
endif

ccflags-y = -fno-stack-protector -lssp
obj-m := u-dma-buf.o

ifndef UDMABUF_MAKE_TARGET
  KERNEL_VERSION_LT_5 ?= $(shell awk '/^VERSION/{print int($$3) < 5}' $(KERNEL_SRC_DIR)/Makefile)
  ifeq ($(KERNEL_VERSION_LT_5), 1)
    UDMABUF_MAKE_TARGET ?= modules
  else
    UDMABUF_MAKE_TARGET ?= modules
  endif
endif

all:
        make -C $(KERNEL_SRC_DIR) ARCH=$(ARCH) CROSS_COMPILE=$(CROSS_COMPILE) M=$(PWD) $(UDMABUF_MAKE_TARGET)

clean:
        make -C $(KERNEL_SRC_DIR) ARCH=$(ARCH) CROSS_COMPILE=$(CROSS_COMPILE) M=$(PWD) clean

ikwzm, 2022-01-16:
I'm sorry I couldn't help you.

In my environment it looks like this:

```console
ichiro@ubuntu:~/work/udmabuf$ uname -a
Linux ubuntu 5.4.0-90-generic #101~18.04.1-Ubuntu SMP Fri Oct 22 09:25:04 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
ichiro@ubuntu:~/work$ git clone https://github.com/ikwzm/udmabuf
Cloning into 'udmabuf'...
remote: Enumerating objects: 802, done.        
remote: Counting objects: 100% (128/128), done.        
remote: Compressing objects: 100% (94/94), done.        
remote: Total 802 (delta 64), reused 97 (delta 34), pack-reused 674        
Receiving objects: 100% (802/802), 377.40 KiB | 2.27 MiB/s, done.
Resolving deltas: 100% (461/461), done.
ichiro@ubuntu:~/work$ cd udmabuf
ichiro@ubuntu:~/work/udmabuf$ make
make -C /lib/modules/5.4.0-90-generic/build ARCH=x86_64 CROSS_COMPILE= M=/home/ichiro/work/udmabuf obj-m=u-dma-buf.o u-dma-buf.ko
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-90-generic'
  CC [M]  /home/ichiro/work/udmabuf/u-dma-buf.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC [M]  /home/ichiro/work/udmabuf/u-dma-buf.mod.o
  LD [M]  /home/ichiro/work/udmabuf/u-dma-buf.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-90-generic'
```

```console
fpga@debian-fpga:~/work$ uname -a
Linux debian-fpga 5.10.0-xlnx-v2021.1-zynqmp-fpga #2 SMP Fri Nov 5 07:40:48 JST 2021 aarch64 GNU/Linux
fpga@debian-fpga:~/work$ git clone https://github.com/ikwzm/udmabuf
Cloning into 'udmabuf'...
remote: Enumerating objects: 802, done.
remote: Counting objects: 100% (128/128), done.
remote: Compressing objects: 100% (94/94), done.
remote: Total 802 (delta 64), reused 97 (delta 34), pack-reused 674
Receiving objects: 100% (802/802), 377.40 KiB | 8.78 MiB/s, done.
Resolving deltas: 100% (461/461), done.
fpga@debian-fpga:~/work$ cd udmabuf
fpga@debian-fpga:~/work/udmabuf$ make
make -C /lib/modules/5.10.0-xlnx-v2021.1-zynqmp-fpga/build ARCH=arm64 CROSS_COMPILE= M=/home/fpga/work/udmabuf obj-m=u-dma-buf.o u-dma-buf.ko
make[1]: Entering directory '/usr/src/linux-headers-5.10.0-xlnx-v2021.1-zynqmp-fpga'
  CC [M]  /home/fpga/work/udmabuf/u-dma-buf.o
  MODPOST /home/fpga/work/udmabuf/Module.symvers
  CC [M]  /home/fpga/work/udmabuf/u-dma-buf.mod.o
  LD [M]  /home/fpga/work/udmabuf/u-dma-buf.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.10.0-xlnx-v2021.1-zynqmp-fpga'
```

njsys, 2022-01-16:
thanks. 
Closing issue as we have a way to make it work after modifying the Makefile.

closed 2022-01-16 by njsys