ikwzm/udmabuf #87: support for microblaze ?

doonny, 2022-03-19:
Hi, this is a very useful tool.  Is it possible to add support for microblaze processor ?

ikwzm, 2022-03-19:
Thank you for the issue.

I made u-dma-buf for arm or arm64. 
I have never tried to work with microbraze prossor.

doonny, 2022-03-22:
hi @ikwzm thx for the reply.   I have successfully compiled the udmabuf driver for microblaze processor using petalinux 2021.2.
The driver can be successfully loaded into kernel:

root@vc709_plnx_21:~# insmod u-dma-buf.ko udmabuf0=1048576
u-dma-buf u-dma-buf.0: driver probe start.
u-dma-buf udmabuf0: driver version = 3.2.5
u-dma-buf udmabuf0: major number   = 250
u-dma-buf udmabuf0: minor number   = 0
u-dma-buf udmabuf0: phys address   = 0xaed00000
u-dma-buf udmabuf0: buffer size    = 1048576
u-dma-buf u-dma-buf.0: driver installed.

and I could observe from /proc/meminfo that CMA space has also been consumed to allocate the buffer.

But I could not find the dev node under /dev:

root@vc709_plnx_21:~# ls /dev/u*
/dev/urandom

Could give some advice what might be the problem ?

doonny, 2022-03-23:
@ikwzm thx for the advice, after enable udev in petalinux, the u-dma-buf module can be loaded upon bootup:
```
INIT: version 2.97 booting
mount: mounting sysfs on /sys failed: Device or resource busy
/etc/rcS.d/S04mdev: line 10: /proc/sys/kernel/hotplug: No such file or directory
Starting udev
udevd[83]: starting version 3.2.9
random: udevd: uninitialized urandom read (16 bytes read)
random: udevd: uninitialized urandom read (16 bytes read)
random: udevd: uninitialized urandom read (16 bytes read)
udevd[84]: starting eudev-3.2.9
u_dma_buf: loading out-of-tree module taints kernel.
u-dma-buf udmabuf@0x00: driver probe start.
u-dma-buf udmabuf0: driver version = 3.2.5
u-dma-buf udmabuf0: major number   = 251
u-dma-buf udmabuf0: minor number   = 0
u-dma-buf udmabuf0: phys address   = 0xa7d00000
u-dma-buf udmabuf0: buffer size    = 67108864
u-dma-buf udmabuf@0x00: driver installed.
bootlogd: /dev/ttyUL0Fri Mar  9 12:34:56 UTC 2018
```
But still could not find it under /dev:
```
root@vc709_plnx_21:~# ls /dev/u*
/dev/urandom
```

There are some error in the boot logs after adding udev to petalinux, not sure if it was the cause:
```
u-dma-buf udmabuf0: phys address   = 0xa7d00000
u-dma-buf udmabuf0: buffer size    = 67108864
u-dma-buf udmabuf@0x00: driver installed.
bootlogd: /dev/ttyUL0Fri Mar  9 12:34:56 UTC 2018
sh: malloc.c:2394: sysmalloc: Assertion `(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pa.
/etc/rcS.d/S37populate-volatile.sh: line 154:   198 Done                    sed 's/#.*//' "${CFGFILE}"
       199 Aborted                 | while read -r TTYPE TUSER TGROUP TMODE TNAME TLTARGET; do
    test -z "${TLTARGET}" && continue; TNAME=${ROOT_DIR}${TNAME}; [ "${VERBOSE}" != "no" ] && echo "Checking for -${TNAME}-."; [ "${TTYPE}" = "l" ] && { TSOURCE="$TLTARGET"; [ "${VERBOSE}"
        mount --bind \"${TSOURCE}\" \"${TNAME}\""; test "$VOLATILE_ENABLE_CACHE" = yes && echo "$EXEC" >> /etc/volatile.cache.build; continue; }; [ -L "${TNAME}" ] && { [ "${VERBOSE}" != n
        TNAME="$(echo "${TNAME}" | sed -e 's@\(.*\)/.*@\1@')/${NEWNAME}"; [ "${VERBOSE}" != "no" ] && echo "Converted relative linktarget to absolute path -${TNAME}-.";
    else
        TNAME="${NEWNAME}"; [ "${VERBOSE}" != "no" ] && echo "Using absolute link target -${TNAME}-.";
    fi; }; case "${TTYPE}" in
        "f")
            [ "${VERBOSE}" != "no" ] && echo "Creating file -${TNAME}-."; TSOURCE="$TLTARGET"; [ "${TSOURCE}" = "none" ] && TSOURCE=""; create_file "${TNAME}" "${TSOURCE}" &
        ;;
        "d")
            [ "${VERBOSE}" != "no" ] && echo "Creating directory -${TNAME}-."; mk_dir "${TNAME}"
        ;;
        *)
            [ "${VERBOSE}" != "no" ] && echo "Invalid type -${TTYPE}-."; continue
        ;;
    esac;
done
urandom_read: 1 callbacks suppressed
random: dd: uninitialized urandom read (512 bytes read)
Configuring packages on first boot....
 (This may take several minutes. Please do not power off the machine.)

```

doonny, 2022-03-23:
After changing selecting the whole eudev package rather only selecting libudev, the errors was gone in the boot log.
But still could not see the udmabuf device under /dev/

ikwzm, 2022-03-23:
I'm sorry I can't be of any help.
I'm not familiar with eudev.
You may want to check the debug information with ```udevadm monitor```.
Please let us know if you have any new information.

doonny, 2022-03-23:
hi @ikwzm ,  following are the messages printed when add and move the drivier :
```
root@vc709_plnx_21:~# udevadm monitor
monitor will print the received events for:
UDEV - the event which udev sends out after rule processing
KERNEL - the kernel uevent

KERNEL[818.755730] remove   /devices/virtual/misc/u-dma-buf-mgr (misc)
KERNEL[818.803936] remove   /devices/platform/u-dma-buf.0/u-dma-buf/udmabuf0 (u-dma-buf)
KERNEL[818.809830] unbind   /devices/platform/u-dma-buf.0 (platform)
KERNEL[818.816036] remove   /devices/platform/u-dma-buf.0 (platform)
KERNEL[818.829952] remove   /bus/platform/drivers/u-dma-buf (drivers)
KERNEL[818.835897] remove   /class/u-dma-buf (class)
KERNEL[818.838346] remove   /module/u_dma_buf (module)
KERNEL[832.295301] add      /module/u_dma_buf (module)
KERNEL[832.336945] add      /class/u-dma-buf (class)
KERNEL[832.350402] add      /bus/platform/drivers/u-dma-buf (drivers)
KERNEL[832.356573] add      /kernel/software_nodes/node1 (software_nodes)
KERNEL[832.359202] add      /devices/platform/u-dma-buf.0 (platform)
KERNEL[832.416661] add      /devices/platform/u-dma-buf.0/u-dma-buf/udmabuf0 (u-dma-buf)
KERNEL[832.423696] bind     /devices/platform/u-dma-buf.0 (platform)
KERNEL[832.511984] add      /devices/virtual/misc/u-dma-buf-mgr (misc)

```
Is the information sufficient for you

ikwzm, 2022-03-24:
In ZynqMP-FPGA-Linux, the output is as follows.

```console
root@debian-fpga:/home/fpga# udevadm monitor
monitor will print the received events for:
UDEV - the event which udev sends out after rule processing
KERNEL - the kernel uevent

KERNEL[1989.940970] add      /module/u_dma_buf (module)
KERNEL[1989.942721] add      /class/u-dma-buf (class)
KERNEL[1989.942789] add      /bus/platform/drivers/u-dma-buf (drivers)
KERNEL[1989.942827] add      /kernel/software_nodes/node5 (software_nodes)
KERNEL[1989.942872] add      /devices/platform/u-dma-buf.0 (platform)
UDEV  [1989.943530] add      /module/u_dma_buf (module)
UDEV  [1989.944906] add      /bus/platform/drivers/u-dma-buf (drivers)
UDEV  [1989.944986] add      /class/u-dma-buf (class)
KERNEL[1989.945735] add      /devices/platform/u-dma-buf.0/u-dma-buf/udmabuf0 (u-dma-buf)
UDEV  [1989.947321] add      /devices/platform/u-dma-buf.0 (platform)
UDEV  [1989.948651] add      /kernel/software_nodes/node5 (software_nodes)
UDEV  [1989.950287] add      /devices/platform/u-dma-buf.0/u-dma-buf/udmabuf0 (u-dma-buf)
KERNEL[1989.979179] bind     /devices/platform/u-dma-buf.0 (platform)
KERNEL[1989.979285] add      /devices/virtual/misc/u-dma-buf-mgr (misc)
UDEV  [1989.980528] bind     /devices/platform/u-dma-buf.0 (platform)
UDEV  [1989.980794] add      /devices/virtual/misc/u-dma-buf-mgr (misc)
```

As you can see, there is a UDEV line. The UDEV line shows the udev event handler that has completed.

On the other hand, your log does not have a UDEV line. I guess this is because the udev event is not happening in some settings. Check udev settings.

closed 2022-07-06 by doonny