ikwzm/udmabuf #88: dma_alloc_coherent(size=16777216) failed. return(0)

azAghaee, 2022-07-19:
Hi,
I'm trying to use u-dma-buffer on my ultra96 board.
here is my device tree 
/include/ "system-conf.dtsi"
/ {
	reserved-memory {
		#address-cells = <2>;
		#size-cells = <2>;
		ranges;
		image_buf0: image_buf@0 {
			compatible = "shared-dma-pool";
			reusable;
			reg = <0x0 0x20000000 0x0 0x01000000>; 
			linux,cma-default;
			label = "image_buf0";
		};
	};
	udmabuf@0 {
		compatible = "ikwzm,u-dma-buf";
		device-name = "udmabuf0";
		size = <0x0 0x01000000>; // 16MiB
		memory-region = <&image_buf0>;
	};
};

below is part of linux boot report
[image]
[image]
linux version is 4.14
I read whole the related issues but can't fix the problem.
hope you can guide me.

ikwzm, 2022-07-19:
How a physical address is mapped to a virtual address depends on the kernel.
For example, the Linux Kernel for ARM64 has a parameter called CONFIG_ARM64_VA_BITS that specifies the number of bits allocated to a virtual address.
Therefore, I'm not sure if the question address is correct.

azAghaee, 2022-07-19:
Thanks for your answer
In fact I knew about that but when I give that address to Xilinx VDMA core, it return an invalid address!
I can read the address content. perhaps this is an VDMA issue.
Thanks for your hint and driver

ikwzm, 2022-07-19:
The virtual address returned by mmap() is used by the CPU to access the buffer. 
You must specify the physical address for DMA. 
Try specifying the value of ```/sys/class/u-dma-buf/udmabuf0/phys_addr```  for VDMA.

azAghaee, 2022-07-20:
After I replaced the physical address, problem solved.
Your help meant a lot to me.
Thank you for your time and precious guidance.

closed 2022-09-10 by ikwzm