ikwzm/udmabuf #94: Zynq Ultrascale+ (ARM Cortex A-53) Kernel 5.15.19 problem

oguzhancik, 2022-10-04:
Everything compiled good with Petalinux kernel 5.15.19. I'm also creating device tree with dma-coherent property as shown below.

	dma_data_buffer@0x00{
		compatible = "ikwzm,u-dma-buf";
		device-name = "dma_data_buffer";
		size = <0x09400000>;
		dma-coherent; 	
	};

When i try to use it with succesfully working user DMA software with previous kernel 5.10 everything works well. However, with this 5.15.19 kernel it doesn't work. I suspect that it is caused by cache coherency because i'm using hw coherent bus of MPSOC and hardware flush or invalidates the cache over cache coherent bus. Can you help for this issue? I think proper support fr 5.15.19 will fix this issue.

ikwzm, 2022-10-05:
thank you for this precious issue.

Please give me a little more time.
I'm not using Linux Kernel 5.15.x yet.

I would like to ask you some questions
What exactly do you mean by "it doesn't work"?

oguzhancik, 2022-10-05:
I use Xilinx AXI DMA with scatter gather engine from user space. I create all data and descriptor buffers with this module in the userspace. This space is cache coherent. In case of a problem with cache coherency it doesn't trigger when I send the data. It works well with previous kernel version 5.10.

ikwzm, 2022-10-06:
Thanks for your reply.

Let me ask you a few more questions.

Q1. How did you build the Linux Kernel? Petalinux? Yocto? Other? 　

Q2. What is the output of dmesg when u-dma-buf is installed? I especially want to know the u-dma-buf version and physical address.

Sorry for the inconvenience, but thank you in advance.

oguzhancik, 2022-10-09:
Hello 
I'm sharing the answers below.

Q1. I build using Petalinux.
Q2. Here is the info for u-dma-buf

[   11.083838] u-dma-buf dma_data_buffer: driver version = 4.0.0
[   11.083873] u-dma-buf dma_data_buffer: major number   = 237
[   11.083880] u-dma-buf dma_data_buffer: minor number   = 0
[   11.083886] u-dma-buf dma_data_buffer: phys address   = 0x000000005d900000
[   11.083894] u-dma-buf dma_data_buffer: buffer size    = 155189248
[   11.083903] u-dma-buf dma_data_buffer@0x00: driver installed.
[   11.084180] u-dma-buf dma_rx_desc_buffer@0x00: driver probe start.
[   11.156243] u-dma-buf dma_rx_desc_buffer: driver version = 4.0.0
[   11.156282] u-dma-buf dma_rx_desc_buffer: major number   = 237
[   11.156288] u-dma-buf dma_rx_desc_buffer: minor number   = 1
[   11.156294] u-dma-buf dma_rx_desc_buffer: phys address   = 0x0000000066d00000
[   11.156303] u-dma-buf dma_rx_desc_buffer: buffer size    = 25165824
[   11.156313] u-dma-buf dma_rx_desc_buffer@0x00: driver installed.
[   11.156681] u-dma-buf mov_ul_frame_buffer@0x00: driver probe start.
[   11.237942] u-dma-buf mov_ul_frame_buffer: driver version = 4.0.0
[   11.237975] u-dma-buf mov_ul_frame_buffer: major number   = 237
[   11.237982] u-dma-buf mov_ul_frame_buffer: minor number   = 2
[   11.237987] u-dma-buf mov_ul_frame_buffer: phys address   = 0x0000000068500000
[   11.237994] u-dma-buf mov_ul_frame_buffer: buffer size    = 155189248
[   11.238004] u-dma-buf mov_ul_frame_buffer@0x00: driver installed.