ikwzm/udmabuf #96: The reserved space is used to expand the CMA space, and the u-dma-buf driver cannot be used normally

Noah-naihe, 2022-10-28:
I'm sorry, I have one more question here; Here I refer to xilinx's cma space extension for a link:https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18841683/Linux+Reserved+Memory#LinuxReservedMemory-ReservedmemoryforCMA
The cma space was also successfully created.
[image]
Using this udmabuf driver before, it is normal to open up space for multiple Cmas. However, after expanding the cma space, it can not be used normally

[image]

device tree:
	reserved-memory {
     		#address-cells = <2>;
      		#size-cells = <2>;
      		ranges;
  
      		reserved: buffer@0 {
         		compatible = "shared-dma-pool";
         		reusable;
         		reg = <0x00000008 0x00000000 0x0 0x80000000>;
         		linux,cma-default;
      		};
   	};

Noah-naihe, 2022-10-28:
I refer to what you said, add the following device tree node in the device tree, but found that the compiler does not pass. I am using kernel 5.4.0, what is the impact of this
[image]

ikwzm, 2022-10-28:
Thank you for the issue.

The first half of the problem may be cured by specifying the dma-mask property. See this issue #45 for details.

Please tell me what kind of error occurred in the latter half of the problem.

Noah-naihe, 2022-10-31:
I use 16G DDR. According to xilinx system memory allocation, it is 2+14GB allocation, among which 2G is in 0x0-0x80000000 and the other 14GB is in 0x8 00000000. However, in the device tree, the CMA sets the start address at 0x8 00000000 and the length is 2GB, but the physical address is still 2GB ahead.
[image]
Allocating 100MB of space also fails
The 100MB space cannot be allocated successfully. The previous 1MB space can be allocated. So it's not the number of digits in the system.
[image]

ikwzm, 2022-10-31:
u-dma-buf-mgr allocates memory from the default CMA. 
As far as this log is concerned, the default CMA area seems to be below 0x80000000 (2GBi). 
What are your CMA area settings?

noelpedro, 2023-05-05:
1GiB worked for me I allocated 1250MB of CMA and had the reusable flag removed.