ikwzm/udmabuf #102: In Python test code there is a typo

sapertuz, 2023-04-21:
in the python test code:
```python

from udmabuf import Udmabuf
import numpy as np
import time
def test_1(a):
    for i in range (0,9):
        a *= 0
        a += 0x31
if __name__ == '__main__':
    udmabuf      = Udmabuf('udmabuf0')
    test_dtype   = np.uint8
    test_size    = udmabuf.buf_size/(np.dtype(test_dtype).itemsize)
    udmabuf.memmap(dtype=test_dtype, shape=(test_size))
    comparison   = np.zeros(test_size, dtype=test_dtype)
    print ("test_size  : %d" % test_size)
    start        = time.time()
    test_1(udmabuf.mem_map)
    elapsed_time = time.time() - start
    print ("udmabuf-0   : elapsed_time:{0}".format(elapsed_time)) + "[sec]"
    start        = time.time()
    test_1(comparison)
    elapsed_time = time.time() - start
    print ("comparison : elapsed_time:{0}".format(elapsed_time)) + "[sec]"
    if np.array_equal(udmabuf.mem_map, comparison):
        print ("udmabuf-0 == comparison : OK")
    else:
        print ("udmabuf-0 != comparison : NG")
```
in 
```python
    test_1(udmabuf.mem_map)
```
the attribute ```udmabuf.mem_map``` does not exist and the code doesn't work. am I doing something wrong?

sapertuz, 2023-04-21:
I solved it... the code should call the attribute ```udmabuf.array``` instead.
There were also some code typos that I fixed here:
```python
from udmabuf import Udmabuf
import numpy as np
import time
def test_1(a):
    for i in range (0,9):
        a[i] *= 0
        a[i] += 0x31
if __name__ == '__main__':
    udmabuf      = Udmabuf('udmabuf0')
    test_dtype   = np.uint8
    test_size    = udmabuf.buf_size//(np.dtype(test_dtype).itemsize)
    comparison   = np.zeros(test_size, dtype=test_dtype)
    udmabuf.memmap(dtype=test_dtype, shape=(test_size))
    print ("test_size  : %d" % test_size)
    start        = time.time()
    test_1(udmabuf.array)
    elapsed_time = time.time() - start
    print ("udmabuf-0   : elapsed_time: {0}".format(elapsed_time) + " sec" )
    start        = time.time()
    test_1(comparison)
    elapsed_time = time.time() - start
    print ("comparison : elapsed_time:  {0}".format(elapsed_time) + " sec")
    if np.array_equal(udmabuf.array, comparison):
        print ("udmabuf-0 == comparison : OK")
    else:
        print ("udmabuf-0 != comparison : NG")
```

I wonder however if this is correct... in the ```test_1`` it didn't increment in the array. I added it, however... I am also curious about why the range of ```i``` is only from 0 to 9... should it not be 0 to ```test_size-1``` ?

Btw, feel free to use the code to be included in your readme file :)

ikwzm, 2023-04-21:
Thank you for the issue.

Readme.md was corrected.

ikwzm, 2023-04-22:
> I wonder however if this is correct... in the `test_1` it didn't increment in the array. I added it, however... I am also curious about why the range of iis only from 0 to 9... should it not be 0 `totest_size-1` ?

The argument `a` of `test_1` is a numpy array.
Thus, `a *= 0` clears the entire contents of the array, and `a += 0x31` adds 0x31 to each element of the array.
This is repeated 10 times and the time it takes is measured.

closed 2023-04-22 by ikwzm