ikwzm/udmabuf #103: memory-region

noelpedro, 2023-04-28:
Hi Ichiro,

I currently have 1GB of DDR shared with the PL. And I'm currently bottle necked by the read speeds of using /dev/mem to read from this memory.

Can I expect an improvement in read speeds by using the u-dma-buf memory-region configuration? Its seems like by using CMA for the area I will definitely improve read speeds.

ikwzm, 2023-04-29:
If reading from memory is slow, the reason may be that the data cache is disabled.

When reading from /dev/mem with mmap(), the data cache is disabled if the O_SYNC flag is specified when opening; opening without the O_SYNC flag enables the data cache.
Whether the data cache can be enabled or must be disabled depends on whether cache coherency is guaranteed by the hardware.
If cache coherency is guaranteed in hardware, then the data cache can be enabled. However, if cache coherency is not guaranteed by the hardware, the data cache must be disabled.

The above also applies to the use of u-dma-buf.

noelpedro, 2023-05-03:
How about using udmabuf in python's memmap is there an O_SYNC flag enable/disbale ?

ikwzm, 2023-05-04:
Please refer to the following repositories.

* https://github.com/ikwzm/ArgSort-Kv260

noelpedro, 2023-05-05:
The dma-coherent flag for my udmabuf really sped up my memory reads.  I also connected the ACP in the PL.

closed 2023-05-10 by noelpedro