ikwzm/udmabuf #107: Could this be used with V4L2/libcamera buffers on the Raspberry Pi 4 (Arm A72)

octopus-russell, 2023-06-13:
Hi,
We've come across this driver as a potential way of passing a userspace dma buffer to V4L2 instead of V4L2's default mmap mode which is rather slow. Here I see someone's done this achieving a 15x speedup: https://github.com/ikwzm/udmabuf/issues/38
Do you know if this module supports the Raspberry Pi 4? (ARM A72, Debian bullseye, kernel 6.1.21)
Thanks
Russell

ikwzm, 2023-06-13:
Thanks for the issue.

I have only run it on ARM Cortex®-A53 (Xilinx Zynq Ultrascale+ MPSoC) and ARM Cortex®-A9 (Xilinx ZYNQ / Altera CycloneV SoC), I don't know if it works on ARM Cortex®-A72 (Raspberry Pi 4).

It may work on the ARM Cortex®-A72 (Raspberry Pi 4) since it has the same arm64 architecture as the A53.

Please someone give me some information.

kbingham, 2023-07-21:
udmabuf likely isn't a good way to pass the buffers, but if you're experiencing issues with mmap'ing buffers indeed it's because they are likely in uncached memory.

ArmandBENETEAU, 2023-09-08:
Hi all!

Firstly, I want to sincerely say thank you to all the contributors to this project and more specifically to @ikwzm. It seems to be a really dynamic project, where all the issue have answers. This is great! 

I am in a situation a bit like @octopus-russell. Indeed, I want to capture images from a camera using v4l2 and I need to write them at a given address in **physical memory**.  Several solutions exist of course, but it seems from my point of view that using DMA is the most optimal way. In v4l2 it corresponds to the `V4L2_MEMORY_DMABUF` method.

So I've entered the dark world of DMA in Linux. After several hours/days of research around the web, I've come across this device-driver and I thought for one glorious second that I've had found the right way. But after several attempts, I've finally find out that I cannot export dma-buffer file descriptors using `u-dma-buf`...

Then, during another day I've tried to find a tool that allow me to export a dma-buffer file descriptor from a physical memory address... in vain. And a bit randomly, I encounter this issue, that is really close from what I want to do! Thus, may be that I can find the solution here.

Indeed @kbingham you said that _"udmabuf likely isn't a good way to pass the buffers"_ and I'm wondering if you could give me a hint on how to do what I want? I.e: using `V4L2_MEMORY_DMABUF` method to write directly at a given address in physical memory.

Sorry for this looooong text, and have a good day!

PS1: unfortunately since December 2022, the strategy consisting in using `V4L2_MEMORY_USERPTR` method with `u-dma-buf` cannot work due to [this commit](https://lore.kernel.org/lkml/20221206094916.987259-13-sashal@kernel.org/)

PS2: I am aware that may be this is not the right place to ask that, if it is the case, could you redirect me to the right place?

ikwzm, 2023-09-08:
Thank you for your valuable information.

> PS1: unfortunately since December 2022, the strategy consisting in using V4L2_MEMORY_USERPTR method with u-dma-buf cannot work due to [this commit](https://lore.kernel.org/lkml/20221206094916.987259-13-sashal@kernel.org/)

I did not know that the V4L2_MEMORY_USERPTR method was no longer available. It would be a shame if it is no longer available.

This is not yet a decision, but I am currently trying to add the ability to export u-dma-bufs as dma-bufs.
It is currently under development in this branch, but unfortunately it is not working yet. 
I can export, but when I try to remove u-dma-buf from the kernel, the kernel panics.

https://github.com/ikwzm/udmabuf/tree/dma-buf-export-develop

I still have a long way to go, but I will make it public when it is ready.

ArmandBENETEAU, 2023-09-11:
You are welcome. It is still available but, as far as I understand, v4l2 refuses to use it if it implicates in the end direct writing to physical memory address.

Wow, having `u-dma-bufs` as dma-bufs would be great indeed! I hope you will succeed making it work.

Thank you for your answer.