ikwzm/udmabuf #109: AXI MCDAM support?

Livius90, 2023-07-15:
Hi,

Is the [AXI Multichannel DMA](https://www.xilinx.com/products/intellectual-property/axi_mcdma.html) supported to use with u-dma-buf?

> The maximum number of DMA buffers that can be allocated using insmod is 8 (udmabuf0/1/2/3/4/5/6/7).
What does it mean? Maximum 8 DMA buffer is limited only, if it is configured via insmod? What is the maximum if they are defined in device-tree nodes instead of insmod?

ikwzm, 2023-07-16:
Thank you for the issue.

I am not familiar with AXI MCDMA.
u-dma-buf only provides DMA buffers, it does not control DMA. 
There is probably a dedicated Linux Driver for AXI MCDMA. 
Please contact AMD for more information.

I'm sorry I couldn't help you.

Livius90, 2023-07-16:
Hi,

I looked after what is the different between AXI DMA and AXI MCDMA, lets see the following:
- [AXI DMA](https://docs.xilinx.com/viewer/book-attachment/ePquvyIHSl7mKfi0ecEn4Q/CPPzqKuxCU1Q4a3rXOA1jw) supported modes:
    - Optional Independent Scatter/Gather Direct Memory Access (DMA) support
    - Optional Direct Register Mode (no scatter/gather support)
- [AXI MCDMA](https://docs.xilinx.com/v/u/en-US/pg288-axi-mcdma) supported modes:
    - Scatter/Gather support (only)
    
As i understand your u-dma-buf kernel module, it support only the "Direct Register Mode" and "Scatter/Gather" is not supported, am i rigth? In this case AXI MCDMA is not usable via u-dma-buf unofrunately. :cry:

Livius90, 2023-07-16:
@ikwzm 
> The maximum number of DMA buffers that can be allocated using insmod is 8 (udmabuf0/1/2/3/4/5/6/7).

What does it mean? Maximum 8 DMA buffer is limited only, if it is configured via insmod? What is the maximum if they are defined in device-tree nodes instead of insmod?

ikwzm, 2023-07-16:
> What does it mean? Maximum 8 DMA buffer is limited only, if it is configured via insmod? What is the maximum if they are defined in device-tree nodes instead of insmod?

256

See u-dma-buf.c for details.

Livius90, 2023-07-16:
> > What does it mean? Maximum 8 DMA buffer is limited only, if it is configured via insmod? What is the maximum if they are defined in device-tree nodes instead of insmod?
> 
> 256
> 
> See u-dma-buf.c for details.

Thanks, i found it in the code:
https://github.com/ikwzm/udmabuf/blob/master/u-dma-buf.c#L72

Livius90, 2023-07-17:
Thanks the detailed quick answers, your u-dma-buf is awesome.

closed 2023-07-17 by Livius90