ikwzm/udmabuf #111: Using udmabuf on linux zynq7020

lone-boy, 2023-08-12:
Hello, I want to use udmabuf for dma like function. The ddr on the PL side allocates a section of space, the PL writes data into it, and then the PS section reads it.
But when I write device tree like this
reserved-memory {
		#address-cells = <2>;
		#size-cells = <2>;
		ranges;
		image_buf0: image_buf@0 {
			compatible = "shared-dma-pool";
			reusable;
			reg = <0x0 0x32000000 0x0 0x33000000>;
			alignment = <0x0 0x1000>;
			label = "image_buf0";
		};
	};
	
	udmabuf@0{
		compatible = "ikwzm,u-dma-buf";
		device-name = "udmabuf0";
		size = <0x01000000>;
		memory-region = <&image_buf0>;
	};
# dmesg | grep udma
u-dma-buf udmabuf@0: of_reserved_mem_device_init failed. return=-22
u-dma-buf udmabuf@0: driver probe failed. return=-22
u-dma-buf: probe of udmabuf@0 failed with error -22

ikwzm, 2023-08-12:
Thanks for the issue.

I assume reserved-memory initialization has failed, please check dmesg.

ikwzm, 2023-08-12:
Please see issue #45.

lone-boy, 2023-08-14:
That's really nice, thank you very much

closed 2023-08-14 by lone-boy