ikwzm/udmabuf #116: Xilinx Zynqmp:  The read data is inconsistent with the data written by the PL

zhanghongg, 2024-02-19:
The problem is sporadic. 
It seems to always happen at the last position of buf.

ikwzm, 2024-02-19:
Please give more information.

zhanghongg, 2024-02-20:
U-dma-buf devicetree:
	udmabuf@0x00 {
		compatible = "ikwzm,u-dma-buf";
		device-name = "udmabuf0";
		size = <0x14000000>;
	};
U-dma-buf initialization:
    fd_ = open("/dev/udmabuf0", O_RDWR);
    if (fd_ < 0) {   
        log_error("UdmaBuf mmap!");
        assert(false);
    }
    user_ptr_ = mmap64(NULL, map_size_, PROT_READ | PROT_WRITE, MAP_SHARED, fd_, 			base_addr_);
    if(user_ptr_ != MAP_FAILED) {
        log_info("UdmaBuf mmap succuffully!");
        ClearBuf();
    }
    else {
        log_error("UdmaBuf mmap failed!"); 
}
U-dma-buf ClearBuf:
    unsigned char* tmp = reinterpret_cast<unsigned char*>(user_ptr_);
memset(tmp,0,map_size_);

V4l2 Userptr REQBUFS:
    uint64_t reserved_base_addr = 0;
    reserved_memory_ = std::make_shared<UdmaBuf>(reserved_base_addr, req_count * 			buf_length_);

    struct v4l2_requestbuffers reqBufs;
    memset(&reqBufs, 0, sizeof(struct v4l2_requestbuffers));
    reqBufs.count  = req_count;
    reqBufs.type   = V4L2_BUF_TYPE_VIDEO_CAPTURE;
    reqBufs.memory = V4L2_MEMORY_USERPTR;
    if (ioctl(fd_, VIDIOC_REQBUFS, &reqBufs)) {
        log_error("v4l2 VIDIOC_REQBUFS failed: %s", strerror(errno));
        return false;
    }
    video_buf_.resize(req_count);
    dq_video_bufs_.resize(req_count);
    for(int i = 0; i < reqBufs.count; i++){
        struct v4l2_buffer videoBuf;
        memset(&videoBuf, 0, sizeof(struct v4l2_buffer));
        videoBuf.index = i;
        videoBuf.type = V4L2_BUF_TYPE_VIDEO_CAPTURE;
        videoBuf.memory = V4L2_MEMORY_USERPTR;
        if(ioctl(fd_, VIDIOC_QUERYBUF, &videoBuf)){
            log_error("v4l2 VIDIOC_QUERYBUF failed: %s", strerror(errno));
            return false;
        }
        void* myPtr = reserved_memory_->GetMappAddr();
        video_buf_[i] = reinterpret_cast<unsigned char*>(myPtr) + i*buf_length_;
        videoBuf.m.userptr = reinterpret_cast<unsigned long>(video_buf_[i]);

        if(ioctl(fd_, VIDIOC_QBUF, &videoBuf)){
            log_error("v4l2 VIDIOC_QBUF failed: %s", strerror(errno));
            return false;
        }
    }

    return true;

Program running logic and Problem describe:
 
PL will write a value of the last four bytes of buf, and the ps side will write the last four bytes of buf as 0 every time the ps side reads buf. The problem is that when ps obtain buf3,it occasionally reads out the last four byte values of 0, but the pl terminal has written it as a non-0 value.

zhanghongg, 2024-02-20:
U-dma-buf devicetree:
```
	udmabuf@0x00 {
		compatible = "ikwzm,u-dma-buf";
		device-name = "udmabuf0";
		size = <0x14000000>;
	};
```
U-dma-buf initialization:
```
    fd_ = open("/dev/udmabuf0", O_RDWR);
    if (fd_ < 0) {   
        log_error("UdmaBuf mmap!");
        assert(false);
    }
    user_ptr_ = mmap64(NULL, map_size_, PROT_READ | PROT_WRITE, MAP_SHARED, fd_, 			base_addr_);
    if(user_ptr_ != MAP_FAILED) {
        log_info("UdmaBuf mmap succuffully!");
        ClearBuf();
    }
    else {
        log_error("UdmaBuf mmap failed!"); 
}
```
U-dma-buf ClearBuf:
```
    unsigned char* tmp = reinterpret_cast<unsigned char*>(user_ptr_);
memset(tmp,0,map_size_);
```

V4l2 Userptr REQBUFS:
```
    uint64_t reserved_base_addr = 0;
    reserved_memory_ = std::make_shared<UdmaBuf>(reserved_base_addr, req_count * 			buf_length_);

    struct v4l2_requestbuffers reqBufs;
    memset(&reqBufs, 0, sizeof(struct v4l2_requestbuffers));
    reqBufs.count  = req_count;
    reqBufs.type   = V4L2_BUF_TYPE_VIDEO_CAPTURE;
    reqBufs.memory = V4L2_MEMORY_USERPTR;
    if (ioctl(fd_, VIDIOC_REQBUFS, &reqBufs)) {
        log_error("v4l2 VIDIOC_REQBUFS failed: %s", strerror(errno));
        return false;
    }
    video_buf_.resize(req_count);
    dq_video_bufs_.resize(req_count);
    for(int i = 0; i < reqBufs.count; i++){
        struct v4l2_buffer videoBuf;
        memset(&videoBuf, 0, sizeof(struct v4l2_buffer));
        videoBuf.index = i;
        videoBuf.type = V4L2_BUF_TYPE_VIDEO_CAPTURE;
        videoBuf.memory = V4L2_MEMORY_USERPTR;
        if(ioctl(fd_, VIDIOC_QUERYBUF, &videoBuf)){
            log_error("v4l2 VIDIOC_QUERYBUF failed: %s", strerror(errno));
            return false;
        }
        void* myPtr = reserved_memory_->GetMappAddr();
        video_buf_[i] = reinterpret_cast<unsigned char*>(myPtr) + i*buf_length_;
        videoBuf.m.userptr = reinterpret_cast<unsigned long>(video_buf_[i]);

        if(ioctl(fd_, VIDIOC_QBUF, &videoBuf)){
            log_error("v4l2 VIDIOC_QBUF failed: %s", strerror(errno));
            return false;
        }
    }

    return true;
```

Program running logic and Problem describe:
 
[image]

PL will write a value of the last four bytes of buf, and the ps side will write the last four bytes of buf as 0 every time the ps side reads buf. The problem is that when ps obtain buf3,it occasionally reads out the last four byte values of 0, but the pl terminal has written it as a non-0 value.

ikwzm, 2024-02-20:
How is Cache Coherency?

zhanghongg, 2024-02-21:
The code shown above is everything I do

zhanghongg, 2024-02-21:
Please provide some suggestions, thank you.

ikwzm, 2024-02-22:
> I tried using the method that "Manual cache management with the CPU cache still being enabled" in udmabuf readme.
The problem will still occur.

How did you do this?

ikwzm, 2024-02-22:
How many is the value of buf_length_ ?
What is the value of videoBuf.length ?

zhanghongg, 2024-02-22:
When CPU using buf, the following code is roughly executed:

`    
// set sync_offset
    unsigned char  attr[1024];
    unsigned long  sync_offset = 16777216* index;   // index: 0-19
    if ((fd  = open("/sys/class/u-dma-buf/udmabuf0/sync_offset", O_WRONLY)) != -1) {
        sprintf(attr, "%d", sync_offset); /* or sprintf(attr, "0x%x", sync_offset); */
        write(fd, attr, strlen(attr));
        close(fd);
    }
// set sync_size
    unsigned char  attr[1024];
    unsigned long  sync_size = 16777216;  
    if ((fd  = open("/sys/class/u-dma-buf/udmabuf0/sync_size", O_WRONLY)) != -1) {
        sprintf(attr, "%d", sync_size); /* or sprintf(attr, "0x%x", sync_size); */
        write(fd, attr, strlen(attr));
        close(fd);
    }

//  set sync_direction
    unsigned char  attr[1024];
    unsigned long  sync_direction = 1;
    if ((fd  = open("/sys/class/u-dma-buf/udmabuf0/sync_direction", O_WRONLY)) != -1) {
        sprintf(attr, "%d", sync_direction);
        write(fd, attr, strlen(attr));
        close(fd);
    }
// set sync_for_cpu
    unsigned char  attr[1024];
    unsigned long  sync_for_cpu = 1;
    if ((fd  = open("/sys/class/u-dma-buf/udmabuf0/sync_for_cpu", O_WRONLY)) != -1) {
        sprintf(attr, "%d", sync_for_cpu);
        write(fd, attr, strlen(attr));
        close(fd);
    }
// set sync_for_device
    unsigned char  attr[1024];
    unsigned long  sync_for_device = 0;
    if ((fd  = open("/sys/class/u-dma-buf/udmabuf0/sync_for_device", O_WRONLY)) != -1) {
        sprintf(attr, "%d", sync_for_device);
        write(fd, attr, strlen(attr));
        close(fd);
    }
`

zhanghongg, 2024-02-22:
When CPU using buf, the following code is roughly executed:

`// set sync_offset
unsigned char attr[1024];
unsigned long sync_offset = 16777216* index; // index: 0-19
if ((fd = open("/sys/class/u-dma-buf/udmabuf0/sync_offset", O_WRONLY)) != -1) {
sprintf(attr, "%d", sync_offset); /* or sprintf(attr, "0x%x", sync_offset); /
write(fd, attr, strlen(attr));
close(fd);
}
// set sync_size
unsigned char attr[1024];
unsigned long sync_size = 16777216;
if ((fd = open("/sys/class/u-dma-buf/udmabuf0/sync_size", O_WRONLY)) != -1) {
sprintf(attr, "%d", sync_size); / or sprintf(attr, "0x%x", sync_size); */
write(fd, attr, strlen(attr));
close(fd);
}

// set sync_direction
unsigned char attr[1024];
unsigned long sync_direction = 1;
if ((fd = open("/sys/class/u-dma-buf/udmabuf0/sync_direction", O_WRONLY)) != -1) {
sprintf(attr, "%d", sync_direction);
write(fd, attr, strlen(attr));
close(fd);
}
// set sync_for_cpu
unsigned char attr[1024];
unsigned long sync_for_cpu = 1;
if ((fd = open("/sys/class/u-dma-buf/udmabuf0/sync_for_cpu", O_WRONLY)) != -1) {
sprintf(attr, "%d", sync_for_cpu);
write(fd, attr, strlen(attr));
close(fd);
}
// set sync_for_device
unsigned char attr[1024];
unsigned long sync_for_device = 0;
if ((fd = open("/sys/class/u-dma-buf/udmabuf0/sync_for_device", O_WRONLY)) != -1) {
sprintf(attr, "%d", sync_for_device);
write(fd, attr, strlen(attr));
close(fd);
}`

ikwzm, 2024-02-22:
What I would like to know is not so much when, but when are you syncing?

zhanghongg, 2024-02-22:
When CPU using buf, the following code is roughly executed:
```

// set sync_offset
unsigned char attr[1024];
unsigned long sync_offset = 16777216* index; // index: 0-19
if ((fd = open("/sys/class/u-dma-buf/udmabuf0/sync_offset", O_WRONLY)) != -1) {
sprintf(attr, "%d", sync_offset); /* or sprintf(attr, "0x%x", sync_offset); /
write(fd, attr, strlen(attr));
close(fd);
}
// set sync_size
unsigned char attr[1024];
unsigned long sync_size = 16777216;
if ((fd = open("/sys/class/u-dma-buf/udmabuf0/sync_size", O_WRONLY)) != -1) {
sprintf(attr, "%d", sync_size); / or sprintf(attr, "0x%x", sync_size); */
write(fd, attr, strlen(attr));
close(fd);
}

// set sync_direction
unsigned char attr[1024];
unsigned long sync_direction = 1;
if ((fd = open("/sys/class/u-dma-buf/udmabuf0/sync_direction", O_WRONLY)) != -1) {
sprintf(attr, "%d", sync_direction);
write(fd, attr, strlen(attr));
close(fd);
}
// set sync_for_cpu
unsigned char attr[1024];
unsigned long sync_for_cpu = 1;
if ((fd = open("/sys/class/u-dma-buf/udmabuf0/sync_for_cpu", O_WRONLY)) != -1) {
sprintf(attr, "%d", sync_for_cpu);
write(fd, attr, strlen(attr));
close(fd);
}
// set sync_for_device
unsigned char attr[1024];
unsigned long sync_for_device = 0;
if ((fd = open("/sys/class/u-dma-buf/udmabuf0/sync_for_device", O_WRONLY)) != -1) {
sprintf(attr, "%d", sync_for_device);
write(fd, attr, strlen(attr));
close(fd);
}

```

ikwzm, 2024-02-22:
Umm...
I do not know what you are doing.
Please show me the source code, not the text.

zhanghongg, 2024-02-22:
After obtaining the v4l2 buf, I executed the sync code above

ikwzm, 2024-02-22:
> After obtaining the v4l2 buf, I executed the sync code above

It does not work at all.

zhanghongg, 2024-02-22:
> What I would like to know is not so much when, but when are you syncing?

what means？

> It does not work at all.

why...

zhanghongg, 2024-02-23:
Thank you very much for your answer. The purpose of using udmabuf is to improve the speed of memcpy. What should I do now?

ikwzm, 2024-02-23:
I ask again.

How many is the value of buf_length_ ?
What is the value of videoBuf.length ?

zhanghongg, 2024-02-23:
Both are 16,777,216.

zhanghongg, 2024-02-23:
Thank you very much for your help!

> Why does the ps side write the last four bytes of buf as every time the ps side reads buf ?
The next time the PL end receives this buf, it will determine some logic based on this memory address

> Can you show me the source code for this part?
OK!

Main logic code:
```
    unsigned char* buf = v4l2fd_.DqBuf();
    ...
    ... /// use this buf
    ...
    uint64_t* countBuf = reinterpret_cast<uint64_t*>(buf + stripBufSize - sizeof(uint64_t));
    *countBuf =0;
    v4l2fd_.QBuf();
```

```
unsigned char* DqBuf() {
    memset(&active_video_buf_, 0, sizeof(struct v4l2_buffer));
    active_video_buf_.type = V4L2_BUF_TYPE_VIDEO_CAPTURE;
    active_video_buf_.memory = V4L2_MEMORY_USERPTR;
    if(ioctl(fd_, VIDIOC_DQBUF, &active_video_buf_)){
        log_error("v4l2 VIDIOC_DQBUF failed: %s", strerror(errno));
        return 0;
    }
    reserved_memory_->Sync(buf_length_, active_video_buf_.index*buf_length_, 1);
    return video_buf_[active_video_buf_.index];
}
```

```
bool V4l2FdUserPtr::QBuf() {
    reserved_memory_->Sync(buf_length_, active_video_buf_.index*buf_length_, 0);
    if(ioctl(fd_, VIDIOC_QBUF, &active_video_buf_)){
        log_error("v4l2 VIDIOC_QBUF failed: %s", strerror(errno));
        return false;
    }
    return true;
}
```

```
void UdmaBuf::Sync(unsigned long sync_size, unsigned long sync_offset,
                   unsigned int sync_direction) {
    int fd;
    {
        unsigned char  attr[1024];
        if ((fd  = open("/sys/class/u-dma-buf/udmabuf0/sync_offset", O_WRONLY)) != -1) {
            sprintf((char *)attr, "%d", sync_offset); /* or sprintf(attr, "0x%x", sync_offset); */
            write(fd, attr, strlen((const char *)attr));
            close(fd);
        }
    }
    {
        unsigned char  attr[1024];
        if ((fd  = open("/sys/class/u-dma-buf/udmabuf0/sync_size", O_WRONLY)) != -1) {
            sprintf((char *)attr, "%d", sync_size); /* or sprintf(attr, "0x%x", sync_size); */
            write(fd, attr, strlen((const char *)attr));
            close(fd);
        }
    }

    {
        unsigned char  attr[1024];
        if ((fd  = open("/sys/class/u-dma-buf/udmabuf0/sync_direction", O_WRONLY)) != -1) {
            sprintf((char *)attr, "%d", sync_direction);
            write(fd, attr, strlen((const char *)attr));
            close(fd);
        }
    }

    {
        unsigned char  attr[1024];
        unsigned long  sync_for_cpu = sync_direction == 1? 1:0;
        if ((fd  = open("/sys/class/u-dma-buf/udmabuf0/sync_for_cpu", O_WRONLY)) != -1) {
            sprintf((char *)attr, "%d", sync_for_cpu);
            write(fd, attr, strlen((const char *)attr));
            close(fd);
        }
    }

    {
        unsigned char  attr[1024];
        unsigned long  sync_for_device = sync_direction == 1? 0:1;
        if ((fd  = open("/sys/class/u-dma-buf/udmabuf0/sync_for_device", O_WRONLY)) != -1) {
            sprintf((char *)attr, "%d", sync_for_device);
            write(fd, attr, strlen((const char *)attr));
            close(fd);
        }
    }

}
```

ikwzm, 2024-02-23:
Where is the value of *countBuf read and checked?

zhanghongg, 2024-02-24:
> Main logic code:
> 
>     unsigned char* buf = v4l2fd_.DqBuf();
>     ...
>     ... /// use this buf
>     ...
>     uint64_t* countBuf = reinterpret_cast<uint64_t*>(buf + stripBufSize - sizeof(uint64_t));
>     *countBuf =0;
>     v4l2fd_.QBuf();

After DqBuf,  Check at this location. " /// use this buf"

ikwzm, 2024-02-24:
Thank you.
By the way, this is a confirmation, Did it work correctly with V4L2_MEMORY_MMAP instead of V4L2_MEMORY_USERPTR?

ikwzm, 2024-02-24:
It is better not to use memset() to clear u-dma-bufs. > #38

ikwzm, 2024-02-24:
>> What is the device driver?
>
> Are you referring to the v4l2 driver？Is there anything I should pay attention to in this regard?

Yes, what v4l2 driver are you using?

> Firstly, is my usage of udmabuf roughly correct?

Specify sync_direction as 2 as follows: If the V4L2 driver is capturing, use DMA_FROM_DEVICE(=2).

```
void UdmaBuf::Sync(unsigned long sync_size, unsigned long sync_offset,
                   unsigned int sync_for_device) {
    int fd;
    {
        unsigned char  attr[1024];
        if ((fd  = open("/sys/class/u-dma-buf/udmabuf0/sync_offset", O_WRONLY)) != -1) {
            sprintf((char *)attr, "%d", sync_offset); /* or sprintf(attr, "0x%x", sync_offset); */
            write(fd, attr, strlen((const char *)attr));
            close(fd);
        }
    }
    {
        unsigned char  attr[1024];
        if ((fd  = open("/sys/class/u-dma-buf/udmabuf0/sync_size", O_WRONLY)) != -1) {
            sprintf((char *)attr, "%d", sync_size); /* or sprintf(attr, "0x%x", sync_size); */
            write(fd, attr, strlen((const char *)attr));
            close(fd);
        }
    }

    {
        unsigned char  attr[1024];
        unsigned int   sync_direction = 2; // DMA_FROM_DEVICE if V4L2_BUF_TYPE_VIDEO_CAPTURE
        if ((fd  = open("/sys/class/u-dma-buf/udmabuf0/sync_direction", O_WRONLY)) != -1) {
            sprintf((char *)attr, "%d", sync_direction);
            write(fd, attr, strlen((const char *)attr));
            close(fd);
        }
    }

    if (sync_for_device == 0) 
    {
        unsigned char  attr[1024];
        unsigned long  sync_for_cpu = 1;
        if ((fd  = open("/sys/class/u-dma-buf/udmabuf0/sync_for_cpu", O_WRONLY)) != -1) {
            sprintf((char *)attr, "%d", sync_for_cpu);
            write(fd, attr, strlen((const char *)attr));
            close(fd);
        }
    }
    else
    {
        unsigned char  attr[1024];
        unsigned long  sync_for_device = 1;
        if ((fd  = open("/sys/class/u-dma-buf/udmabuf0/sync_for_device", O_WRONLY)) != -1) {
            sprintf((char *)attr, "%d", sync_for_device);
            write(fd, attr, strlen((const char *)attr));
            close(fd);
        }
    }

}
```
> Secondly, in order to achieve my goal that improve the speed of memcpy, is my solution direction correct.

I believe the method is correct. 
However, it may not work with some V4L2 drivers. 
Therefore, please let me know what you are using for your V4L2 driver.

ikwzm, 2024-02-24:
There is another way to speed up mmap.
It is to use S_AXI_HPC0 or S_AXI_HPC1 or S_AXI_ACP for the PL -> PS interface to perform cache coherency in hardware and then set the dma-coherent property in the V4L2 driver device tree.
By doing so, the cache is also enabled in a way using V4L2_MEMORY_MMAP.

For more information on cache coherency, please refer to the following URL

https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18842098/Zynq+UltraScale+MPSoC+Cache+Coherency

zhanghongg, 2024-02-25:
Thank you.

> There is another way to speed up mmap.
> It is to use S_AXI_HPC0 or S_AXI_HPC1 or S_AXI_ACP for the PL -> PS interface to perform cache coherency in hardware and then set the dma-coherent property in the V4L2 driver device tree.
> By doing so, the cache is also enabled in a way using V4L2_MEMORY_MMAP.
> 
> Please refer to the above for details> https://github.com/ikwzm/udmabuf/issues/107
> 
> For more information on cache coherency, please refer to the following URL
> 
> https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18842098/Zynq+UltraScale+MPSoC+Cache+Coherency

OK! I have also considered this method, but the PL segment requires 4 axi_hp buses.

> Therefore, please let me know what you are using for your V4L2 driver.

The v4l2 driver has customized some logic to tell the PL end the writable buf address. Should I provide you with the driver source code? If so, I will provide the source code on Monday.

zhanghongg, 2024-02-26:
v4l2 driver:
```
#include <linux/module.h>
#include <linux/delay.h>
#include <linux/errno.h>
#include <linux/fs.h>
#include <linux/kernel.h>
#include <linux/slab.h>
#include <linux/mm.h>
#include <linux/ioport.h>
#include <linux/init.h>
#include <linux/sched.h>
#include <linux/pci.h>
#include <linux/random.h>
#include <linux/version.h>
#include <linux/mutex.h>
#include <linux/videodev2.h>
#include <linux/dma-mapping.h>
#include <linux/interrupt.h>
#include <linux/kthread.h>
#include <linux/highmem.h>
#include <linux/freezer.h>
#include <media/videobuf-vmalloc.h>
#include <media/v4l2-device.h>
#include <media/v4l2-ioctl.h>
#include <linux/platform_device.h>
#include <linux/gpio.h>
#include <linux/of_gpio.h>

#include "extractor_sw.h"
#include "extractor_hw.h"

/* 
 * trans video driver
 */
#define MAX_WIDTH   8192
#define MAX_HEIGHT  2048
 
#define CAPTURE_DRV_NAME "Strip driver"
#define PVI_MODULE_NAME "Strip"

void __iomem  *extractor_base = 0;
void __iomem  *sta_info = 0;

static struct extractor_port *file2port(struct file *file)
{
	return container_of(file->private_data, struct extractor_port, fh);
}

static void start_extractor(struct extractor_dev *xdev, struct extractor_buffer *buf, unsigned char buf_flag)
{
	dma_addr_t dma_addr;
	dma_addr = vb2_dma_contig_plane_dma_addr(&buf->v4l2_buf.vb2_buf, 0);
	if(buf_flag == 0) {
		iowrite32(dma_addr, extractor_base + XWR2DDR_CTRL_ADDR_DATA_DATA_A);
		iowrite32(dma_addr, extractor_base + XWR2DDR_CTRL_ADDR_DATA_DATA2_A);
	}
	else {
		iowrite32(dma_addr, extractor_base + XWR2DDR_CTRL_ADDR_DATA_DATA_B);
		iowrite32(dma_addr, extractor_base + XWR2DDR_CTRL_ADDR_DATA_DATA2_B);
	}
	//v4l2_info(&xdev->v4l2_dev, "dma_addr=0x%x\n", dma_addr);
}
/*
static void stop_dma(void)
{
}
*/
/*
 * Videobuf operations
[... 559 lines cut ...]
static struct platform_driver extractor_pdrv = {
	.probe		= extractor_probe,
	.remove		= extractor_remove,
	.driver		= {
		.name	= CAPTURE_DRV_NAME,
		.owner	= THIS_MODULE,
		.of_match_table = extractor_of_match,
	},
};

static int extractor_init(void)
{
	return platform_driver_register(&extractor_pdrv);
}
 
static void extractor_exit(void)
{
	platform_driver_unregister(&extractor_pdrv);
}
 
module_init(extractor_init);
module_exit(extractor_exit);
 
MODULE_LICENSE("GPL");
```

ikwzm, 2024-02-27:
I found that your V4L2 driver uses videobuf2-dma-contig.

```C
static int alloc_port(struct extractor_dev *xdev)
{
    :
	q->mem_ops = &vb2_dma_contig_memops;
    :    
}
```

With videobuf2-dma-contig, cache synchronization is automatically performed inside the Linux Kernel, so there is no need to explicitly sync u-dma-bufs on the user application side.

## vb2_ioctl_qbuf

Let's follow how vb2_ioctl_qbuf handles this.

### vb2_ioctl_qbuf()

https://elixir.bootlin.com/linux/v6.1.70/source/drivers/media/common/videobuf2/videobuf2-v4l2.c#L1052

```C
int vb2_ioctl_qbuf(struct file *file, void *priv, struct v4l2_buffer *p)
{
	struct video_device *vdev = video_devdata(file);

	if (vb2_queue_is_busy(vdev->queue, file))
		return -EBUSY;
	return vb2_qbuf(vdev->queue, vdev->v4l2_dev->mdev, p);
}
EXPORT_SYMBOL_GPL(vb2_ioctl_qbuf);
```
vb2_ioctl_qbuf() calls vb2_qbuf().

### vb2_qbuf()

https://elixir.bootlin.com/linux/v6.1.70/source/drivers/media/common/videobuf2/videobuf2-v4l2.c#L802

```C
int vb2_qbuf(struct vb2_queue *q, struct media_device *mdev,
	     struct v4l2_buffer *b)
{
	struct media_request *req = NULL;
	int ret;

	if (vb2_fileio_is_active(q)) {
		dprintk(q, 1, "file io in progress\n");
		return -EBUSY;
	}

	ret = vb2_queue_or_prepare_buf(q, mdev, b, false, &req);
	if (ret)
		return ret;
	ret = vb2_core_qbuf(q, b->index, b, req);
	if (req)
		media_request_put(req);
	return ret;
}
EXPORT_SYMBOL_GPL(vb2_qbuf);
```

vb2_qbuf() calls vb2_core_qbuf()

### vb2_core_qbuf()

https://elixir.bootlin.com/linux/v6.1.70/source/drivers/media/common/videobuf2/videobuf2-core.c#L1651

```C
int vb2_core_qbuf(struct vb2_queue *q, unsigned int index, void *pb,
		  struct media_request *req)
[... 243 lines cut ...]

	/* This takes care of DMABUF and user-enforced cache sync hint */
	if (buf->vb->skip_cache_sync_on_finish)
		return;

	if (!buf->non_coherent_mem)
		return;

	/* Non-coherent MMAP only */
	if (buf->vaddr)
		invalidate_kernel_vmap_range(buf->vaddr, buf->size);

	/* For both USERPTR and non-coherent MMAP */
	dma_sync_sgtable_for_cpu(buf->dev, sgt, buf->dma_dir);
}
```

vb2_dc_finish() calls dma_sync_sgtable_for_cpu() if memory type is USERPTR.
This function performs cache synchronization.

### Conclusion

With videobuf2-dma-contig, cache synchronization is automatically performed inside the Linux Kernel, so there is no need to explicitly sync u-dma-bufs on the user application side.

The cause of the problem is becoming more and more difficult to understand.

zhanghongg, 2024-02-29:
Many thanks.

If udmabuf requests an additional buf_size_ memory size than v4l2 requires, the problem will no longer occur.

Namely, apply for udmabuf with a size of (req_count+1) * buf'length_,

The number of bufs registered to V4L2 is req_count, with a size of buf_length_.
```

bool V4l2FdUserPtr::ReqBufs(uint32_t req_count) {
    uint64_t reserved_base_addr = 0;
    reserved_memory_ = std::make_shared<UdmaBuf>(reserved_base_addr, (req_count+1) * buf_length_);

    struct v4l2_requestbuffers reqBufs;
    memset(&reqBufs, 0, sizeof(struct v4l2_requestbuffers));
    reqBufs.count  = req_count;
    reqBufs.type   = V4L2_BUF_TYPE_VIDEO_CAPTURE;
    reqBufs.memory = V4L2_MEMORY_USERPTR;
    if (ioctl(fd_, VIDIOC_REQBUFS, &reqBufs)) {
        log_error("v4l2 VIDIOC_REQBUFS failed: %s", strerror(errno));
        return false;
    }
    video_buf_.resize(req_count);
    dq_video_bufs_.resize(req_count);
    for(int i = 0; i < reqBufs.count; i++){
        struct v4l2_buffer videoBuf;
        memset(&videoBuf, 0, sizeof(struct v4l2_buffer));
        videoBuf.index = i;
        videoBuf.type = V4L2_BUF_TYPE_VIDEO_CAPTURE;
        videoBuf.memory = V4L2_MEMORY_USERPTR;
        if(ioctl(fd_, VIDIOC_QUERYBUF, &videoBuf)){
            log_error("v4l2 VIDIOC_QUERYBUF failed: %s", strerror(errno));
            return false;
        }
        /// 每个buf的尺寸
        // buf_length_ = videoBuf.length;
        printf("VIDIOC_QUERYBUF videoBufSize: %d, SetFmtBufSize: %d\n", videoBuf.length, buf_length_);
        void* myPtr = reserved_memory_->GetMappAddr();
        video_buf_[i] = reinterpret_cast<unsigned char*>(myPtr) + i*buf_length_;
        videoBuf.m.userptr = reinterpret_cast<unsigned long>(video_buf_[i]);

        if(ioctl(fd_, VIDIOC_QBUF, &videoBuf)){
            log_error("v4l2 VIDIOC_QBUF failed: %s", strerror(errno));
            return false;
        }
    }

    return true;
}
```

ikwzm, 2024-02-29:
What is the next output result?
```C
        printf("VIDIOC_QUERYBUF videoBufSize: %d, SetFmtBufSize: %d\n", videoBuf.length, buf_length_);
```

What is the value of req_count?

What is the value of stripBufSize?

What is the next process?　Do you have the source code?
```C
    video_buf_.resize(req_count);
```

What is the next process?　Do you have the source code?
```C
    dq_video_bufs_.resize(req_count);
```

zhanghongg, 2024-03-01:
> What is the next output result?
> 
>         printf("VIDIOC_QUERYBUF videoBufSize: %d, SetFmtBufSize: %d\n", videoBuf.length, buf_length_);

Both are 16,777,216.

> What is the value of req_count?

During the testing process, there were 4.

> What is the value of stripBufSize?

It is also 16,777,216.

> What is the next process?　Do you have the source code?
> 
>     video_buf_.resize(req_count);

It is only used in DqBuf.

```
unsigned char* V4l2FdUserPtr::DqBuf() {
    memset(&active_video_buf_, 0, sizeof(struct v4l2_buffer));
    active_video_buf_.type = V4L2_BUF_TYPE_VIDEO_CAPTURE;
    active_video_buf_.memory = V4L2_MEMORY_USERPTR;
    if(ioctl(fd_, VIDIOC_DQBUF, &active_video_buf_)){
        log_error("v4l2 VIDIOC_DQBUF failed: %s", strerror(errno));
        return 0;
    }
    // reserved_memory_->Sync(buf_length_, active_video_buf_.index*buf_length_, 1);
    return video_buf_[active_video_buf_.index];
}
```

> What is the next process?　Do you have the source code?
> 
>     dq_video_bufs_.resize(req_count);

Please ignore it, it is currently not being used in any actual interface used.

ikwzm, 2024-03-01:
> What is the value of req_count?
>
>> During the testing process, there were 4.

What is the value of reqBuf.count after the next ioctl(fd_, VIDIOC_REQBUFS,&reqBufs)?

```C
bool V4l2FdUserPtr::ReqBufs(uint32_t req_count) {
    uint64_t reserved_base_addr = 0;
    reserved_memory_ = std::make_shared<UdmaBuf>(reserved_base_addr, (req_count+1) * buf_length_);

    struct v4l2_requestbuffers reqBufs;
    memset(&reqBufs, 0, sizeof(struct v4l2_requestbuffers));
    reqBufs.count  = req_count;
    reqBufs.type   = V4L2_BUF_TYPE_VIDEO_CAPTURE;
    reqBufs.memory = V4L2_MEMORY_USERPTR;
    if (ioctl(fd_, VIDIOC_REQBUFS, &reqBufs)) {
        log_error("v4l2 VIDIOC_REQBUFS failed: %s", strerror(errno));
        return false;
    }
    video_buf_.resize(req_count);
    dq_video_bufs_.resize(req_count);
    for(int i = 0; i < reqBufs.count; i++){
```
 Is it the same value as req_count?

ikwzm, 2024-03-01:
```C
    reserved_memory_ = std::make_shared<UdmaBuf>(reserved_base_addr, (req_count+1) * buf_length_);
```

Where is the source code for the UdmaBuf class?

zhanghongg, 2024-03-05:
> Is it the same value as req_count?

yes, it is.

> Where is the source code for the UdmaBuf class?

```
UdmaBuf::UdmaBuf(uint64_t base_addr, uint64_t map_size): base_addr_(base_addr), map_size_(map_size) {
    fd_ = open("/dev/udmabuf0", O_RDWR);
    if (fd_ < 0) {   
        log_error("UdmaBuf mmap!");
        assert(false);
    }
    user_ptr_ = mmap64(NULL, map_size_, PROT_READ | PROT_WRITE, MAP_SHARED, fd_, base_addr_);
    if(user_ptr_ != MAP_FAILED) {
        log_info("UdmaBuf mmap succuffully!");
        ClearBuf();
    }
    else {
        log_error("UdmaBuf mmap failed!"); 
    }
}
```

ikwzm, 2024-03-05:
Where is ClearBuf located?
Can you show me the source code for "all" of the Udmabuf class, not just the constructor?

zhanghongg, 2024-03-05:
```
#include "UdmaBuf.h"
#include <unistd.h>
#include <fcntl.h>
#include <sys/mman.h>
#include <assert.h>
#include "log.h"
#include <string.h>
UdmaBuf::UdmaBuf(uint64_t base_addr, uint64_t map_size): base_addr_(base_addr), map_size_(map_size) {
    fd_ = open("/dev/udmabuf0", O_RDWR);
    if (fd_ < 0) {   
        log_error("UdmaBuf mmap!");
        assert(false);
    }
    user_ptr_ = mmap64(NULL, map_size_, PROT_READ | PROT_WRITE, MAP_SHARED, fd_, base_addr_);
    if(user_ptr_ != MAP_FAILED) {
        log_info("UdmaBuf mmap succuffully!");
        ClearBuf();
    }
    else {
        log_error("UdmaBuf mmap failed!"); 
    }
}

UdmaBuf::~UdmaBuf() {
    close(fd_);
    munmap(user_ptr_, map_size_);
}

void* UdmaBuf::GetMappAddr() {
    return user_ptr_;
}

void UdmaBuf::SetSyncMode(uint8_t mode){

    unsigned char attr[1024];
    int tmpfd_;
    unsigned long  sync_mode = mode;
    if ((tmpfd_  = open("/sys/class/u-dma-buf/udmabuf0/sync_mode", O_WRONLY)) != -1) {
        sprintf((char *)attr, "%d", sync_mode);            
        write(tmpfd_, attr, strlen((const char *)attr));
        close(tmpfd_);
    }
    else {
        log_error("SetSyncMode failed!"); 
    }
}

void UdmaBuf::ClearBuf() {
    /// no memset!!
    unsigned char* tmp = reinterpret_cast<unsigned char*>(user_ptr_);
            memset(tmp,0,map_size_);
    // for(uint64_t i=0; i < map_size_; ++i) {
    //     memset(tmp,0,map_size);
    //     tmp[i] = 0;
    // }            
}

void UdmaBuf::Sync(unsigned long sync_size, unsigned long sync_offset,
                   unsigned int sync_direction) {
    int fd;
    {
        unsigned char  attr[1024];
        if ((fd  = open("/sys/class/u-dma-buf/udmabuf0/sync_offset", O_WRONLY)) != -1) {
            sprintf((char *)attr, "%d", sync_offset); /* or sprintf(attr, "0x%x", sync_offset); */
            write(fd, attr, strlen((const char *)attr));
            close(fd);
        }
    }
    {
        unsigned char  attr[1024];
        if ((fd  = open("/sys/class/u-dma-buf/udmabuf0/sync_size", O_WRONLY)) != -1) {
            sprintf((char *)attr, "%d", sync_size); /* or sprintf(attr, "0x%x", sync_size); */
            write(fd, attr, strlen((const char *)attr));
            close(fd);
        }
    }

    {
        unsigned char  attr[1024];
        if ((fd  = open("/sys/class/u-dma-buf/udmabuf0/sync_direction", O_WRONLY)) != -1) {
            sprintf((char *)attr, "%d", sync_direction);
            write(fd, attr, strlen((const char *)attr));
            close(fd);
        }
    }

    {
        unsigned char  attr[1024];
        unsigned long  sync_for_cpu = sync_direction == 1? 1:0;
        if ((fd  = open("/sys/class/u-dma-buf/udmabuf0/sync_for_cpu", O_WRONLY)) != -1) {
            sprintf((char *)attr, "%d", sync_for_cpu);
            write(fd, attr, strlen((const char *)attr));
            close(fd);
        }
    }

    {
        unsigned char  attr[1024];
        unsigned long  sync_for_device = sync_direction == 1? 0:1;
        if ((fd  = open("/sys/class/u-dma-buf/udmabuf0/sync_for_device", O_WRONLY)) != -1) {
            sprintf((char *)attr, "%d", sync_for_device);
            write(fd, attr, strlen((const char *)attr));
            close(fd);
        }
    }

}
```

ikwzm, 2024-03-05:
umm...
Did I make a bad point? 
Where is u-dma-buf.h?

zhanghongg, 2024-03-05:
Sorry.

```
#pragma once
#include <cstdint>
#include <stdint.h>

class UdmaBuf {
public:
    UdmaBuf(uint64_t base_addr, uint64_t map_size);
    virtual ~UdmaBuf();
    void* GetMappAddr();
    uint64_t GetMapSize() const {
        return map_size_;
    }
    void ClearBuf();

    void Sync(unsigned long sync_size, unsigned long sync_offset, unsigned int sync_direction);
protected:
     

    // As listed below, sync_mode can be used to configure the cache behavior when the O_SYNC flag is present in open():
    // sync_mode=0: CPU cache is enabled regardless of the O_SYNC flag presence.
    // sync_mode=1: If O_SYNC is specified, CPU cache is disabled. If O_SYNC is not specified, CPU cache is enabled.
    // sync_mode=2: If O_SYNC is specified, CPU cache is disabled but CPU uses write-combine when writing data to DMA buffer improves performance by combining multiple write accesses. If O_SYNC is not specified, CPU cache is enabled.
    // sync_mode=3: If O_SYNC is specified, DMA coherency mode is used. If O_SYNC is not specified, CPU cache is enabled.
    // sync_mode=4: CPU cache is enabled regardless of the O_SYNC flag presence.
    // sync_mode=5: CPU cache is disabled regardless of the O_SYNC flag presence.
    // sync_mode=6: CPU uses write-combine to write data to DMA buffer regardless of O_SYNC presence.
    // sync_mode=7: DMA coherency mode is used regardless of O_SYNC presence.
    void SetSyncMode(uint8_t mode);
private:
    uint64_t base_addr_;
    uint64_t map_size_;
    int fd_;
    void* user_ptr_;
};

```

ikwzm, 2024-03-06:
The source code you gave us uses memset to clear u-dma-buf, have you checked the behavior without memset?
If not, please check the behavior again with Clear without memset.

zhanghongg, 2024-03-06:
Yes, I have checked.
Nothing happened, and it takes a long time.

```
void UdmaBuf::ClearBuf() {
    /// no memset!!
    unsigned char* tmp = reinterpret_cast<unsigned char*>(user_ptr_);
    for(uint64_t i=0; i < map_size_; ++i) {
        tmp[i] = 0;
    }            
}

```

ikwzm, 2024-03-06:
> Yes, I have checked.
> Nothing happened, and it takes a long time.

Nothing happend? 
Has the buffer been cleared?
Did the value written by PL match the value read by the CPU?

zhanghongg, 2024-03-06:
The phenomenon is consistent.

> Has the buffer been cleared? 

yes, it has been cleared.

> Did the value written by PL match the value read by the CPU?

mismatching.

ikwzm, 2024-10-23:
Sorry, I know that V4L2_MEMORY_USERPTR + u-dma-buf will not work on newer versions of Linux.
The following fix was made to the Linux Kernel around December 2022 and applied starting with Linux Kernel 5.15.

 * [[PATCH AUTOSEL 6.0 13/13] v4l2: don't fall back to follow_pfn() if pin_user_pages_fast() fails](https://lore.kernel.org/lkml/20221206094916.987259-13-sashal@kernel.org/)

As a result of this fix, the method of using V4L2_MEMORY_USERPTR by allocating a buffer with u-dma-buf no longer works.

The above post points out that V4L2_MEMORY_USERPTR was originally deprecated and that the fallback to VM_PFMAP/VM_IO is fundamentally dangerous. As a result, it appears that the restrictions on using memory areas on userspace as buffers for V4L2 have been tightened, and as a result, buffers allocated with u-dma-buf can no longer be used.

With this change, I can no longer run V4L2_MEMORY_USERPTR + u-dma-buf in our environment. Therefore, I cannot reproduce your issue in my environment.

The problem of slow V4L2 running on ZynqMP/Zynq/RaspberryPi has a solution in newer Linux using dma-heap.
Here is an article describing this solution. It is written in Japanese, but please translate it into your language.

 * [[Solving the problem of no performance with V4L2 streaming I/O with dma-heap]@Qiita](https://qiita.com/ikwzm/items/00359768a7c982a4ddc0)

zhanghongg, 2024-10-24:
Great! I have read your solution. I plan to try it in kernel version 5.15.

Could you please help me confirm whether my current driver can be adapted to this modification? Thank you very much.

[[](url)](https://github.com/ikwzm/udmabuf/issues/116#issuecomment-1963418931)