ikwzm/udmabuf #118: Difference with upstream driver

enceladus13, 2024-07-05:
Hi,

What is the main difference of this driver with the mainline/upstream driver?

Here is the subsystem: https://github.com/torvalds/linux/tree/master/drivers/dma-buf

And here is the udmabuf.c file : https://github.com/torvalds/linux/blob/master/drivers/dma-buf/udmabuf.c

Due to the size of the files, I struggle to understand what are the differences between the 2.

Thanks in advance!

ikwzm, 2024-07-05:
They are quite different.

enceladus13, 2024-07-05:
Hi!

Could you be a bit more specific? I have a project where I need to access from userspace DMA capable RAM and your driver seems like a good fit. I am trying to understand though what is the main difference with the already upstreamed driver.

I understand that the upstream driver is a full subsystem that can be used by other kernel subsystems. What is the main difference that makes this more suitable though? Because I saw this udmabuf.c file and I am trying to understand if in the end the kernel driver already offers what this driver offers

ikwzm, 2024-07-05:
Sorry, I don't know much about udmabuf that is upstreamed to the mainline.

My driver was released in 2015 under the name udmabuf. 
The purpose of my driver is to allocate contiguous blocks of memory in kernel space as DMA buffers and make them available from user space.

After that, Linux Kernel 4.20, released in 2018, added udmabuf, which has the same name but a different functionality.
I don't know much about what this newly added udmabuf does.
However, having the same name is bad enough, so I renamed my driver from udmabuf to u-dma-buf.

If you want to know the difference between these two drivers, please examine the udmabuf upstreamed to the mainline yourself.