ikwzm/udmabuf #120: memcpy from udmabuf mmap buffer slow than buffer from malloc

joshuafc, 2024-08-23:
ZYNQ 7035

```
/ {
        reserved-memory {
                #address-cells = <1>;
                #size-cells = <1>;
                ranges;
                shared_to_pl_buf0: shared_to_pl_buf@0 {
                        compatible = "shared-dma-pool";
                        reusable;
                        reg = <0x1b000000 0x04000000>;
                        label = "shared_to_pl_buf0";
                        //linux,cma-default;
                };
        };

        udmabuf@0 {
                compatible = "ikwzm,u-dma-buf";
                device-name = "shared_to_pl_buf";
                size = <0x04000000>; 
                memory-region = <&shared_to_pl_buf0>;
                dma-coherent;
        };
};
```

```
#define _POSIX_C_SOURCE 199309L
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <unistd.h>
#include <sys/types.h>
#include <sys/stat.h>
#include <fcntl.h>
#include <sys/mman.h>
#include <time.h>

#define PORT 18081
#define BUFFER_SIZE (16 * 1024 * 1024)
#define COPY_COUNT 32

int main() {
    char *udmabuf_buffer_cached;
    char *udmabuf_buffer_nocached;
    char *malloc_buffer = (char*)malloc(BUFFER_SIZE);
    char *dst = (char*)malloc(BUFFER_SIZE);
    
    int fd1 = 0;
    if ((fd1  = open("/dev/shared_to_pl_buf", O_RDWR)) != -1) {
        udmabuf_buffer_cached = mmap(NULL, BUFFER_SIZE, PROT_READ, MAP_SHARED, fd1, 0);
    }

    int fd2 = 0;
    if ((fd2  = open("/dev/shared_to_pl_buf", O_RDWR | O_SYNC)) != -1) {
        udmabuf_buffer_nocached = mmap(NULL, BUFFER_SIZE, PROT_READ, MAP_SHARED, fd2, 0);
    }

    // --------------- //

    struct timespec start, end;
    clock_gettime(CLOCK_MONOTONIC, &start);

    for (int i = 0; i < COPY_COUNT; ++i) {
        memcpy(dst, udmabuf_buffer_cached, BUFFER_SIZE);
    }

    clock_gettime(CLOCK_MONOTONIC, &end);

    long seconds = end.tv_sec - start.tv_sec;
    long nanoseconds = end.tv_nsec - start.tv_nsec;
    double elapsed = seconds + nanoseconds*1e-9;

    printf("UDMABUF with cache: memcpy cost: %.9f seconds\n", elapsed);

    // --------------- //

    clock_gettime(CLOCK_MONOTONIC, &start);

    for (int i = 0; i < COPY_COUNT; ++i) {
        memcpy(dst, udmabuf_buffer_nocached, BUFFER_SIZE);
    }

    clock_gettime(CLOCK_MONOTONIC, &end);

    seconds = end.tv_sec - start.tv_sec;
    nanoseconds = end.tv_nsec - start.tv_nsec;
    elapsed = seconds + nanoseconds*1e-9;

    printf("UDMABUF without cache: memcpy cost: %.9f seconds\n", elapsed);

    // --------------- //

    clock_gettime(CLOCK_MONOTONIC, &start);

    for (int i = 0; i < COPY_COUNT; ++i) {
        memcpy(dst, malloc_buffer, BUFFER_SIZE);
    }

    clock_gettime(CLOCK_MONOTONIC, &end);

    seconds = end.tv_sec - start.tv_sec;
    nanoseconds = end.tv_nsec - start.tv_nsec;
    elapsed = seconds + nanoseconds*1e-9;

    printf("MALLOC: memcpy cost: %.9f seconds\n", elapsed);

    return 0;
}
```

```
UDMABUF with cache: memcpy cost: 1.433987362 seconds
UDMABUF without cache: memcpy cost: 7.333746588 seconds
MALLOC: memcpy cost: 0.262290923 seconds
```

ikwzm, 2024-08-23:
Thank you for the issue.
This is a very interesting issue.

I confirmed the same issue here with ZYBO-Z7.

```consolegcc test-org.c -o test-org
fpga@debian-fpga:~/work/u-dma-buf-test$ ./test-org
UDMABUF with cache: memcpy cost: 1.537138997 seconds
UDMABUF without cache: memcpy cost: 7.634434165 seconds
MALLOC: memcpy cost: 0.271438302 seconds
```

Of particular interest was the malloc_buffer bandwidth.
This program will result in a transfer of 16MB x 32 x 2 = 1024MB since BUFFER_SIZE = 16MB, COPY_COUNT = 32, and memcpy does reads and writes simultaneously.
In the case of malloc_buffer, this is completed in 0.27 seconds, resulting in a bandwidth of 3800 MB/sec.
This bandwidth clearly far exceeds the DRAM bandwidth of the ZYNQ-7000.
I assume that something special is going on.

As a trial, I initialized malloc_buffer with memset as follows and then memcpy.

```#define _POSIX_C_SOURCE 199309L
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <unistd.h>
#include <sys/types.h>
#include <sys/stat.h>
#include <fcntl.h>
#include <sys/mman.h>
#include <time.h>

#define BUFFER_SIZE (16 * 1024 * 1024)
#define COPY_COUNT 32

int main() {
    char *udmabuf_buffer_cached;
    char *udmabuf_buffer_nocached;
    char *malloc_buffer = (char*)malloc(BUFFER_SIZE);
    char *dst = (char*)malloc(BUFFER_SIZE);
    
    int fd1 = 0;
    if ((fd1  = open("/dev/shared_to_pl_buf", O_RDWR)) != -1) {
        udmabuf_buffer_cached = mmap(NULL, BUFFER_SIZE, PROT_READ, MAP_SHARED, fd1, 0);
    }

    int fd2 = 0;
    if ((fd2  = open("/dev/shared_to_pl_buf", O_RDWR | O_SYNC)) != -1) {
        udmabuf_buffer_nocached = mmap(NULL, BUFFER_SIZE, PROT_READ, MAP_SHARED, fd2, 0);
    }

    // --------------- //

    struct timespec start, end;
    clock_gettime(CLOCK_MONOTONIC, &start);

    for (int i = 0; i < COPY_COUNT; ++i) {
        memcpy(dst, udmabuf_buffer_cached, BUFFER_SIZE);
    }

    clock_gettime(CLOCK_MONOTONIC, &end);

    long seconds = end.tv_sec - start.tv_sec;
    long nanoseconds = end.tv_nsec - start.tv_nsec;
    double elapsed = seconds + nanoseconds*1e-9;

    printf("UDMABUF with cache: memcpy cost: %.9f seconds\n", elapsed);

    // --------------- //

    clock_gettime(CLOCK_MONOTONIC, &start);

    for (int i = 0; i < COPY_COUNT; ++i) {
        memcpy(dst, udmabuf_buffer_nocached, BUFFER_SIZE);
    }

    clock_gettime(CLOCK_MONOTONIC, &end);

    seconds = end.tv_sec - start.tv_sec;
    nanoseconds = end.tv_nsec - start.tv_nsec;
    elapsed = seconds + nanoseconds*1e-9;

    printf("UDMABUF without cache: memcpy cost: %.9f seconds\n", elapsed);

    // --------------- //
    memset(malloc_buffer, 0xCC, BUFFER_SIZE);
    clock_gettime(CLOCK_MONOTONIC, &start);

    for (int i = 0; i < COPY_COUNT; ++i) {
        memcpy(dst, malloc_buffer, BUFFER_SIZE);
    }

    clock_gettime(CLOCK_MONOTONIC, &end);

    seconds = end.tv_sec - start.tv_sec;
    nanoseconds = end.tv_nsec - start.tv_nsec;
    elapsed = seconds + nanoseconds*1e-9;

    printf("MALLOC: memcpy cost: %.9f seconds\n", elapsed);

    return 0;
}
```

The results were as follows

```console
shell$ gcc test-new.c -o test-new
shell$ ./test-new
UDMABUF with cache: memcpy cost: 1.592706256 seconds
UDMABUF without cache: memcpy cost: 8.096199140 seconds
MALLOC: memcpy cost: 1.543264081 seconds
```
The bandwidth for malloc_buffer is now the same as for udmabuf_cached.

Since malloc_buffer was not initialized, the compiler or library may have recognized it as an undefined state and performed some special optimization.
I would like to figure out what is going on, but time is short. Please give me more time.

ikwzm, 2024-08-24:
I now know why memcpy is fast when using an uninitialized buffer as the transfer source.

When malloc allocates a large size buffer, it requests memory (pages) from the linux kernel with the mmap system call.
The linux kernel does not immediately allocate a physical page when a page is requested. 
Instead, it prepares only one physical page and temporarily allocates that physical page to all virtual pages.
Then, only when a write occurs to a virtual page, it catches the exception that occurred and then assigns a new physical page.
This is COW (Copy On Write).

This means that all virtual areas are assigned the same single page until they are initialized after allocating pages with malloc.
If memcpy is performed using the virtual area in this state as the transfer source, all reads are performed to the same page. Moreover, since the size of one page is smaller than that of L1-Cache, all reads are placed in the cache.

Therefore, memcpy effectively causes only write transactions to the DRAM.
Access to DRAM requires turnaround time between reads and writes, so if reads and writes are mixed, transfer efficiency drops.
However, if only write accesses occur, transfers can be performed up to the bus bandwidth limit.

This is why memcpy is fast when using an uninitialized buffer as the transfer source.

Can I close this issue since it has nothing to do with u-dma-buf?

joshuafc, 2024-08-25:
Thank you for your detailed analysis. I tried using memset on a buffer allocated by malloc and indeed got the same performance results as you.

I recently tried submitting a u-dma-buf mapped buffer to the TCP stack with MSG_ZEROCOPY to avoid copying from user space to kernel space.

From https://stackoverflow.com/questions/58627200/zero-copy-user-space-tcp-send-of-dma-mmap-coherent-mapped-memory , I learned that:

the underlying problem is that zerocopy networking does not work for memory that has been mapped using remap_pfn_range() (which dma_mmap_coherent() happens to use under the hood as well). The reason is that this type of memory (with the VM_PFNMAP flag set) does not have metadata in the form of struct page* associated with each page, which it needs. The solution then is to allocate the memory in a way that struct page*s are associated with the memory.

Therefore, I had to try copying the data into a buffer allocated by malloc in user space and sending that buffer with MSG_ZEROCOPY. I'm wondering if this can improve performance.

However, none of the tests used memset on the buffer, so all my experimental results are wrong. Thank you for reminding me about memset.

joshuafc, 2024-08-25:
Without the MSG_ZEROCOPY, malloced buffer can archive TCP send at 750Mbites/second
With the MSG_ZEROCOPY, malloced buffer can archive TCP send at 891Mbites/second
Without the MSG_ZEROCOPY, u-dma-buf buffer can archive TCP send at 755Mbites/second
With the MSG_ZEROCOPY, u-dma-buf buffer will get a "Bad Address" error when call to send.

With a memcpy from u-dma-buf buffer to malloced buffer, each time copy 16 MiB, with the MSG_ZEROCOPY, only archive TCP send at 655Mbites/second

This is a different topic now, so I'll close this issue.

closed 2024-08-25 by joshuafc