ikwzm/udmabuf #122: Unresolved dma_buf_export, dma_buf_fd & dma_buf_put in latest tag 4.8.0

stevejames, 2024-10-14:
Hello Kawazome san,
This is just to let you know that today I tried latest tag v4.8.0 and found these missing:
    ERROR: modpost: "dma_buf_export" [udmabuf-4.8.0/u-dma-buf.ko] undefined!
    ERROR: modpost: "dma_buf_fd" [udmabuf-4.8.0/u-dma-buf.ko] undefined!
    ERROR: modpost: "dma_buf_put" [udmabuf-4.8.0/u-dma-buf.ko] undefined!
It seems these are indeed missing from u-dma-buf.c. I have no problem with the preceding tag v4.7.3.
Thank you sincerely for time maintaining this package!
Regards, Steve.

ikwzm, 2024-10-14:
Thank you for the issue.

Sorry, I did not encounter such a problem in my environment. 
I have confirmed the operation in the following environment here.

 * https://github.com/ikwzm/ZynqMP-FPGA-Ubuntu22.04-Desktop/tree/v3.0.0
 * https://github.com/ikwzm/ZynqMP-FPGA-Debian11/tree/v1.0.1

Please tell me in what environment did you encounter that problem?

stevejames, 2024-10-17:
Hello Kawazome san,

Thank you for your reply.

The environment I was using was NXP's linux-lf-6.1.55-2.2.0 [1] built with OpenWrt v23.05.2 [2].
However since then I have moved the kernel forward to lf-6.6.36-2.1.0 [3] and I did not experience this issue.

But to be honest I'm quite sure that the issue I raised here was the result of a bad build on my part while switching between combinations of kernel version and kernel configuration. So my apologies for the unwarranted noise! I will close this issue as invalid.

Sorry again,
Steve.

[1] - https://github.com/nxp-qoriq/linux/releases/tag/lf-6.1.55-2.2.0
[2] - https://github.com/openwrt/openwrt/tree/v23.05.2
[3] - https://github.com/nxp-qoriq/linux/releases/tag/lf-6.6.36-2.1.0

closed 2024-10-17 by stevejames

ikwzm, 2024-10-17:
Thank you very much.

I did a little research and found a config named TRIM_UNUSED_KSYMS in kernel/module/Kconfig.

```
config TRIM_UNUSED_KSYMS
	bool "Trim unused exported kernel symbols" if EXPERT
	depends on !COMPILE_TEST
	help
	  The kernel and some modules make many symbols available for
	  other modules to use via EXPORT_SYMBOL() and variants. Depending
	  on the set of modules being selected in your kernel configuration,
	  many of those exported symbols might never be used.

	  This option allows for unused exported symbols to be dropped from
	  the build. In turn, this provides the compiler more opportunities
	  (especially when using LTO) for optimizing the code and reducing
	  binary size.  This might have some security advantages as well.

	  If unsure, or if you need to build out-of-tree modules, say N.

```

If I set this to 'y' and build, the unused symbol information among the EXPORT_SYMBOLed symbols will not be included in Module.symvers.
On Linux built in this way, undefined may occur at modpost time when building to add Kernel Module later.
Perhaps this could have been 'y'.