ikwzm/udmabuf #123: Unresolved dma_buf_export, dma_buf_fd & dma_buf_put in latest tag 4.8.0

jmwilson, 2024-10-23:
Revisiting #122, one explanation is that these 3 symbols are defined in `drivers/dma-buf`, which is conditionally built when `CONFIG_DMA_SHARED_BUFFER=y`. For in-tree build, add `select DMA_SHARED_BUFFER` to the Kconfig file to guarantee these are available.

ikwzm, 2024-10-23:
Thank you for your valuable advice.

Please give me some more time to consider.

ikwzm, 2024-10-24:
This has been tentatively fixed in https://github.com/ikwzm/udmabuf/commit/28c82418a01d3777592e46b5d9b5f5f328bab01d
Changes to Kconfig are still under consideration.

ikwzm, 2024-10-30:
https://github.com/ikwzm/udmabuf/tree/v5.0.0 has been released to solve this issue.
v5.0.0 allows selection of features via Kconfig.
This includes, following your advice, `select DMA_SHARED_BUFFER` if `U_DMA_BUF_EXPORT=y`.
Thanks for your advice.

closed 2024-10-30 by ikwzm