ikwzm/udmabuf #126: Aligned memory and size of CMA

pierredaye, 2024-12-02:
Dear @ikwzm,

First of all, thank you for your amazing work! Your driver helps me a lot in the design of my system on an ultrascale+.
As you are an expert on the memory management, I was wondering if you knew the requirements to set up a reserved memory region that could be used by your driver. From my understanding, I though that they had to be aligned on 0x1000 (409 bytes). 
However, when setting the reserved memory in my device tree like this:
```
reserved-memory {
        #address-cells = <2>;
        #size-cells = <2>;
        ranges;

        image_buf0: image_buf@0 {
                compatible = "shared-dma-pool";
                reusable;
                reg = <0x0 0x40000000 0x0 0x4C0000>;
                label = "image_buf0";
        };

        image_buf1: image_buf@1 {
                compatible = "shared-dma-pool";
                reusable;
                reg = <0x0 0x41400000 0x0 0x25C00000>;
                label = "image_buf1";
        };
};
```
The reserved memory for image_buf0 could not be created by the kernel (and therefore, the udma buffer could not be created). I also read that the size and start address should be of a power of 2. But image_buf1 does not have a size corresponding to a power of two...

Therefore, I am confused. Do you have any information on the requirements for start address and size?

Thanks

ikwzm, 2024-12-02:
Thank you for the issue.

The Linux CMA region has alignment constraints.
For MPSoC, the address and size that can be specified in the regs property of reserved-memory must be in units of 0x00400000 (4MiB).

Try change the size to 0x00800000 from 0x4C0000  as follows:

```
reserved-memory {
        #address-cells = <2>;
        #size-cells = <2>;
        ranges;

        image_buf0: image_buf@0 {
                compatible = "shared-dma-pool";
                reusable;
                reg = <0x0 0x40000000 0x0 0x00800000>;
                label = "image_buf0";
        };

        image_buf1: image_buf@1 {
                compatible = "shared-dma-pool";
                reusable;
                reg = <0x0 0x41400000 0x0 0x25C00000>;
                label = "image_buf1";
        };
};
```

pierredaye, 2025-01-29:
Thank you for your answer. This solved my issue!

closed 2025-01-29 by pierredaye