ikwzm/udmabuf #127: The dma-coherent default setting

TieraLai, 2025-01-09:
Hi,

I'm using the u-dma-buf module on an OS with Linux kernel version 5.15.136, running on an ARM Cortex-A78AE CPU.

I install the module using the insmod command and plan to use udmabuf as the PCI RX/TX DMA buffer.

However, I encountered an issue: when I load the module with insmod, the default value of dma-coherent is set to 0.

How can I change the value of dma-coherent?

P.S. I'm a newcomer trying to figure this out. Any guidance would be greatly appreciated.

ikwzm, 2025-01-09:
Thanks for the issue.

If u-dma-buf is created by the argument of the insmod command, the value of dma-coherent is set to the Linux Kernel default value. In principle, it is impossible to change this value later.

However, if you add a u-dma-buf under a bus (e.g., simple-bus such as /soc) using the device tree, adding a property called dma-coherent to the u-dma-buf node will set dma-coherent to 1 and adding the property dma-noncoherent to the u-dma-buf node will set dma-coherent to 0.

See of_dma_configure() and of_dma_is_coherent() in Linux Kernel for details.

ikwzm, 2025-02-17:
Thank you for the issue.
Sorry, the current u-dma-buf is not guaranteed to work under iommu.